zephyrproject-rtos/zephyr #17719: Document how to use native posix target to create FS images

vanwinkeljan, 2019-07-23:
**Is your enhancement proposal related to a problem? Please describe.**
A byproduct of the flash + fuse support for the native_posix is that initial target FS images can be created on a linux host system but no documentation exists how to achieve this.

**Describe the solution you'd like**
Update native_posix board documentation.

**Describe alternatives you've considered**
Update fs shell sample documentation.

pkral78, 2019-07-23:
Note that it would be useful to create standalone utility for that purpose. What I'm not sure is whatever it should be added as another sample in subsys/fs or as separate utility (build itself on top of Zephyr) and be part of SDK. However, the former should be good start. The utility itself should be able to prepare flash filesystem in binary/hex/elf form for separate handling or further merging with targer firmware.

Another step might be to extend build system (aka Zephyr macros) with related task.

pkral78, 2019-07-23:
The utility should support as many flash fs types as available in that branch. Some fs uses direct flash device, ELM FAT uses disk layer to access flash. FAT has already plenty of tools available around so it seem to be superfluous to include support for it, however one general utility has also its advantages in build pipeline.

vanwinkeljan, 2019-07-23:
> The utility itself should be able to prepare flash filesystem in binary/hex/elf form for separate handling or further merging with targer firmware.

I was thinking to add a new west command to take care of the extraction of the FS partition, if possible (to be investigated)

> The utility should support as many flash fs types as available in that branch

This is the advantage of using the native_posix board is that it will support any FS available in the zephyr tree as it is just provides a flash driver at the lower level and the host fs access uses the high level zephyr FS API. Only question that remains is where to put the utility.

pkral78, 2019-07-24:
> This is the advantage of using the native_posix board is that it will support any FS available in the zephyr tree as it is just provides a flash driver at the lower level and the host fs access uses the high level zephyr FS API. Only question that remains is where to put the utility.

Of course, but the utility have to deal with filesystem structures initialization which is always FS specific.

closed 2024-02-09 by aescolar