zephyrproject-rtos/zephyr #17732: cannot use bt_conn_security in connected callback

maz3max, 2019-07-23:
**Describe the bug**
I preprogrammed a BLE SC bonding on a central and a peripheral device and wanted to enable encryption when the connection is established. For doing this, I called `bt_conn_security` in the connected callback. This seems to result in an error which terminates the connection.

**To Reproduce**
* have to devices already bonded with BLE SC
* both have corresponding `bt_keys` objects with keys: 34, flags: 17
* in the `connected` callback of one both devices, add
```
int ret = bt_conn_security(conn, BT_SECURITY_FIPS);
if (ret) {
    LOG_INF("Kill connection: insufficient security %i", ret);
    bt_conn_disconnect(conn, BT_HCI_ERR_INSUFFICIENT_SECURITY);
} else {
    LOG_INF("bt_conn_security successful");
}
```
* let peripheral advertise, central connects on receipt of adv.
* the devices disconnect shortly after
* if both device enforce encryption like this, `bt_conn_security` fails on the peripheral with code -22 (EINVAL) causing the app the kill the connection
* if only one of both devices enforces encryption,
    * central gets disconnected with code 8 (BT_HCI_ERR_CONN_TIMEOUT)
    * peripheral gets disconnected with code 61 (BT_HCI_ERR_TERM_DUE_TO_MIC_FAIL)

**Expected behavior**
* the devices set up the security level successfully and stay connected.

**Impact**
* I have to restructure my whole app idea if I cannot find a workaround.

**Screenshots or console output**
Are posted here: https://github.com/zephyrproject-rtos/zephyr/issues/15256#issuecomment-514254728
and here: https://github.com/zephyrproject-rtos/zephyr/issues/15256#issuecomment-514256427

joerchan, 2019-07-24:
To summarize my investigation into this issue, there are 2 separate issues:

> if both device enforce encryption like this, bt_conn_security fails on the peripheral with code -22 (EINVAL) causing the app the kill the connection

The peripheral side of `bt_conn_security` fails with FIPS level because the requested security cannot be reached since no pairing callbacks has been set, which would result in a Just Works pairing (no input no output). 
Possible workarounds;
 - Only request security from central side
 - Request Low security from peripheral side

> if only one of both devices enforces encryption,
central gets disconnected with code 8 (BT_HCI_ERR_CONN_TIMEOUT)
peripheral gets disconnected with code 61 (BT_HCI_ERR_TERM_DUE_TO_MIC_FAIL)

This happened because the central sends LL control procedure PDUs after sending LL_ENC_REQ. This is invalid behaviour of the peer.
This should be handled by issue #15256

aescolar, 2019-07-24:
CC @cvinayak

joerchan, 2019-07-29:
Added PR for 1#
2# should be handled by different issue.

ioannisg, 2019-07-30:
> 
> 
> Added PR for 1#
> 2# should be handled by different issue.

@joerchan your PR will close this issue, though. So we need to report 2# to another issue, right?

PR #17836 (2019-07-31): Bluetooth: SMP: Fix peripheral security request with authenticated keys
The call to bt_conn_security will send a SMP security request for
peripheral, and for central it will initiate LL encryption.
A call to bt_conn_security with no IO capabilities but authenticated
keys has been distributed, would succeed on central side, but fail on
peripheral side with error code -22.

The keys could have been either:
 - Preprogrammed
 - IO capabilities may have changed.
 - OOB bonding may been used.

Fix so that Peripheral can send a security request if the bond
information is already established.

Fixes:  [... cut]
closed 2019-07-31 by carlescufi