zephyrproject-rtos/zephyr #17738: STATIC_ASSERT no longer defined when CONFIG_NEWLIB_LIBC is enabled

jlienau03, 2019-07-23:
**Describe the bug**
When you include CONFIG_NEWLIB_LIBC in your proj.conf file, the static_assert macro stops working.   It seems like the static_assert function isn't supported in newlib. When I try building the application it used assert.h from the zephyr library  <..>\ncs\zephyr\lib\libc\minimal\include\assert.h  

When I build with newlib it pulls assert.h from my toolchain directoy instead <...>gnuarmemb\arm-none-eabi\include\assert.h  

**To Reproduce**
Easiest way to reproduce the error is to use the event manager example from Nordic's nRF repository here: https://github.com/NordicPlayground/fw-nrfconnect-nrf/tree/master/samples/event_manager

The example will build with no issues.
Now add CONFIG_NEWLIG_LIBC=y to the proj.cof file and rebuild.
The build will fail stating that static_assert is undefined.

**Impact**
Unable to use the Event Manager with Zephyr.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain GNU Arm ebedded - q2 2018

I'm using the nRF9160 development kit from Nordic along with the Nordic edition of SEGGER Embedded Studio V4.18.

aescolar, 2019-07-24:
@andrewboie can you help here?

pabigot, 2019-08-19:
This is a duplicate of #11754.  See also [this recent fw-nrfconnect-nrf commit](https://github.com/NordicPlayground/fw-nrfconnect-nrf/pull/1003/commits/a8bbb58640ff9a4351c896c1657decb5f47e8731).

That `static_assert` should never have been added to `<assert.h>` is a different issue which I'll follow up on.

closed 2019-08-19 by pabigot