zephyrproject-rtos/zephyr #17743: cross compiling for RISCV32 fails as compiler flags are not supplied by board but must be in target.cmake

egcd32, 2019-07-24:
**Describe the bug**
When cross-compiling the Zephyr hello world sample on Windows 10 using Freedom Studio 4.7.2 2019-03, which includes riscv64-unknown-elf-gcc-8.2.0-2019.02.0, for a SiFive HiFive1 rev B board, ninja reports compiler warnings on pointer/int conversion and the link step fails due to ABI incompatibility.  This is because the compiler flags needed for RISCV32 are not being supplied to gcc. Freedom Studio's gcc can compile for either 64 or 32 but defaults to 64  I had thought that would be set for the board, but the board file was being loaded too late and was not taking effect. 

I was able to work around this issue by changing the zephyr/cmake/compiler/gcc/target.cmake file and adding at about line 80, after the other two architecture specific compiler flag sections:

```
elseif("${ARCH}" STREQUAL "riscv32")

list(APPEND TOOLCHAIN_C_FLAGS
   -march=rv32imac -mabi=ilp32 -mcmodel=medlow
   )
```

**To Reproduce**
Steps to reproduce the behavior:
1. Install a compiler for RISCV32 and RISCV64.
2. Point to that compiler with set ZEPHYR_TOOLCHAIN_VARIANT=cross-compile
set CROSS_COMPILE=C:/.../FreedomStudio-4.7.2.2019-03-4/SiFive/toolchain/riscv64-unknown-elf-gcc-8.2.0-2019.02.0/bin/riscv64-unknown-elf-
3. cd samples  cd hello_world   mkdir build   cd build
4. cmake -GNinja -DBOARD=hifive1_revb .. 
5. ninja -j 1
6. See error   ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'

**Expected behavior**
Compiles to an ELF file for RISCV32.

**Impact**
Would prefer if those compiler flags were in target.cmake in Zephyr for RISCV32, similar to ARM architecture.

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain: GCC riscv64-unknown-elf-gcc-8.2.0 packaged by SiFive in FreedomStudio-4.7.2.2019-03-4 for the HiFive1 rev B board

ioannisg, 2019-07-30:
@SebastianBoe can you help here?

SebastianBoe, 2019-07-30:
Hi,

a patch to this effect submitted as a PR would be welcome.

Although perhaps it should be applied in zephyr/arch/riscv32/CMakeLists.txt to be consistent with how the -mabi and -march is set in ARM.

galak, 2019-08-10:
Is this still an issue with the recent qemu_riscv64 merge we now set these flags:

https://github.com/zephyrproject-rtos/zephyr/blob/master/cmake/compiler/gcc/target.cmake#L72

egcd32, 2019-08-10:
Thanks. There is still an issue, though it appears to be specific to cross compiling with the GCC 8.2.0 packaged in SiFive FreedomStudio-4.7.2.2019-05.0-win32.   The above change to target.cmake uses -mabi=ilp32 -march=rv32ima however SiFive doesn't have a GCC library for rv32ima. (It does have rv32i, rv32im, rv32imac, rv32imafc...)
As a result, in that cross-compiler,
gcc -mabi=ilp32 -march=rv32ima  --print-gcc-file-name   
returns an elf64-littleriscv libgcc.a that isn't correct, while
gcc -mabi=ilp32 -march=rv32imac --print-gcc-file-name   
returns an elf32-littleriscv libgcc.a

jaronkelleher, 2020-03-04:
Does Pull Request #23265 solve this?

kbrunham-intel, 2022-01-22:
I still see this issue when using the SiFive GNU Embedded Toolchain v2020.12.8 and release v2.7.1. Please reopen this issue.

closed 2022-03-15 by mbolivar-nordic