zephyrproject-rtos/zephyr #17745: stm32: Move clock configuration to device tree

erwango, 2019-07-24:
**Is your enhancement proposal related to a problem? Please describe.**
Current clock-control configuration (ie system clock source configuration) for stm32 based devices currently happens via Kconfig. To comply with overall device configuration, and follow other zephyr clock evolution (#15363, #16958) this should be moved to device tree.

Supersets #17703

**Describe the solution you'd like**
Bindings used for clock configuration should be similar to usual Linux clocks bindings.
Only difference would be new bindings introduced for pll parameters (various divisors, pre-divisors and multipliers) following current Kconfig scheme. This deviates from usual Linux binding which usually provides the output pll freq and use an algo at run time to calculate the various parameters allowing to set the required freq. At a later step, we might study the run time calculation option but for now, considering the range of clock-control IPs to study, keeping the current configuration scheme should speed up the implementation.
Same choice would be done for bus prescalers.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

b0661, 2019-07-24:
> Only difference would be new bindings introduced for pll parameters (various divisors, pre-divisors and multipliers) following current Kconfig scheme

What is the benefit to move application specific configurations (various divisors, pre-divisors and multipliers) to DTS?

DTS is a hardware description - with some exceptions. Fixed clocks (SoC internal & external on board) are hardware. Clock (tree) selection and clock divisor configuration is in my view related to the application and shall therefor be defined by Kconfig. An application developer may select a different clock tree based on the application needs using the same hardware/ board.

erwango, 2019-07-25:
@b0661 

Since last year discussions, community agreed that DTS would be used for hardware description AND its boot-time configuration. This is reflected in zephyr documentation in following section: [Device Tree vs Kconfig](https://docs.zephyrproject.org/latest/guides/dts/index.html?#device-tree-vs-kconfig). In that view, clocks configuration is part of hardware configuration and moving it to DTS fits that split.
This is also matching the DTS practice done in other systems using it such as Linux kenel.

b0661, 2019-07-25:
@erwango 

> Since last year discussions, community agreed that DTS would be used for hardware description AND its boot-time configuration.

Thanks for the info.

>  In that view, clocks configuration is part of hardware configuration and moving it to DTS fits that split.

As an application developer will I still be able to change the clock configuration _WITHOUT_ changing the DTS?

erwango, 2019-07-26:
@b0661 

> As an application developer will I still be able to change the clock configuration WITHOUT changing the DTS?

This won't be possible indeed. And tbh, I think it's for the best since having mixed dts/Kconfig would be source of bugs, confusion and finally misuse.
Though, one option I see is to support run time clock configuration, which is not available for now.

gon1332, 2019-07-26:
Just to put more info on runtime clock configuration there this relative issue: https://github.com/zephyrproject-rtos/zephyr/issues/12134

PR #34120 (2021-04-27): STM32: Move RCC and system clocks configuration to device tree.
Provide initial and minimum changes to introduce device tree based clocks configuration
for STM32 based targets.

This initial PR extends to L4, WB and F4 series to ensure minimum scalability of the
proposed solution. Work on other series will be started once this RFC is approved.

Several requirements were taken into account:
- The solution should coexist with current Kconfig based configuration during the transition period
- Once transition is started, we should be able to generate deprecation warnings for Kconfig configuration  [... cut]
closed 2021-04-27 by carlescufi