zephyrproject-rtos/zephyr #17748: stm32: clock-control: Remove usage of SystemCoreClock

erwango, 2019-07-24:
**Is your enhancement proposal related to a problem? Please describe.**
STM32 clock control driver is using variable `SystemCoreClock` to hold core clock frequency.
It was used in order to avoid usage of constant symbol `CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC`
Reason was cpu freq setting is usually done in 2 step with a default value at boot and the configured value set up at clock_control driver init.

Requires:
- removing direct usage of `CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC` (#16958)
- move clock configuration from device tree (#17745)
- get `sys_clock_hw_cycles_per_sec` to return updated clock through implementation of CONFIG_TIMER_READS_ITS_FREQUENCY_AT_RUNTIME for STM32 (or similar)

ikerperezdelpalomar, 2020-07-29:
Hi, I am willing to get learn about Zephyr so I considered that this might be a nice issue that start working from. Is it ok if I try to solve it?

galak, 2020-07-29:
> Hi, I am willing to get learn about Zephyr so I considered that this might be a nice issue that start working from. Is it ok if I try to solve it?

@ikerperezdelpalomar are you familiar with devicetree?

ikerperezdelpalomar, 2020-07-31:
> > Hi, I am willing to get learn about Zephyr so I considered that this might be a nice issue that start working from. Is it ok if I try to solve it?
> 
> @ikerperezdelpalomar are you familiar with devicetree?

A bit, I worked in project that involved devicetree 1 year ago.

erwango, 2021-06-04:
Closing as I'm not certain this is the way to go anymore.
At least, this is not a goal by itself anymore

closed 2021-06-04 by erwango