zephyrproject-rtos/zephyr #17751: build is broken for mec15xxevb_assy6853

andrewboie, 2019-07-24:
```
$ sanitycheck -p mec15xxevb_assy6853
JOBS: 6
Building testcase defconfigs...
Filtering test cases...
143 tests selected, 123437 tests discarded due to filters
total complete:    5/ 141   3%  failed:    0

mec15xxevb_assy6853       samples/nfc/nfc_hello/sample.nfc.hello             FAILED: build_error
	see: sanity-out/mec15xxevb_assy6853/samples/nfc/nfc_hello/sample.nfc.hello/build.log

total complete:   25/ 141  17%  failed:    1

mec15xxevb_assy6853       samples/subsys/console/echo/sample.subsys.console.echo FAILED: build_error
	see: sanity-out/mec15xxevb_assy6853/samples/subsys/console/echo/sample.subsys.console.echo/build.log

total complete:   29/ 141  20%  failed:    2

mec15xxevb_assy6853       samples/subsys/console/getchar/sample.subsys.console.getchar FAILED: build_error
	see: sanity-out/mec15xxevb_assy6853/samples/subsys/console/getchar/sample.subsys.console.getchar/build.log

total complete:   30/ 141  21%  failed:    3

mec15xxevb_assy6853       samples/subsys/console/getline/sample.subsys.console.getline FAILED: build_error
	see: sanity-out/mec15xxevb_assy6853/samples/subsys/console/getline/sample.subsys.console.getline/build.log

total complete:   52/ 141  36%  failed:    4

mec15xxevb_assy6853       tests/drivers/uart/uart_basic_api/peripheral.uart  FAILED: build_error
	see: sanity-out/mec15xxevb_assy6853/tests/drivers/uart/uart_basic_api/peripheral.uart/build.log

total complete:   53/ 141  37%  failed:    5

mec15xxevb_assy6853       tests/drivers/uart/uart_basic_api/peripheral.uart.shell FAILED: build_error
	see: sanity-out/mec15xxevb_assy6853/tests/drivers/uart/uart_basic_api/peripheral.uart.shell/build.log

total complete:  141/ 141  100%  failed:    6
135 of 141 tests passed with 0 warnings in 420 seconds
```

andrewboie, 2019-07-24:
In all cases the error is:

```
In file included from /home/apboie/projects/zephyr2/zephyr/include/arch/arm/arch.h:27,
                 from /home/apboie/projects/zephyr2/zephyr/include/arch/cpu.h:17,
                 from /home/apboie/projects/zephyr2/zephyr/include/kernel_includes.h:34,
                 from /home/apboie/projects/zephyr2/zephyr/include/kernel.h:17,
                 from /home/apboie/projects/zephyr2/zephyr/drivers/serial/uart_ns16550.c:24:
/home/apboie/projects/zephyr2/zephyr/sanity-out/mec15xxevb_assy6853/samples/nfc/nfc_hello/sample.nfc.hello/zephyr/include/generated/uart_ns16550_port_2.h: In function 'irq_config_func_2':
/home/apboie/projects/zephyr2/zephyr/sanity-out/mec15xxevb_assy6853/samples/nfc/nfc_hello/sample.nfc.hello/zephyr/include/generated/generated_dts_board_fixups.h:53:41: error: 'DT_INST_2_NS16550_IRQ_0_PRIORITY' undeclared (first use in this function); did you mean 'DT_INST_0_NS16550_IRQ_0_PRIORITY'?
 #define DT_UART_NS16550_PORT_2_IRQ_PRI  DT_INST_2_NS16550_IRQ_0_PRIORITY
                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/apboie/projects/zephyr2/zephyr/include/arch/arm/cortex_m/irq.h:85:28: note: in definition of macro 'Z_ARCH_IRQ_CONNECT'
  z_irq_priority_set(irq_p, priority_p, flags_p); \
                            ^~~~~~~~~~
/home/apboie/projects/zephyr2/zephyr/sanity-out/mec15xxevb_assy6853/samples/nfc/nfc_hello/sample.nfc.hello/zephyr/include/generated/uart_ns16550_port_2.h:94:2: note: in expansion of macro 'IRQ_CONNECT'
  IRQ_CONNECT(DT_UART_NS16550_PORT_2_IRQ,
  ^~~~~~~~~~~
/home/apboie/projects/zephyr2/zephyr/sanity-out/mec15xxevb_assy6853/samples/nfc/nfc_hello/sample.nfc.hello/zephyr/include/generated/uart_ns16550_port_2.h:95:7: note: in expansion of macro 'DT_UART_NS16550_PORT_2_IRQ_PRI'
       DT_UART_NS16550_PORT_2_IRQ_PRI,
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/apboie/projects/zephyr2/zephyr/sanity-out/mec15xxevb_assy6853/samples/nfc/nfc_hello/sample.nfc.hello/zephyr/include/generated/generated_dts_board_fixups.h:53:41: note: each undeclared identifier is reported only once for each function it appears in
 #define DT_UART_NS16550_PORT_2_IRQ_PRI  DT_INST_2_NS16550_IRQ_0_PRIORITY
                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/apboie/projects/zephyr2/zephyr/include/arch/arm/cortex_m/irq.h:85:28: note: in definition of macro 'Z_ARCH_IRQ_CONNECT'
  z_irq_priority_set(irq_p, priority_p, flags_p); \
                            ^~~~~~~~~~
/home/apboie/projects/zephyr2/zephyr/sanity-out/mec15xxevb_assy6853/samples/nfc/nfc_hello/sample.nfc.hello/zephyr/include/generated/uart_ns16550_port_2.h:94:2: note: in expansion of macro 'IRQ_CONNECT'
  IRQ_CONNECT(DT_UART_NS16550_PORT_2_IRQ,
  ^~~~~~~~~~~
/home/apboie/projects/zephyr2/zephyr/sanity-out/mec15xxevb_assy6853/samples/nfc/nfc_hello/sample.nfc.hello/zephyr/include/generated/uart_ns16550_port_2.h:95:7: note: in expansion of macro 'DT_UART_NS16550_PORT_2_IRQ_PRI'
       DT_UART_NS16550_PORT_2_IRQ_PRI,
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/build.make:62: recipe for target 'zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/uart_ns16550.c.obj' failed

```

galak, 2019-07-24:
Fix is:

```
diff --git a/soc/arm/microchip_mec/mec1501/dts_fixup.h b/soc/arm/microchip_mec/mec1501/dts_fixup.h
index 6a01ed5b7a..48c72e7c2c 100644
--- a/soc/arm/microchip_mec/mec1501/dts_fixup.h
+++ b/soc/arm/microchip_mec/mec1501/dts_fixup.h
@@ -28,7 +28,7 @@
 #define DT_UART_NS16550_PORT_0_BAUD_RATE \
        DT_NS16550_400F2400_CURRENT_SPEED
 #define DT_UART_NS16550_PORT_0_IRQ             DT_NS16550_400F2400_IRQ_0
-#define DT_UART_NS16550_PORT_0_IRQ_PRI         DT_INST_0_NS16550_IRQ_0_PRIORITY
+#define DT_UART_NS16550_PORT_0_IRQ_PRI         DT_NS16550_400F2400_IRQ_0_PRIORITY
 #define DT_UART_NS16550_PORT_0_IRQ_FLAGS       0 /* Deault */
 
 #define DT_UART_NS16550_PORT_1_BASE_ADDR \
@@ -38,7 +38,7 @@
 #define DT_UART_NS16550_PORT_1_BAUD_RATE \
        DT_NS16550_400F2800_CURRENT_SPEED
 #define DT_UART_NS16550_PORT_1_IRQ             DT_NS16550_400F2800_IRQ_0
-#define DT_UART_NS16550_PORT_1_IRQ_PRI         DT_INST_1_NS16550_IRQ_0_PRIORITY
+#define DT_UART_NS16550_PORT_1_IRQ_PRI         DT_NS16550_400F2800_IRQ_0_PRIORITY
 #define DT_UART_NS16550_PORT_1_IRQ_FLAGS       0 /* Default */
 
 #define DT_UART_NS16550_PORT_2_BASE_ADDR \
@@ -48,7 +48,7 @@
 #define DT_UART_NS16550_PORT_2_BAUD_RATE \
        DT_NS16550_400F2C00_CURRENT_SPEED
 #define DT_UART_NS16550_PORT_2_IRQ             DT_NS16550_400F2C00_IRQ_0
-#define DT_UART_NS16550_PORT_2_IRQ_PRI         DT_INST_2_NS16550_IRQ_0_PRIORITY
+#define DT_UART_NS16550_PORT_2_IRQ_PRI         DT_NS16550_400F2C00_IRQ_0_PRIORITY
 #define DT_UART_NS16550_PORT_2_IRQ_FLAGS       0 /* Default */
 
```

andrewboie, 2019-07-24:
@galak thanks!!

closed 2019-07-24 by andrewboie

galak, 2019-07-24:
Similar fix needed for `soc/arm/microchip_mec/mec1701/dts_fixup.h`

PR #17754 (2019-07-24): mec1501: fix dts build error
This file wasn't migrated to the new DTS define for the
IRQ priority.

Fixes: #17751