zephyrproject-rtos/zephyr #17764: Broken link to latest development version of docs

ulfalizer, 2019-07-25:
The link to the latest development version at the top of https://docs.zephyrproject.org/1.14.0/guides/dts/index.html should point to https://docs.zephyrproject.org/latest/guides/dts/index.html, but points to https://docs.zephyrproject.org/guides/dts/index.html, which is a 404.

Might be more broken links. Only stumbled upon that page.

carlescufi, 2019-07-25:
[image]
To clarify, the broken link highlighted

galak, 2019-08-22:
@dbkinder any update on this?

dbkinder, 2019-08-22:
Sorry, I missed this being assigned to me. The header on documents generated for a tagged release (not from the master branch) have this "The latest document version of this page ..." notice with a link to the same page generated from the master branch.  When this header was put in, the master branch documents were in the root folder and all the tagged release versions were in corresponding versioned folder (e.g., /1.14.0).   Then time moved on and the master-branch documents were all moved into a /latest/ folder (instead of the server root) and redirects were put on the server to help reduce the large number of 404 errors that would have occurred.  It seems like those redirects may be missing now. (@nashif ?) The simplest "fix" is to simply remove this header or at least remove the link trying to get to content that's no longer where it's pointing.  (And we'd need to regenerate the 1.14 docs.)

PR #18597 (2019-08-23): doc: update link from released doc to master
Documents generated for a tagged release have a header that mentions
that the "latest" (master branch) version of the doc may be more up to
date, but the link to the corresponding master doc is missing the
/latest/ (previously the master docs were in the root folder.

Fixes: #17764
commit fbcbc2ed (2019-08-23): doc: update link from released doc to master

Documents generated for a tagged release have a header that mentions
that the "latest" (master branch) version of the doc may be more up to
date, but the link to the corresponding master doc is missing the
/latest/ (previously the master docs were in the root folder.

Fixes: #17764
commit 2c8ef376 (2019-08-23): doc: update link from released doc to master

Documents generated for a tagged release have a header that mentions
that the "latest" (master branch) version of the doc may be more up to
date, but the link to the corresponding master doc is missing the
/latest/ (previously the master docs were in the root folder.

Fixes: #17764
commit 781258f0 (2019-08-23): doc: update link from released doc to master

Documents generated for a tagged release have a header that mentions
that the "latest" (master branch) version of the doc may be more up to
date, but the link to the corresponding master doc is missing the
/latest/ (previously the master docs were in the root folder.

Fixes: #17764
closed 2019-08-23 by nashif