zephyrproject-rtos/zephyr #17778: Microchip XEC rtos Timer breaks gpios and k_sleep?

tbursztyka, 2019-07-25:
**Describe the bug**
Before commit bdaab8cfa0071127bbe28fcce33a0037fb661077 code below was working:
```#include <zephyr.h>
#include <sys/printk.h>
#include <device.h>
#include <drivers/gpio.h>

#include <soc.h>

#define GPIO_OUT_DRV_NAME DT_GPIO_XEC_GPIO140_176_LABEL
#define GPIO_OUT_PIN MCHP_GPIO_157

void main(void)
{
	struct device *gpio_out_dev;
	int ret;

	gpio_out_dev = device_get_binding(GPIO_OUT_DRV_NAME);
	if (!gpio_out_dev) {
		printk("Cannot find %s!\n", GPIO_OUT_DRV_NAME);
		return;
	}

	/* Setup GPIO output */
	ret = gpio_pin_configure(gpio_out_dev, GPIO_OUT_PIN, (GPIO_DIR_OUT));
	if (ret) {
		printk("Error configuring pin %d!\n", GPIO_OUT_PIN);
	}

	while (1) {
		gpio_pin_write(gpio_out_dev, GPIO_OUT_PIN, 0);
		k_sleep(300);
		gpio_pin_write(gpio_out_dev, GPIO_OUT_PIN, 1);
		k_sleep(300);
	}
}
```

led3 on MEC evaluation board is then flashing.

After that commit: it does not work anymore.
Reverting that commit makes things work again.

**To Reproduce**
Steps to reproduce the behavior:
1. get the code to compile against board mec15xxevb_assy6853 (replace the hello_world sample for instance)
2. flash (with flashrom or dediprog's software)

**Expected behavior**
led 3 on the evaluation board should blink

**Impact**
gpio and/or k_sleep are affected by this commit? 

I really need this board to work (still having issues with UART2, partially solved via #17763) and this kind of commit is going against that. I reverted on my local branch obviously, and it worked fine again.

tbursztyka, 2019-07-25:
@scottwcpg have a look please.

scottwcpg, 2019-07-25:
I have fixed this issue in PR 17776.
Root cause: k_busy_wait takes a parameter in units of microseconds and its default implementation calls the timer driver k_cycle_get_32. This will not work for timer drivers using a 32KHz time base. This fix in PR 1776 enables CONFIG_ARCH_HAS_CUSTOM_BUSY_WAIT and implements z_arch_busy_wait in the MEC1501 SoC layer. SoC layer z_arch_busy_wait uses a MEC1501 32-bit peripheral counter configured for 16MHz, count-up, auto-reload, and no interrupt. 16MHz was chosen as it allow a shift of 4 to convert microseconds to 16MHz time units instead of the more expensive multiply.

albertofloyd, 2019-07-30:
@scottwcpg  This PR only resolves https://github.com/zephyrproject-rtos/zephyr/issues/17897 
Lets address this k_sleep issue inaccuracy issue in in a different PR.

albertofloyd, 2019-08-14:
@tbursztyka can we close this?

tbursztyka, 2019-08-15:
yes your patch fixed the issue

closed 2019-08-15 by tbursztyka