zephyrproject-rtos/zephyr #17782: network tests failing with 'unexpected eof' code coverage enabled in qemu_x86 (TSS)

camicarballo, 2019-07-25:
``` 
$ ./scripts/sanitycheck --coverage -p qemu_x86 -T tests/net/
JOBS: 4
Cleaning output directory /home/camelot/zephyrproject/zephyr/sanity-out
Building testcase defconfigs...
Filtering test cases...
51 tests selected, 21150 tests discarded due to filters
total complete:    9/  51  17%  failed:    0

qemu_x86                  tests/net/iface/net.iface.no_userspace_allowed     FAILED: unexpected eof
	see: sanity-out/qemu_x86/tests/net/iface/net.iface.no_userspace_allowed/handler.log

total complete:   10/  51  19%  failed:    1

qemu_x86                  tests/net/iface/net.iface.with_userspace           FAILED: unexpected eof
	see: sanity-out/qemu_x86/tests/net/iface/net.iface.with_userspace/handler.log

total complete:   19/  51  37%  failed:    2

qemu_x86                  tests/net/lib/mqtt_packet/net.mqtt.packet          FAILED: handler_crash
	see: sanity-out/qemu_x86/tests/net/lib/mqtt_packet/net.mqtt.packet/handler.log

total complete:   34/  51  66%  failed:    3

qemu_x86                  tests/net/socket/getaddrinfo/net.socket            FAILED: handler_crash
	see: sanity-out/qemu_x86/tests/net/socket/getaddrinfo/net.socket/handler.log

total complete:   35/  51  68%  failed:    4

qemu_x86                  tests/net/socket/getnameinfo/net.socket            FAILED: unexpected eof
	see: sanity-out/qemu_x86/tests/net/socket/getnameinfo/net.socket/handler.log

total complete:   36/  51  70%  failed:    5

qemu_x86                  tests/net/socket/misc/net.socket                   FAILED: handler_crash
	see: sanity-out/qemu_x86/tests/net/socket/misc/net.socket/handler.log

total complete:   37/  51  72%  failed:    6

qemu_x86                  tests/net/socket/net_mgmt/net.socket.mgmt          FAILED: unexpected eof
	see: sanity-out/qemu_x86/tests/net/socket/net_mgmt/net.socket.mgmt/handler.log

total complete:   38/  51  74%  failed:    7

qemu_x86                  tests/net/ptp/clock/net.ptp.clock                  FAILED: timeout
	see: sanity-out/qemu_x86/tests/net/ptp/clock/net.ptp.clock/handler.log

total complete:   39/  51  76%  failed:    8

qemu_x86                  tests/net/socket/register/net.socket.register.tls  FAILED: unexpected eof
	see: sanity-out/qemu_x86/tests/net/socket/register/net.socket.register.tls/handler.log

total complete:   40/  51  78%  failed:    9

qemu_x86                  tests/net/socket/select/net.socket                 FAILED: handler_crash
	see: sanity-out/qemu_x86/tests/net/socket/select/net.socket/handler.log

total complete:   41/  51  80%  failed:   10

qemu_x86                  tests/net/socket/tcp/net.socket.tcp                FAILED: unexpected eof
	see: sanity-out/qemu_x86/tests/net/socket/tcp/net.socket.tcp/handler.log

total complete:   42/  51  82%  failed:   11

qemu_x86                  tests/net/socket/udp/net.socket.udp                FAILED: unexpected eof
	see: sanity-out/qemu_x86/tests/net/socket/udp/net.socket.udp/handler.log

total complete:   47/  51  92%  failed:   12

qemu_x86                  tests/net/utils/net.util                           FAILED: unexpected eof
	see: sanity-out/qemu_x86/tests/net/utils/net.util/handler.log

total complete:   51/  51  100%  failed:   13

```
Enabling CONFIG_COVERAGE exposes test failures with networking tests under tests/net/. We need these be able to build and run successfully to assess code coverage of the network stack.

Upon further inspection, 5 of these are failing on line 395 in arch/x86/core/ia32/crt0.S
- tests/net/iface 
- tests/net/socket/tcp
- tests/net/utils
- tests/net/socket/udp
- tests/net/socket/net_mgmt

Reproduce with:

``` sanitycheck --coverage -p qemu_x86 -T tests/net/ ```

andrewboie, 2019-08-02:
_main_tss doesn't have the expected data in it, everything is zeroed. Expecting to see the .ss0 and .esp0 fields set.

```
(gdb) b crt0.S:278
Breakpoint 1 at 0x108c: file /home/apboie/projects/zephyr2/zephyr/arch/x86/core/ia32/crt0.S, line 278.
(gdb) p _main_tss
$1 = {backlink = 0, reserved_1 = 0, esp0 = 0, ss0 = 0, reserved_2 = 0, esp1 = 0, ss1 = 0, 
  reserved_3 = 0, esp2 = 0, ss2 = 0, reserved_4 = 0, cr3 = 0, eip = 0, eflags = 0, eax = 0, ecx = 0, 
  edx = 0, ebx = 0, esp = 0, ebp = 0, esi = 0, edi = 0, es = 0, reserved_5 = 0, cs = 0, reserved_6 = 0, 
  ss = 0, reserved_7 = 0, ds = 0, reserved_8 = 0, fs = 0, reserved_9 = 0, gs = 0, reserved_10 = 0, 
  ldt_ss = 0, reserved_11 = 0, t = 0 '\000', reserved_12 = 0, iomap = 0}
(gdb) 

```

andrewboie, 2019-08-02:
_main_tss is supposed to look like this:

```
$1 = {backlink = 0, reserved_1 = 0, esp0 = 4378624, ss0 = 16, reserved_2 = 0, esp1 = 0, ss1 = 0, 
  reserved_3 = 0, esp2 = 0, ss2 = 0, reserved_4 = 0, cr3 = 0, eip = 0, eflags = 0, eax = 0, ecx = 0, 
  edx = 0, ebx = 0, esp = 0, ebp = 0, esi = 0, edi = 0, es = 0, reserved_5 = 0, cs = 0, reserved_6 = 0, 
  ss = 0, reserved_7 = 0, ds = 0, reserved_8 = 0, fs = 0, reserved_9 = 0, gs = 0, reserved_10 = 0, 
  ldt_ss = 0, reserved_11 = 0, t = 0 '\000', reserved_12 = 0, iomap = 0}
```

andrewboie, 2019-08-02:
Past a certain point, all memory is filled with 0xFF. What is special about address 0x460670 that it starts happening there?

```
0x4604f0 <__gcov_.sys_dlist_init+4>:	0x202947c4	0x52501ce1	0xa43083b8	0x00000001
0x460500 <__gcov_.sys_dlist_init+20>:	0x0042c1e8	0x00452c80	0x3428e926	0x7c45def6
0x460510 <__gcov_.z_hdlr_k_futex_wait+12>:	0xb1f015fa	0x00000006	0x00425920	0x00452c0
0x460520 <__gcov_.z_impl_k_futex_wait+4>:	0x7d36ff34	0xde6f9b43	0xcb8e8281	0x0000009
0x460530 <__gcov_.z_impl_k_futex_wait+20>:	0x00425960	0x00452c80	0x75d1eae7	0x15ac8ec
0x460540 <__gcov_.z_hdlr_k_futex_wake+12>:	0xb1f015fa	0x00000006	0x004259c0	0x00452c0
0x460550 <__gcov_.z_impl_k_futex_wake+4>:	0x3ccffcf5	0xe9dd15ab	0x759f93ff	0x000000b
0x460560 <__gcov_.z_impl_k_futex_wake+20>:	0x00425a00	0x00452c80	0x3135f19c	0x7229ef0
0x460570 <__gcov_.k_futex_find_data+12>:	0x9946e09f	0x00000004	0x00425a60	0x00452c0
0x460580 <__gcov_.z_ready_thread+4>:	0x20b1e5b7	0x72430cd2	0xed537659	0x00000004
0x460590 <__gcov_.z_ready_thread+20>:	0x00425a80	0x00452c80	0x008363fa	0x489440bb
0x4605a0 <__gcov_.z_is_thread_ready+12>:	0x6e7f1575	0x00000005	0x00425aa0	0x00452c0
0x4605b0 <__gcov_.z_is_thread_timeout_active+4>:	0x43a53e76	0x96f6f490	0x7cb34af9	2
0x4605c0 <__gcov_.z_is_thread_timeout_active+20>:	0x0042c1f0	0x00452c80	0x58a4a6da	2
0x4605d0 <__gcov_.z_is_thread_prevented_from_running+12>:	0xdb5de9e8	0x00000001	0x0042c20
0x4605e0 <__gcov_.z_is_inactive_timeout+4>:	0x407ea241	0x9df0cda6	0x7cb34af9	0x0000002
0x4605f0 <__gcov_.z_is_inactive_timeout+20>:	0x0042c208	0x00452c80	0x15e00cc6	0x0934c82
0x460600 <__gcov_.z_set_thread_return_value+12>:	0xa43083b8	0x00000001	0x0042c218	0
0x460610 <__gcov_.k_spin_unlock+4>:	0x35b25dc2	0x079073d1	0xd4f15868	0x00000007
0x460620 <__gcov_.k_spin_unlock+20>:	0x00425ae0	0x00452c80	0x5896acf9	0x5f4dda15
0x460630 <__gcov_.k_spin_lock+12>:	0xf4905a4e	0x00000008	0x00425b20	0x00452c80
0x460640 <__gcov_.z_arch_irq_unlock+4>:	0x7c9f1c72	0x8c092b7c	0x6e6d39f5	0x00000003
0x460650 <__gcov_.z_arch_irq_unlock+20>:	0x0042c220	0x00452c80	0x1a931129	0xebb9d1d
0x460660 <__gcov_.z_arch_irq_lock+12>:	0x7cb34af9	0x00000002	0x0042c238	0x00452c80
0x460670 <__gcov_.z_do_irq_unlock+4>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x460680 <__gcov_.z_do_irq_unlock+20>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x460690 <__gcov_._do_irq_lock+12>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x4606a0 <__gcov_.sys_dnode_is_linked+4>:	0xffffffff	0xffffffff	0xffffffff	0xfffffff
0x4606b0 <__gcov_.sys_dnode_is_linked+20>:	0xffffffff	0xffffffff	0xffffffff	0xffffff-
f
0x4606c0 <__gcov_.atomic_get+12>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x4606d0 <__gcov_.init_mem_domain_module+4>:	0xffffffff	0xffffffff	0xffffffff	0xfffffff
0x4606e0 <__gcov_.init_mem_domain_module+20>:	0xffffffff	0xffffffff	0xffffffff	0xfffffff
0x4606f0 <__gcov_.k_mem_domain_remove_thread+12>:	0xffffffff	0xffffffff	0xffffffff	f
0x460700 <__gcov_.k_mem_domain_add_thread+4>:	0xffffffff	0xffffffff	0xffffffff	0xfffffff
0x460710 <__gcov_.k_mem_domain_add_thread+20>:	0xffffffff	0xffffffff	0xffffffff	0xfffffff
0x460720 <__gcov_.k_mem_domain_remove_partition+12>:	0xffffffff	0xffffffff	0xffffffff	f
0x460730 <__gcov_.k_mem_domain_add_partition+4>:	0xffffffff	0xffffffff	0xffffffff	f
0x460740 <__gcov_.k_mem_domain_add_partition+20>:	0xffffffff	0xffffffff	0xffffffff	f
0x460750 <__gcov_.k_mem_domain_destroy+12>:	0xffffffff	0xffffffff	0xffffffff	0xfffffff
0x460760 <__gcov_.k_mem_domain_init+4>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x460770 <__gcov_.k_mem_domain_init+20>:	0xffffffff	0xffffffff	0xffffffff	0xfffffff
0x460780 <__gcov_.k_spin_unlock+12>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x460790 <__gcov_.k_spin_lock+4>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x4607a0 <__gcov_.k_spin_lock+20>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x4607b0 <__gcov_.z_arch_irq_unlock+12>:	0xffffffff	0xffffffff	0xffffffff	0xfffffff
0x4607c0 <__gcov_.z_arch_irq_lock+4>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x4607d0 <__gcov_.z_arch_irq_lock+20>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x4607e0 <__gcov_.z_do_irq_unlock+12>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x4607f0 <__gcov_._do_irq_lock+4>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x460800 <__gcov_._do_irq_lock+20>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x460810 <__gcov_.sys_dlist_remove+12>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x460820 <__gcov_.sys_dlist_append+4>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x460830 <__gcov_.sys_dlist_append+20>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x460840 <__gcov_.sys_dlist_peek_next+12>:	0xffffffff	0xffffffff	0xffffffff	0xfffffff
0x460850 <__gcov_.sys_dlist_peek_next_no_check+4>:	0xffffffff	0xffffffff	0xffffffff	f
0x460860 <__gcov_.sys_dlist_peek_next_no_check+20>:	0xffffffff	0xffffffff	0xffffffff	f
0x460870 <__gcov_.sys_dlist_peek_head+12>:	0xffffffff	0xffffffff	0xffffffff	0xfffffff
0x460880 <__gcov_.sys_dlist_is_empty+4>:	0xffffffff	0xffffffff	0xffffffff	0xfffffff
0x460890 <__gcov_.sys_dlist_is_empty+20>:	0xffffffff	0xffffffff	0xffffffff	0xfffffff
0x4608a0 <__gcov_.sys_dnode_init+12>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x4608b0 <__gcov_.sys_dlist_init+4>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x4608c0 <__gcov_.sys_dlist_init+20>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x4608d0 <__gcov_.z_hdlr_k_object_alloc+12>:	0xffffffff	0xffffffff	0xffffffff	0xfffffff
[... cut ...]
0x460910 <__gcov_.validate_any_object+4>:	0xffffffff	0xffffffff	0xffffffff	0xfffffff
0x460920 <__gcov_.validate_any_object+20>:	0xffffffff	0xffffffff	0xffffffff	0xfffffff
0x460930 <__gcov_.z_obj_validation_check+12>:	0xffffffff	0xffffffff	0xffffffff	0xfffffff
0x460940:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x460950 <z_libc_partition>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x460960 <obj_rb_tree>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
0x460970 <obj_list+4>:	0xffffffff	0xffffffff	0xffffffff	0xffffffff
```

andrewboie, 2019-08-02:
> What is special about address 0x460670 that it starts happening there?

The physical address where _main_tss is copied from 0xa1990 is within the area for video memory 0x000A0000  - 0x000FFFFF!!

https://wiki.osdev.org/Memory_Map_(x86)

Incredible that we have not run into this before, live and learn..this only happens if the built application text/ROM size is very large, which happens on these network tests

PR #17989 (2019-08-07): x86: memory management adjustments
- The memory map for ia32 SOC was using the first megabyte. This has reserved memory regions in it, relocate the start of memory for ia32 to 1MB.
- Generate page tables at runtime. Delete old buggy MMU table generation script
- qemu_x86 config adjustments
- Enable very early console for qemu_x86_coverage
- X86 now handles all its arch-specific initialization tasks before z_cstart() is called.

Please see individual commit messages for details.

Fixes: #17783
Fixes: #17782 
Fixes: #17784
commit ce3cc4f9 (2019-08-07): x86: ia32: do not use the first megabyte

After witnessing some strange errors with memory not being
what it should be, lifiting everything above 1MB has solved
it. The Zephyr binary was being loaded into memory containing
reserved regions, resulting in data corruption.

We still simulate XIP for testing purposes by setting up the
memory map as follows:

0x000000 - 0x0FFFFF : Non-present
0x100000 - 0x4FFFFF : "Flash" ROM region
0x500000 - 0x8FFFFF : "SRAM" RAM region

For a total of 9 megabytes of physical RAM used.

Fixes problems observed in some large tests when code coverage
is enabled (wh [... cut]
closed 2019-08-07 by andrewboie