zephyrproject-rtos/zephyr #17787: openocd unable to flash hello_world to cc26x2r1_launchxl

szawasky, 2019-07-25:
**Describe the bug**
west flash fails to write hello_world to the LaunchXL CC26X2R1, complaining that the debug regions are unpowered.

**To Reproduce**
follow the instructions here:
https://docs.zephyrproject.org/latest/boards/arm/cc26x2r1_launchxl/doc/index.html
Use a LAUNCHXL-CC26X2R1

**Expected behavior**
west flash completes successfully, and "hello world" is printed on the serial interface

**Impact**
I'm unable to use this CC26x2 device (show stopper)

**Screenshots or console output**-- west flash: using runner openocd

It took a while to figure out that the boards ship with only some of the JTAG jumpers in place.
I had to add jumpers to TDO, TDI, and SWD. Prior to doing that I was getting a similar error, but notably OpenOCD was reporting: "tap-disabled"

I can get a little further with these jumpers installed, but only to the point where it complains "Error: Debug regions are unpowered, an unexpected reset might have happened".

Here is the full output:

Open On-Chip Debugger 0.10.0+dev-g1df07a9a4-dirty (2019-06-13-13:14)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 2500 kHz
srst_only separate srst_gates_jtag srst_open_drain connect_deassert_srst
adapter_nsrst_delay: 100
Info : XDS110: connected
Info : XDS110: firmware version = 2.3.0.14
Info : XDS110: hardware version = 0x0023
Info : XDS110: connected to target via JTAG
Info : XDS110: TCK set to 2500 kHz
Info : clock speed 2500 kHz
Info : JTAG tap: cc26x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc26x2.cpu enabled
Info : cc26x2.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* cc26x2.cpu         cortex_m   little cc26x2.cpu         halted
Info : JTAG tap: cc26x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc26x2.cpu enabled
Error: Debug regions are unpowered, an unexpected reset might have happened
Error: JTAG-DP STICKY ERROR
Error: Could not find MEM-AP to control the core
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: Invalid ACK (0) in DAP response
Error: DP initialisation failed

Info : XDS110: disconnected

**Environment (please complete the following information):**
 - OS: Linux (Ubuntu 18.04.2)
 - Toolchain: Zephyr SDK
git rev-parse HEAD
2463ded4c84cc8e0f1de3d3966c67db0c025e113

**Additional context**
I'm able to use west/ninja to flash an nrf52840_pca10056 without incident.

ioannisg, 2019-07-30:
FYI @MaureenHelm @bwitherspoon

galak, 2019-08-12:
Which SDK version are you using.  There's a fix in SDK 0.10.2 for this platform which should hopefully address the issue you are seeing.

szawasky, 2019-08-13:
I've been using 0.10.0 and 0.10.1. I will see if 0.10.2 helps, thanks!

galak, 2019-08-20:
> I've been using 0.10.0 and 0.10.1. I will see if 0.10.2 helps, thanks!

Any update?

galak, 2019-08-22:
Closing this as its works for myself and @vanti with the newer SDK.  If this is still an issue please reopen.

closed 2019-08-22 by galak

cfriedt-friedtco, 2019-09-04:
I'm receiving the same issue. Built with zephyrproject-rtos/openocd branch `zephyr-20190226`. Also tried branch `zephyr-20190130`. Same problems.

archseer, 2020-03-05:
@galak I also have this issue present on all openocd versions. I think it's XDS110 firmware dependent (it's broken for me on 3.0.0.7)

vanti, 2020-03-05:
@archseer Have you tried the workaround in https://github.com/zephyrproject-rtos/zephyr/issues/21372? It is needed on newer XDS firmware 3.x until OpenOCD is fixed.

IvoBCD, 2020-09-29:
@archseer I'm also seeing this, even using the OpenOCD included in Zephyr SDK 0.11.4.  The fix from  #21372 does not help.

I am able to flash the board using TI's Uniflash, so I'm confident the HW is OK.

```
Open On-Chip Debugger 0.10.0+dev-01341-g580d06d9d-dirty (2020-06-25-12:07)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1500 kHz

Info : XDS110: connected
Info : XDS110: firmware version = 3.0.0.13
Info : XDS110: hardware version = 0x0023
Info : XDS110: connected to target via JTAG
Info : XDS110: TCK set to 2500 kHz
Info : clock speed 1500 kHz
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: cc26x2.jrc: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State
--  ------------------ ---------- ------ ------------------ ------------
 0* cc26x2.cpu         cortex_m   little cc26x2.cpu         tap-disabled

Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: cc26x2.jrc: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Target not examined yet
Error: auto_probe failed
auto erase enabled
Info : XDS110: disconnected
```

vanti, 2020-09-29:
@IvoBCD I am also on Zephyr SDK 0.11.4 and have firmware 3.0.0.13, and 'west flash' seems to work fine for me on Rev B of the cc26x2r1_launchxl:

```
Open On-Chip Debugger 0.10.0+dev-01341-g580d06d9d-dirty (2020-06-25-12:07)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1500 kHz

Info : XDS110: connected
Info : XDS110: firmware version = 3.0.0.13
Info : XDS110: hardware version = 0x0023
Info : XDS110: connected to target via JTAG
Info : XDS110: TCK set to 2500 kHz
Info : clock speed 1500 kHz
Info : JTAG tap: cc26x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc26x2.cpu enabled
Info : cc26x2.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* cc26x2.cpu         cortex_m   little cc26x2.cpu         running

Info : JTAG tap: cc26x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc26x2.cpu enabled
Error: Debug regions are unpowered, an unexpected reset might have happened
Error: JTAG-DP STICKY ERROR
Error: Could not find MEM-AP to control the core
target halted due to debug-request, current mode: Thread 
xPSR: 0x61000000 pc: 0x1000118e msp: 0x11001ff0
auto erase enabled
wrote 360448 bytes from file /home/vwan/zephyrproject-work/zephyr/samples/hello_world/build/zephyr/zephyr.hex in 10.956926s (32.126 KiB/s)

Info : JTAG tap: cc26x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc26x2.cpu enabled
Error: Debug regions are unpowered, an unexpected reset might have happened
Error: JTAG-DP STICKY ERROR
Error: Could not find MEM-AP to control the core
target halted due to debug-request, current mode: Thread 
xPSR: 0x61000000 pc: 0x1000118e msp: 0x11001ff0
verified 360448 bytes in 2.518154s (139.785 KiB/s)

Info : JTAG tap: cc26x2.jrc tap/device found: 0x3bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x3)
Info : JTAG tap: cc26x2.cpu enabled
Error: Debug regions are unpowered, an unexpected reset might have happened
Error: JTAG-DP STICKY ERROR
Error: Could not find MEM-AP to control the core
shutdown command invoked
Info : XDS110: disconnected
```

Given you have the ability to flash using Uniflash, could you use it (or TI Code Composer Studio) to flash the `zephyr.elf` binary from hello world, then retry with OpenOCD? I am wondering if what you had previously flashed on the board is somehow causing this.

IvoBCD, 2020-09-30:
@vanti It turns out that I had to populate the TDO and TDI jumpers on the LaunchPad board, for OpenOCD to work.  TI's Uniflash doesn't need those.