zephyrproject-rtos/zephyr #17794: Timeutil_api test fails with sanitycheck on iotdk board.

chen-png, 2019-07-26:
**Describe the bug**
The timeutil_api test fails with sanitycheck on iotdk board. We are seeing the error message of "gmtime disagreed with gmtime_r".

**To Reproduce**
Steps to reproduce the behavior:
 ./scripts/sanitycheck -p iotdk --device-testing --device-serial /dev/ttyUSB0 --testcase-root tests/lib/timeutil/ --detailed-report ./test_report.xml

**Screenshots or console output**
JOBS: 8
Cleaning output directory /home/chenpeng/zephyrproject/zephyr/sanity-out
Building testcase defconfigs...
Filtering test cases...
1 tests selected, 190 tests discarded due to filters
total complete:    0/   1   0%  failed:    0

iotdk                     tests/lib/timeutil/libraries.libc.minimal          FAILED: failed
        see: sanity-out/iotdk/tests/lib/timeutil/libraries.libc.minimal/handler.log

total complete:    1/   1  100%  failed:    1
0 of 1 tests passed with 0 warnings in 32 seconds

[image]

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr SDk
 - Commit ID: 491702a6c6

IRISZZW, 2019-07-29:
in test_gmtime function[test_gmtime.c], struct tm tm is a uninitialized local variable. and tm.tm_isdst is never used. so the value of tm.tm_isdst is inscrutable.
and struct tm *tp is pointed to a static static struct tm shared which defined in gmtime function[gmtime.c]. although shared.tm_isdst is never used, but it will initialize to 0 by z_bss_zero function.

ioannisg, 2019-07-31:
@chen-png @IRISZZW  I wonder if you could quickly check the provided bug-fix PR, #17889 and see if it solves this issue.

chen-png, 2019-07-31:
Yes, the bug-fix PR #17889 has sloved this issue. Thanks.

PR #17889 (2019-07-31): libc/minimal: fix reproducibility of gmtime
struct tm has fields that were not being set by the implementation,
causing the test to fail when the uninitialized values were compared
with a static initialized result.  Zero the structure before filling it.

Closes #17794
commit b8af1a6a (2019-07-31): libc/minimal: fix reproducibility of gmtime

struct tm has fields that were not being set by the implementation,
causing the test to fail when the uninitialized values were compared
with a static initialized result.  Zero the structure before filling it.

Closes #17794
closed 2019-07-31 by chen-png