zephyrproject-rtos/zephyr #17800: Bluetooth: GATT: Write Without Reponse to invalid handle asserts

paulgavrikov, 2019-07-26:
**Describe the bug**
Issuing a Write Without Response command to an invalid handle (i.e. this handle does not exist). This [line of code](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_conn.c#L725) asserts

**To Reproduce**
Steps to reproduce the behavior:

```
u8_t payload[5];
memset(payload, 1, sizeof(payload));
bt_gatt_write_without_response(pConn, 0x1234 payload,
			sizeof(payload), FALSE);
```

**Expected behavior**
The previous behavior was that the write command returned 0 and no callback etc. was triggered. In case that the remote (G)ATT is notifying of the error, it would be great to trigger the write callback.

**Impact**
Our Zephyr system is behaving according to remote user input, so we cannot assume that the user passes a valid handle (think typo) hence this issue is crucial to us.

**Screenshots or console output**

```
d_00: @00:00:00.672567  [00:00:00.672,546] <err> bt_ctlr_llsw_nordic_lll_conn: assert: 'link' failed
d_00: @00:00:00.672567  @ /home/ntrnd/subversion/components/zephyr/system/Zephyr/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_conn.c:725:
d_00: @00:00:00.672567  [00:00:00.672,546] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops
d_00: @00:00:00.672567  [00:00:00.672,546] <err> os: Current thread: 0x566e0cc0 (idle)
d_00: @00:00:00.672567  [00:00:00.672,546] <err> os: Halting system
d_00: @00:00:00.672567 ERROR: Exiting due to fatal error
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): Kubuntu 18.04
 - Toolchain (e.g Zephyr SDK, ...): Zephyr SDK 0.10.1
 - Commit SHA or Version used:  SHA 

**Additional context**
Compiled for nrf52_bsim

paulgavrikov, 2019-07-26:
@cvinayak I do not have rights to assign you.

carlescufi, 2019-07-26:
Not sure if this is better fixed at the Host or Controller level

Vudentz, 2019-08-02:
All we do in the host is to convert to over the byte order if needed:

 	cmd->handle = sys_cpu_to_le16(handle);

Im surprised this even compile though.

MariuszSkamra, 2019-08-02:
"In case that the remote (G)ATT is notifying of the error, it would be great to trigger the write callback."
There shall be no response (even ATT Error Response to this command).

Core 5.1 Vol 3, Part G, 4.9.1 Write Without Response
"If the Characteristic Value write request is the wrong size, or has an invalid
value as defined by the profile, then the write shall not succeed and no error
shall be generated by the server."

paulgavrikov, 2019-08-02:
> Core 5.1 Vol 3, Part G, 4.9.1 Write Without Response
> "If the Characteristic Value write request is the wrong size, or has an invalid
> value as defined by the profile, then the write shall not succeed and no error
> shall be generated by the server."

It's not wrong size, nor invalid value. It is a nonexistent handle but I am not that familiar with the GATT spec so maybe it is defined somewhere else.

MariuszSkamra, 2019-08-02:
Basically you have two types of ATT transactions:
ATT Request <-> ATT Response
ATT Command
the difference is that there is not response to ATT Command.

joerchan, 2019-08-16:
Cannot reproduce with the bug-description.
Root cause must be something other than the call to GATT Write without Response with Invalid Handle.
The handling of the invalid handle seems to be correct when receiving, and when sending it's not used for anything.

Also the thread seems to be the IDLE thread. 
```
[00:00:00.672,546] <err> os: Current thread: 0x566e0cc0 (idle)
```
We need an updated way to reproduce this issue

paulgavrikov, 2019-08-16:
Cannot reproduce it anymore as well, might have been something else indeed.

closed 2019-08-16 by paulgavrikov