zephyrproject-rtos/zephyr #17805: [Zepyhr v1.14.0 and master] Unable to run commands of mcumgr tool over UART like reset

paanastasiadis, 2019-07-26:
**Bug Description**
I am using a nucleo_l496zg board connected via usb and i am trying to establish communication between mcumgr tool and the board. My src files are based on smp_svr sample and as an example i am trying to send a reset command ( mcumgr reset -c myserial01) where myserial01 profile was created like this (./mcumgr conn add myserial01 type=serial connstring="dev=/dev/ttyACM0,baud=115200"). Whether on v14 or master version, i am getting for output this message "Error: NMP timeout"
Difference between zephyr v14 and master version is that for the latter, if i use smp shell on board(by changing prj.conf accordingly) instead of just smp uart, i manage to run successfully the reset command. On v14, i am getting mpu fault on gtkterm by using  smp shell config (CONFIG_MCUMGR_SMP_SHELL=y, CONFIG_SHELL=y,). With uart only, board is being flashed normally, although still getting the  "error: nmp timeout" when i am using mcumgr tool.

My main.c file is like the smp_svr sample but very simplified. I kept only the code 
regarding CONFIG_MCUMGR_CMD_OS_MGMT ,in order for me to use the os commands and deleted everything regarding bluetooth, stats etc.. Basically, main simply register os mcumgr with os_mgmt_register_group() and contains a while(1) loop with k_sleep(1000) inside.

My prj.conf:

CONFIG_MCUMGR=y

CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048

CONFIG_MCUMGR_SMP_UART=y
CONFIG_MCUMGR_SMP_SHELL=n
CONFIG_SHELL=n

CONFIG_MCUMGR_CMD_OS_MGMT=y

CONFIG_NET_BUF=y
CONFIG_NET_BUF_USER_DATA_SIZE=7

**Reproduce**
Steps to reproduce the behavior:
1. inside smp_svr sample in zephyr, change src/main.c content like the screenshot below  
2. change prj.conf with the above configurations
3. west build -b nucleo_l496zg on a nucleo_l496zg board connected via usb
4. west flash
5../mcumgr conn add myserial01 type=serial connstring="dev=/dev/ttyACM0,baud=115200"
6. mcumgr reset -c myserial01

**Expected behavior**
Using mcumgr command, output should be "Done" and device should be reset itself.

**Impact**
showstopper

**Screenshots or console output**
console output: Error: NMP timeout

main.c
[image]

Error on v14 version when using smp shell configurations (error on V14 ONLY, master is properly working with smp shell)
CONFIG_MCUMGR_SMP_SHELL=y
CONFIG_SHELL=y
[image]

**Environment**
 - OS: Linux Fedora 30 
 - Toolchain  Zephyr SDK
 - Zephyr: v1.14.0-branch or master version

carlescufi, 2019-11-21:
@paanastasiadis is this still an issue?

gon1332, 2019-11-21:
@carlescufi Since then a workaround solution was to use v2.0 version of Zephyr. It is true that it is long time since we last used this functionality. You may close it if there isn't anyone else seeing this.

When we revisit it, we will reopen the issue if it doesn't work under v1.14.

closed 2019-11-21 by carlescufi