zephyrproject-rtos/zephyr #17809: Bluetooth Mesh message cached too early when LPN

richardbarlow, 2019-07-26:
**Describe the bug**

When a device is operating as an LPN and polls its friend there is a small window in which it may receive messages from non-friends in the mesh as well as the response to the poll from its friend. The network layer will add any messages it receives to the network message cache, regardless of what credentials it is encrypted with (network or friend) and regardless of the current friendship state of the LPN. If a message is received that is ultimately destined for the LPN, but was not sent by the friend, the network layer will cache it and pass it onto the transport layer, where it will be rejected:

https://github.com/zephyrproject-rtos/zephyr/blob/bcd60189c2568bcb9e56df0af2d31d7bbd6a438a/subsys/bluetooth/mesh/transport.c#L1405-L1410

If one of the messages received in this fashion is ultimately destined for the LPN, it will get rejected by the network layer when the friend sends it to the LPN in response to a poll:

https://github.com/zephyrproject-rtos/zephyr/blob/bcd60189c2568bcb9e56df0af2d31d7bbd6a438a/subsys/bluetooth/mesh/net.c#L1000-L1003

When this happens the message is not forwarded onto to the transport layer and the state machine for handling the LPN behaviour will attempt to poll the friend a few more times (seemingly getting no response). After a few attempts it clears its friendship.

**To Reproduce**
Steps to reproduce the behavior:

I don't have an MWE, but I would expect to see this occur whenever an LPN is working in a mesh that has a reasonably high rate of traffic. In my case the LPN is sending a segmented message to a unicast address and the destination device responds with a single-segment segmented message almost immediately.

**Expected behavior**
Unfortunately I can't see how this scenario is meant to be handled in the spec. In fact there's not too much said at all about the Network Message Cache.

I would expect that, when an LPN has established friendship it simply ignores all messages received that are not encrypted with the friend credentials. This would prevent messages being added to the cache which are then subsequently ignored (but will need to be accepted later on).

**Screenshots or console output**
Apologies for the text-in-screenshot. It does at least preserve the colouring, which helps to interpret it. Note that this is the combined log from two devices (a friend node 'HUB' and an LPN 'SENSOR').

The key parts are the 'Ignoring unexpected message in Low Power mode' at 53:234,619. The friend receives this message and queues it up in buffer 0x20007ce8. At 53:815,673 the LPN rejects the message from the friend ('Duplicate found in Network Message Cache') that the friend just sent, which if you look two lines up was the one from buffer 0x20007ce8.

[image]
[image]

**Environment (please complete the following information):**
 - OS: Linux (Fedora 29)
 - Toolchain: gcc-arm-none-eabi-6-2017-q2-update
 - Commit SHA or Version used: 3bbcdc79d0f2682bd16af0a61ec632950a4d8f20

richardbarlow, 2019-08-05:
I'm about to implement a workaround for this since we have a client that requires this bug to be fixed in their product. I can see two obvious approaches:

 1. Prevent the message received using non-friend credentials from being added to the message cache in the first place. This will require the network layer to know whether the device is an LPN with an established friendship.
 2. Remove the message from the message cache when the message is dropped by the transport layer as an unexpected message in Low Power mode. This will require the transport layer to notify the network layer to remove the message from the cache and the ability for the network to perform this removal.

I don't like option 1 as it breaks the abstraction between the network and transport layers (the network layer requires visibility into the state of the transport layer). I think implementing option 2 is reasonably trivial. It requires adding a new function to the `net.h` interface to request the removal of a message from the cache. This function simply needs to find the message in the cache and set it to 0 (since 0 is never a valid hash for a message).

Does this sounds reasonable? If you think that it sounds like a valid approach for fixing the bug I can submit a pull request when it's implemented.

jhedberg, 2019-08-05:
@richardbarlow sorry for the delay in responding to this - I've been on summer vacation for the past weeks (officially back next week). Would it make sense to split `msg_cache_match()` into two separate functions: one for looking for a match, and another for updating the cache. The check for a match would occur in the current place, but updating the cache would only happen when the transport layer has successfully accepted the message. A similar fix was recently made to the Replay Protection List handling. Do you see any issue with this kind of approach (e.g. something in the spec or test spec forbidding it)?

jhedberg, 2019-08-05:
@richardbarlow actually, I think my suggestion is not a good idea, since the network message cache is supposed to cover many more messages than just those passed to our transport layer.

jhedberg, 2019-08-05:
@richardbarlow thinking a bit more about this: I'd consider your suggestion 2 with a slight modification: use the EAGAIN return code from `bt_mesh_trans_recv()` as an indication that the transport layer wants to attempt receiving the message again at a later point in time (pretty consistent with what this POSIX error actually means) and then have net.c remove the entry from the cache. That way there's no need to export a new function through net.h.

richardbarlow, 2019-08-05:
@jhedberg I was also on holiday last week (hence the gap between reporting and fixing). Please don't feel any need to respond to any of this until you're back from your vacation.

I've implemented the fix that you suggested of removing the message from the cache when `bt_mesh_trans_recv()` returns `EAGAIN`. It works beautifully. In the real-world application I have here the LPNs have gone from losing friendship almost every time they try to poll their friend for messages to never losing friendship. With the bug I was seeing a long-term average current draw of ~360µA per LPN. With the fix in place this has reduced to ~58µA.

PR #18026 (2019-08-15): Bluetooth: Mesh: LPN: Remove msg from cache on rejection
When in Low Power mode an LPN may receive messages sent by nodes other
than its friend during its brief receive window. These messages get
rejected by the transport layer. At some point in the future the LPN
will receive these messages from its friend, however they will have
already been added to the network message cache earlier.

When the transport layer rejects a message due to it being received from
a non-friend node while in Low Power mode it must be removed from the
network message cache so that it can be correctly received from it [... cut]
commit 5d5706e7 (2019-08-15): Bluetooth: Mesh: LPN: Remove msg from cache on rejection

When in Low Power mode an LPN may receive messages sent by nodes other
than its friend during its brief receive window. These messages get
rejected by the transport layer. At some point in the future the LPN
will receive these messages from its friend, however they will have
already been added to the network message cache earlier.

When the transport layer rejects a message due to it being received from
a non-friend node while in Low Power mode it must be removed from the
network message cache so that it can be correctly received from i [... cut]
closed 2019-08-15 by carlescufi