zephyrproject-rtos/zephyr #17814: Zephyr support for NXP i.MX8M SoC

Mani-Sadhasivam, 2019-07-27:
**Is your feature request related to a problem? Please describe.**

I have an NXP i.MX8M EVK board and I'd like to contribute the SoC and board support to Zephyr for running on the M4 core.

**Describe the solution you'd like**

I got the basic SoC/board port worked on:
https://github.com/Mani-Sadhasivam/zephyr/tree/imx8m

But for submitting it upstream, I'd like to get below points clarified:

1. NXP only supplies MCUXpresso based SDK for i.MX8 series SoCs which differs from the FreeRTOS SDK available for i.MX6/7 family. So, I have imported the SDK to `ext/hal/nxp/mcux/devices/MIMX8MM6`. Is this approach correct?

2. How is the `system_<soc>_cm4` file in SDK gets compiled? I can't get it to build by default without adding a separate rule:
https://github.com/Mani-Sadhasivam/zephyr/blob/imx8m/ext/hal/nxp/mcux/CMakeLists.txt#L63

3. The UART driver available based on MCUX SDK (up to 2.5.0), is not compatible with i.MX8M (SDK version 2.6.0). So, I have created a separate UART driver:
https://github.com/Mani-Sadhasivam/zephyr/blob/imx8m/drivers/serial/uart_mcux_imx.c
Is this approach correct or I have to modify one of the existing MCUX drivers?

Mani-Sadhasivam, 2019-07-27:
@MaureenHelm Could you please take a look?

galak, 2019-07-27:
> 3\. The UART driver available based on MCUX SDK (up to 2.5.0), is not compatible with i.MX8M (SDK version 2.6.0). So, I have created a separate UART driver:
>     [Mani-Sadhasivam/zephyr:drivers/serial/uart_mcux_imx.c@`imx8m`](https://github.com/Mani-Sadhasivam/zephyr/blob/imx8m/drivers/serial/uart_mcux_imx.c) [](/Mani-Sadhasivam/zephyr/blob/HEAD@{2019-07-27T08:57:23Z}/drivers/serial/uart_mcux_imx.c)
>     Is this approach correct or I have to modify one of the existing MCUX drivers?

How different is the SDK uart driver?

Mani-Sadhasivam, 2019-07-27:
> > 3. The UART driver available based on MCUX SDK (up to 2.5.0), is not compatible with i.MX8M (SDK version 2.6.0). So, I have created a separate UART driver:
> > [Mani-Sadhasivam/zephyr:drivers/serial/uart_mcux_imx.c@`imx8m`](https://github.com/Mani-Sadhasivam/zephyr/blob/imx8m/drivers/serial/uart_mcux_imx.c) 
> > Is this approach correct or I have to modify one of the existing MCUX drivers?
> 
> How different is the SDK uart driver?

Some of the API got changed. For instance,

```
UART_GetStatusFlags(config->base)
```
got changed to
```
UART_GetStatusFlag(config->base, kUART_TxReadyFlag)
```

Mani-Sadhasivam, 2019-08-07:
@MaureenHelm Any inputs?

Mani-Sadhasivam, 2019-08-24:
@MaureenHelm ping

raveslave, 2019-11-21:
just checking in, any ideas on when this might be

MaureenHelm, 2019-12-11:
> just checking in, any ideas on when this might be

See #20328

PR #20328 (2020-08-14): Add i.MX8M Mini SoC and board support
Hello,

This PR adds SoC and board support for NXP i.MX8M Mini SoC. NXP has a wide variety of i.MX8 series SoCs with different configurations. So I'm not exactly sure how to model the SoC hierarchy here. But what I have done is to create a separate `mimx8mm_m4` SoC directory for i.MX8M Mini SoCs.

And the board support is added for the i.MX8M Mini LPDDR EVK board. The model name for this board is `8MMINILPD4-EVK` but since we can't use that name directly, I have added it as `mimx8mm_evk`.

The corresponding PR for adding SDK support to NXP module is here:  [... cut]
closed 2020-08-14 by MaureenHelm