zephyrproject-rtos/zephyr #17816: LVGL V5.3 build error if CONFIG_LVGL_COLOR_16_SWAP=y

explora26, 2019-07-27:
**Describe the bug**
I'm trying to development my project with LVGL, but the LCD color is incorrect due to data swap.
So I enable CONFIG_LVGL_COLOR_16_SWAP, but got build error.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/gui/lvgl
2. add configs to boards/nrf52840_pca10056.conf
    CONFIG_LVGL_COLOR_DEPTH_16=y
    CONFIG_LVGL_BITS_PER_PIXEL=0
    CONFIG_LVGL_COLOR_16_SWAP=y
    CONFIG_ILI9340_RGB565
3. west build --board=nrf52840_pca10056
4. See error

**Screenshots or console output**
```
[134/206] Building C object modules/lvgl/CMakeFiles/..__modules__lib__gui__lvgl.dir/lv_objx/lv_canvas.c.obj
FAILED: modules/lvgl/CMakeFiles/..__modules__lib__gui__lvgl.dir/lv_objx/lv_canvas.c.obj 
/home/explora26/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBUILD_VERSION=zephyr-v1.14.0-2663-g3d4409403d14 -DKERNEL -DLV_CONF_INCLUDE_SIMPLE=1 -DNRF52840_XXAA -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I/media/explora26/project/zephyrproject/zephyr/lib/gui/lvgl -I/media/explora26/project/zephyrproject/zephyr/kernel/include -I/media/explora26/project/zephyrproject/zephyr/arch/arm/include -I/media/explora26/project/zephyrproject/zephyr/include -I/media/explora26/project/zephyrproject/zephyr/include/drivers -Izephyr/include/generated -I/media/explora26/project/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf52 -I/media/explora26/project/zephyrproject/zephyr/soc/arm/nordic_nrf/include -I/media/explora26/project/zephyrproject/zephyr/lib/libc/minimal/include -I/media/explora26/project/zephyrproject/zephyr/ext/hal/cmsis/Include -I/media/explora26/project/zephyrproject/zephyr/ext/hal/nordic/nrfx -I/media/explora26/project/zephyrproject/zephyr/ext/hal/nordic/nrfx/drivers/include -I/media/explora26/project/zephyrproject/zephyr/ext/hal/nordic/nrfx/hal -I/media/explora26/project/zephyrproject/zephyr/ext/hal/nordic/nrfx/mdk -I/media/explora26/project/zephyrproject/zephyr/ext/hal/nordic/. -isystem /home/explora26/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include -isystem /home/explora26/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/include-fixed -Os -nostdinc -imacros/media/explora26/project/zephyrproject/zephyr/samples/gui/lvgl/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mthumb -mcpu=cortex-m4 -imacros/media/explora26/project/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/media/explora26/project/zephyrproject/zephyr/samples/gui/lvgl=CMAKE_SOURCE_DIR -fmacro-prefix-map=/media/explora26/project/zephyrproject/zephyr=ZEPHYR_BASE -ffunction-sections -fdata-sections -mabi=aapcs -march=armv7e-m -std=c99 -MD -MT modules/lvgl/CMakeFiles/..__modules__lib__gui__lvgl.dir/lv_objx/lv_canvas.c.obj -MF modules/lvgl/CMakeFiles/..__modules__lib__gui__lvgl.dir/lv_objx/lv_canvas.c.obj.d -o modules/lvgl/CMakeFiles/..__modules__lib__gui__lvgl.dir/lv_objx/lv_canvas.c.obj   -c /media/explora26/project/zephyrproject/modules/lib/gui/lvgl/lv_objx/lv_canvas.c
/media/explora26/project/zephyrproject/modules/lib/gui/lvgl/lv_objx/lv_canvas.c: In function 'lv_canvas_mult_buf':
/media/explora26/project/zephyrproject/modules/lib/gui/lvgl/lv_objx/lv_canvas.c:344:33: error: 'lv_color_t' {aka 'union <anonymous>'} has no member named 'green'; did you mean 'green_h'?
             canvas_buf_color[j].green = (uint16_t) ((uint16_t) canvas_buf_color[j].green * copy_buf_color[j].green) >> 8;
                                 ^~~~~
                                 green_h
/media/explora26/project/zephyrproject/modules/lib/gui/lvgl/lv_objx/lv_canvas.c:344:84: error: 'lv_color_t' {aka 'union <anonymous>'} has no member named 'green'; did you mean 'green_h'?
             canvas_buf_color[j].green = (uint16_t) ((uint16_t) canvas_buf_color[j].green * copy_buf_color[j].green) >> 8;
                                                                                    ^~~~~
                                                                                    green_h
/media/explora26/project/zephyrproject/modules/lib/gui/lvgl/lv_objx/lv_canvas.c:344:110: error: 'lv_color_t' {aka 'union <anonymous>'} has no member named 'green'; did you mean 'green_h'?
             canvas_buf_color[j].green = (uint16_t) ((uint16_t) canvas_buf_color[j].green * copy_buf_color[j].green) >> 8;
                                                                                                              ^~~~~
                                                                                                              green_h
[139/206] Building C object modules/lvgl/CMakeFiles/..__modules__lib__gui__lvgl.dir/lv_objx/lv_chart.c.obj
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain :Zephyr SDK
 - Commit SHA: bcd60189c2568bcb9e56df0af2d31d7bbd6a438a

**Additional context**
LVGL commit SHA: c43a2bbe52cc803d12bc12888b38105cf1e179d0

vanwinkeljan, 2019-07-27:
This is not a zephyr issue but a bug in LVGL V5.3, see https://github.com/littlevgl/lvgl/issues/802

vanwinkeljan, 2019-07-28:
@explora26  Pushed a PR backporting the fix for this issue but keep in mind that the documentation on the LVGL webpages is for V6.0 and not for V5.3 and possibly other fixes went in to the master branch but never made it into a V5 release.

In case you are interested in V6 support with zephyr see PR #17629

explora26, 2019-07-29:
Thanks for the fix, I will verify the patch this week.

vanwinkeljan, 2019-07-29:
Reopened issue as it needs the fix of PR #17824

embeddedt, 2019-07-29:
> possibly other fixes went in to the master branch but never made it into a V5 release.

That's correct. After a release all bugfixes are put in `master`.

PR #17824 (2019-07-29): west: gui: Updated lvgl revision
Updated lvgl revision to take in fix for 16-bit color space swap.

Fixes #17816
commit 9d2669f1 (2019-07-29): west: gui: Updated lvgl revision

Updated lvgl revision to take in fix for 16-bit color space swap.

Fixes #17816
closed 2019-07-29 by ioannisg