zephyrproject-rtos/zephyr #17820: Mesh  bug report In access.c

menglingao1997, 2019-07-27:
hello everyone:
 When I first call bt_mesh_model_publish on access.c and do not use retransmission, the internal call to pub_send_cb.publish_send, the next_period function is executed, and the pub->period_start state is not initialized, which may cause a malfunction.

## Suggest：

* Add conditional judgments in bt_mesh_model_publish, for example:
```c
if(!pub->count)
{
    pub->period_start=k_uptime_get32();
}
```
* remove pub->period_start=k_uptime_get32 in mod_publish func.

aescolar, 2019-07-29:
CC @jhedberg @Vudentz

carlescufi, 2019-08-22:
@jhedberg @trond-snekvik can you comment on this one?

jhedberg, 2019-08-22:
This does look like a potential bug to me. Still looking at the code to confirm it and figure out whether your proposal is the right fix or not.

PR #18577 (2019-08-22): Bluetooth: Mesh: Fix publication period timestamp initialization
The period_start timestamp wasn't getting properly initialized in the
case that retransmission was not being used. In the case of
retransmission the timestamp was getting updated in the mod_publish()
delayed work callback. Add a send_start callback and do the
initialization there, since this covers both the retransmission as
well as the no-retransmission cases.

Fixes #17820
commit 71d00c1b (2019-08-22): Bluetooth: Mesh: Fix publication period timestamp initialization

The period_start timestamp wasn't getting properly initialized in the
case that retransmission was not being used. In the case of
retransmission the timestamp was getting updated in the mod_publish()
delayed work callback. Add a send_start callback and do the
initialization there, since this covers both the retransmission as
well as the no-retransmission cases.

Fixes #17820
PR #18586 (2019-08-29): [Backport v1.14-branch] Bluetooth: Mesh: Fix publication period timestamp initialization
Backport #18577

Fixes #17820
commit 13d04abd (2019-08-29): Bluetooth: Mesh: Fix publication period timestamp initialization

The period_start timestamp wasn't getting properly initialized in the
case that retransmission was not being used. In the case of
retransmission the timestamp was getting updated in the mod_publish()
delayed work callback. Add a send_start callback and do the
initialization there, since this covers both the retransmission as
well as the no-retransmission cases.

Fixes #17820
closed 2019-08-22 by jhedberg