zephyrproject-rtos/zephyr #17821: Mesh Bug on access.c

menglingao1997, 2019-07-27:
# On line 235 in access.c

```c
if(pub->count)
{
    k_delayed_work_cancel(&pub->timer);
}
```

## description

when call bt_mesh_model_publish function in application, and use retransmit, for example,use pub->count not equal zero. so first bt_mesh_model_publish call from app, and mod_publish will be call after Publish Period timer time out, when count minus one and equal zero, the mod_publish func will call bt_mesh_model_publish again, when result is success, so the publish->count with reset, but In the above code, we see that when count is not zero, the timer is canceled, so how to continue publish?

jhedberg, 2019-08-13:
The timer will be reset through the `publish_sent()` callback in `access.c` which gets called when the PDU has been transmitted. Is this again a GATT-related issue? In that case `bt_mesh_net_send()` calls the callback synchronously, meaning the branch you've mentioned above would indeed cancel this. That said, why would you be using retransmissions for GATT? That doesn't make sense to me.

jhedberg, 2019-08-13:
That said, the call to `k_delayed_work_cancel()` does look redundant to me - perhaps it was needed before but some refactoring eliminated the need for it. I'm still looking through the code to verify this.

PR #18223 (2019-08-14): Bluetooth: Mesh: Fix canceling publication retransmission timer
This branch is inherited from old code that used to start the timer as
soon as a message was queued for transmission rather than when it was
actually transmitted (the case today). It'll also cause a race in case
the publication goes over the GATT layer since the "sent" callback
happens synchronously in that case.

Fixes #17821
commit a43ef332 (2019-08-14): Bluetooth: Mesh: Fix canceling publication retransmission timer

This branch is inherited from old code that used to start the timer as
soon as a message was queued for transmission rather than when it was
actually transmitted (the case today). It'll also cause a race in case
the publication goes over the GATT layer since the "sent" callback
happens synchronously in that case.

Fixes #17821
closed 2019-08-14 by jhedberg