zephyrproject-rtos/zephyr #17832: x86: update mmustructs.h to support long mode

andrewboie, 2019-07-28:
arch/x86/include/ia32/mmustructs.h contains all the definitions for page tables.

Right now, this only implements page tables for PAE on 32-bit x86. Need to add support for 64-bit page tables as well. I think the following needs to be done:

1. X86_MMU_GET_* macros should take a pdpt pointer for PAE, and a pml4 pointer for long mode
2. When long mode enabled, need a definition for PDPTEs which specify a 1GB area, instead of a page directory. Need a container union similar to union x86_mmu_pde.
3. When long mode enabled, need a X86_MMU_GET_PDPT_ADDR() macro
4. Need some ifdefs for the different fields in all the page table structure when long mode is enabled
5. Need MMU_*_MASK macros for both types of PDPT, and PML4
6. Our new coding guidelines don't allow anonymous unions. Remove all the value fields in the table entry structures, and instead create some inline functions to convert table entries into u64_t values.

We probably don't need to update anything related to MMU_BOOT_REGION() as the plan is to deprecate these and build all page tables at runtime.

gnuless, 2019-10-07:
@andrewboie I dunno how far along you are in this (if at all), but a shortcut here would be to assume everything we want to physically address is <= 512GB, and then use a "fixed" PML4 for each CPU and just change its first entry before "reloading" CR3.  I see little reason to expand the logic to 4 levels -- if we do want to use all 4, I'd suggest maybe we take advantage of the fact that page tables are recursive structures.

andrewboie, 2019-10-08:
> @andrewboie I dunno how far along you are in this (if at all), but a shortcut here would be to assume everything we want to physically address is <= 512GB, and then use a "fixed" PML4 for each CPU [...]

I'm just starting, but this sounds like a good idea. I'm currently adding structure definitions and ifdefs to mmustructs.h for long mode and cleaning up some cruft.

andrewboie, 2019-10-19:
> @andrewboie I dunno how far along you are in this (if at all), but a shortcut here would be to assume everything we want to physically address is <= 512GB, and then use a "fixed" PML4 for each CPU [...]

I didn't get this far. Right now we just generate a single get of page tables (or two if KPTI is turned on).

Back when I was implementing per-thread page tables for 32-bit, I considered having a master PDPT and just updating pointers to the first PDE when switching between threads. I decided not to do it at the time because it would make the switching more complex than a simple cr3 update.

PR #19809 (2019-10-24): x86: intel64: implement runtime page tables
What's changed:

- locore area split into text, rodata, data areas for proper MMU permissions
- necessary linker symbols added to intel64 linker script to compute MMU boot regions, with page alignment if runtime page tables are enabled
- MMU_BOOT_REGION() now uses Z_STRUCT_SECTION_ITERABLE()
- Enable MMU_BOOT_REGION() section for intel64
- Bring in CONFIG_X86_MMU and some related defines to common X86 Kconfig
- Don't set ARCH_HAS_USERSPACE or ARCH_HAS_STACK_PROTECTION for intel64 yet when X86_MMU is enabled, we're not there yet
- Uplevel x86_mmu.c to [... cut]
closed 2019-10-24 by andrewboie