zephyrproject-rtos/zephyr #17833: read-only memory domain partitions may affect supervisor mode on some arches

andrewboie, 2019-07-28:
The intent of the memory domain APIs is to control access to memory regions to user mode. However, in some cases, an active memory partition for a user mode thread may affect the kernel during a system call, or if we get an interrupt while that thread is running.

The specific situation is read-only memory partitions. We have a K_MEM_PARTITION_P_RW_U_RO which allows for a read-only partition that supervisor mode can still write to on ARC, ARM NXP MPU, ARMv2 MPU.

However, on X86 and ARMv8 MPU, due to hardware limitations we only have K_MEM_PARTITION_P_RO_U_RO. For these systems, consider the following:

1. User thread X is running with a memory domain that has a region Y configured with K_MEM_PARTITION_P_RO_U_RO.
2. If the thread make a system call, the system call will get a fatal protection error if it attempts to write to the memory specified by the partition on behalf of the thread.
3. If a hardware interrupt fires, the ISR code will also get a fatal protection error if it attempts to write to the memory specified by the partition.

Further complicating matters is that on x86, the above only happens if KPTI is disabled. If KPTI is turned on, there's a page table swap to the supervisor mode page tables which don't have any of the memory partition policy applied and the ISR or syscall may freely write to the memory.

Filing as an enhancement for now, we need to figure out how to adequately document this detail, and consider the best way to deal with read-only regions, which may involve memory domain API changes, or new assertions, etc.

andrewboie, 2019-07-28:
@ioannisg FYI...is the lack of K_MEM_PARTITION_P_RW_U_RO on ARMv8 an actual HW limitation, or did we just not define it

ioannisg, 2019-07-28:
> @ioannisg FYI...is the lack of K_MEM_PARTITION_P_RW_U_RO on ARMv8 an actual HW limitation, or did we just not define it

It is a HW limitation (the policy is not allowed by the v8m MPU)

dcpleung, 2024-02-08:
This is architecture specific so the next step is to review each architecture and see what can be done.