zephyrproject-rtos/zephyr #17838: state DEVICE_PM_LOW_POWER_STATE of Device Power Management

wentongwu, 2019-07-29:
Currently Device Power Management only support state DEVICE_PM_ACTIVE_STATE and DEVICE_PM_SUSPEND_STATE. And DEVICE_PM_SUSPEND_STATE will be triggered when system will enter into deep sleep state. But for non deep sleep state, device power management will not be considered. State of DEVICE_PM_LOW_POWER_STATE requires less time than DEVICE_PM_SUSPEND_STATE, so I suggest to schedule state DEVICE_PM_LOW_POWER_STATE for devices when system will suspend but not deep sleep.

wentongwu, 2019-07-29:
@pizi-nordic could you please give some comments, thanks a lot.

wentongwu, 2019-08-05:
Consider more, in the documentation we claim support DEVICE_PM_LOW_POWER_STATE, DEVICE_PM_SUSPEND_STATE, DEVICE_PM_OFF_STATE, DEVICE_PM_ACTIVE_STATE states, when system going to suspend state, we let device driver decide himself's state based on the system idle ticks, and then system will make the final decision based on every device's feedback. 
@pizi-nordic could you please take a look this issue and give some comments? Thanks

pizi-nordic, 2019-08-13:
@wentongwu: Hi. I apologize for the delay. I already looked at your PR.

wentongwu, 2019-08-14:
> @wentongwu: Hi. I apologize for the delay. I already looked at your PR.

@pizi-nordic no problem, thanks for the review.

PR #21613 (2020-01-04): power: add DEVICE_PM_LOW_POWER_STATE for device power management
when system going to sleep state(not deep sleep), make peripherals
go to DEVICE_PM_LOW_POWER_STATE state which need less time than
DEVICE_PM_SUSPEND_STATE to save more power in the sleep state.

Fixes: #17838.
PR #21693 (2020-01-08): [Backport 21613 to v1.14 branch] power: add DEVICE_PM_LOW_POWER_STATE for device power management
when system going to sleep state(not deep sleep), make peripherals
go to DEVICE_PM_LOW_POWER_STATE state which need less time than
DEVICE_PM_SUSPEND_STATE to save more power in the sleep state.

Fixes: #17838.
closed 2020-01-04 by nashif