zephyrproject-rtos/zephyr #17843: Bluetooth: controller: v1.14.x release conformance test failures

cvinayak, 2019-07-29:
**Describe the bug**
Fix BT LL TS 5.1 test case failures

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=nrf52840_pca10056 ../samples/bluetooth/hci_uart
3. make

**Expected behavior**
Pass all conformance tests

**Impact**
showstopper

**Screenshots or console output**
-

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: gnuarmemb
 - Commit SHA or Version used: v1.14.1-rc1

**Additional context**
These are failures due to changed test case procedures and/or test case changes.

cvinayak, 2019-08-27:
This is for v1.14.1 and the fix in #17720

carlescufi, 2019-08-27:
> Is this something that needs to be in 2.0 or just 1.14.1?

@rljordan only 1.14.1. The fixes are already in master for 2.0.

PR #17720 (2019-08-14): Bluetooth: controller: Conformance testing backports to v1.14 branch
Related to #17097, #17643, #17030, and #14506.

Fixes #17843.