zephyrproject-rtos/zephyr #17852: Cmsis_rtos_v2_apis test failed on iotdk board.

chen-png, 2019-07-30:
**Describe the bug**
The cmsis_rtos_v2_apis test failed on iotdk board, we found that it will occur "Zephyr fatal error 2: Stack overflow" when running cmsis_rots_v2_apis test.

**To Reproduce**
Steps to reproduce the behavior:
1. source zephyr-env.sh
2. west build -b iotdk -p=auto tests/cmsis_rtos_v2
3. screen /dev/ttyUSB0 115200
4. west flash

**Screenshots or console output**
[image]

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr SDK
 - Commit ID: e806bd977955

IRISZZW, 2019-07-31:
This exception caused by violation stack memory access, as arc hardware stacking has been enable by default.

in function osThreadGetStackSpace called in <zephyr_root>/tests/cmsis_rtos_v2/src/thread_apis.c line 87, it will read from the unallocated stack region, which is not allowed.

you can disable arc hardware stacking use menuconfig.
(Top) → ARC Options → ARCv2 Family Options

[image]

Then rebuild and run.

chen-png, 2019-08-01:
@IRISZZW Yes, you are right. I add one line in the zephyr/tests/cmsis_rtos_v2/prj.conf, 
        +    CONFIG_ARC_HAS_STACK_CHECKING=n
and then the issue has been solved.

ioannisg, 2019-08-27:
@vonhust will you be able to take a look?

maksimmasalski, 2019-09-30:
@vonhust Are you working on that issue?

PR #22520 (2020-02-08): kernel: overhaul unused stack measurement
The existing stack_analyze APIs had some problems:

1. Not properly namespaced
2. Accepted the stack object as a parameter, yet the stack object
   does not contain the necessary information to get the associated
   buffer region, the thread object is needed for this
3. Caused a crash on certain platforms that do not allow inspection
   of unused stack space for the currently running thread
4. No user mode access

We deprecate these functions and add a new API
k_thread_stack_space_get() which addresses all of these issues.

A helper API log_stack_usage [... cut]
commit efc5fe07 (2020-02-08): kernel: overhaul unused stack measurement

The existing stack_analyze APIs had some problems:

1. Not properly namespaced
2. Accepted the stack object as a parameter, yet the stack object
   does not contain the necessary information to get the associated
   buffer region, the thread object is needed for this
3. Caused a crash on certain platforms that do not allow inspection
   of unused stack space for the currently running thread
4. No user mode access
5. Separately passed in thread name

We deprecate these functions and add a new API
k_thread_stack_space_get() which addresses all of these  [... cut]
closed 2020-02-08 by jhedberg