zephyrproject-rtos/zephyr #17853: kernel panic in tests/kernel/sched/schedule_api kernel.sched.multiq

andrewboie, 2019-07-30:
This is an intermittent issue which I believe exposes a bug in the multiq scheduler.

This is the full log:
```
SeaBIOS (version rel-1.12.0-0-ga698c8995f-prebuilt.qemu.org)
Booting from ROM..***** Booting Zephyr OS build zephyr-v1.14.0-2699-ge806bd977955 *****
Running test suite threads_scheduling
===================================================================
starting test - test_bad_priorities
PASS - test_bad_priorities
===================================================================
starting test - test_priority_cooperative
PASS - test_priority_cooperative
===================================================================
starting test - test_priority_preemptible
PASS - test_priority_preemptible
===================================================================
starting test - test_yield_cooperative
PASS - test_yield_cooperative
===================================================================
starting test - test_sleep_cooperative
PASS - test_sleep_cooperative
===================================================================
starting test - test_sleep_wakeup_preemptible
PASS - test_sleep_wakeup_preemptible
===================================================================
starting test - test_pending_thread_wakeup
PASS - test_pending_thread_wakeup
===================================================================
starting test - test_time_slicing_preemptible
PASS - test_time_slicing_preemptible
===================================================================
starting test - test_time_slicing_disable_preemptible
PASS - test_time_slicing_disable_preemptible
===================================================================
starting test - test_lock_preemptible
PASS - test_lock_preemptible
===================================================================
starting test - test_unlock_preemptible
PASS - test_unlock_preemptible
===================================================================
starting test - test_sched_is_preempt_thread
PASS - test_sched_is_preempt_thread
===================================================================
starting test - test_slice_reset

    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_sched_timeslice_reset.c:79: thread_time_slice: (t <= expected_slice_max is false)
timeslice too big, expected 20100000 got 20126992
FAIL - test_slice_reset
===================================================================
starting test - test_slice_scheduling
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
PASS - test_slice_scheduling
===================================================================
starting test - test_priority_scheduling
ASSERTION FAIL [(addr & 0xfffU) == 0U] @ ZEPHYR_BASE/arch/x86/core/ia32/x86_mmu.c:192
	unaligned address provided
FATAL: eax: 0x00000008, ebx: 0x00429ecc, ecx: 0x0000000a, edx: 0x00429ec8
FATAL: esi: 0x00000007, edi: 0x00401fc0, ebp: 0x00429ee4, esp: 0x00429eb8
FATAL: eflags: 0x00000097 cs: 0x0008
FATAL: call trace:
FATAL: eip: 0x00004544
FATAL:      0x00004ffc (0x440000)
FATAL:      0x00004e1a (0x429ecc)
FATAL:      0x000015f5 (0x43609c)
FATAL:      0x00005e24 (0x3f04)
FATAL:      0x00003ed6 (0x400118)
FATAL: >>> ZEPHYR FATAL ERROR 4: Kernel panic
FATAL: Current thread: 0x00401e40 (unknown)
FATAL: Halting system
```

The bug is the panic in test_priority_scheduling.
This panic ONLY occurs if the earlier test_slice_reset case fails. If that case passes, then test_priority_scheduling proceeds without incident. test_slice_reset fails intermittently in QEMU, I had to run this a few times to get it to do it.

I have only seen this in the kernel.sched.multiq scenario; I never see this in the other scenarios in tests/kernel/sched/schedule_api/testcase.yaml

This seems to be some kind of memory corruption; the crash happens in _x86_swap_update_page_tables, which gets called by __swap.

Of particular importance to note:

```
FATAL: Current thread: 0x00401e40 (unknown)
```

_current hasn't been updated yet; that happens later in __swap. the bad addresses sent to the MMU are from thread structures from the *incoming* thread; the above address is the outgoing thread (ztest_thread).

As far as I can tell, the incoming thread pointer is corrupted, it does not correspond to any struct k_thread object.

andrewboie, 2019-07-31:
Hmm mot specific to multiq, as I have discovered. The regular scenario can fail like this as well:

```
===================================================================
starting test - test_slice_reset

    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_sched_timeslice_reset.c:79:)
timeslice too big, expected 20100000 got 20114088
FAIL - test_slice_reset
===================================================================
starting test - test_slice_scheduling
FATAL: ***** CPU Page Fault (error code 0x00000003)
FATAL: Supervisor thread wrote address 0x001143be
FATAL: PDE: 0x025 Present, Read-only, User, Execute Enabled
FATAL: PTE: 0x8000000025 Present, Read-only, User, Execute Disable
FATAL: eax: 0x001143ba, ebx: 0x00402280, ecx: 0x00402280, edx: 0x00000001
FATAL: esi: 0x00450ee4, edi: 0x00402484, ebp: 0x00450f14, esp: 0x00450ed4
FATAL: eflags: 0x00000002 cs: 0x0008 cr3: 0x00467000
FATAL: call trace:
FATAL: eip: 0x000037fc
FATAL:      0x000079ca (0x402484)
FATAL:      0x000082c6 (0x402484)
FATAL:      0x00002f5e (0x402280)
FATAL:      0x00005f92 (0x0)
FATAL:      0x00003ef1 (0x400104)
FATAL: >>> ZEPHYR FATAL ERROR 0: CPU exception
FATAL: Current thread: 0x00402280 (unknown)
FATAL: Halting system
```

In this case, the crash occurred during a z_thread_priority_set() call, in set_child(). The thread pointer is valid.

Like the other case, a previous test had to fail.

andrewboie, 2019-07-31:
Seen on ARM this is not x86 specific:

```
===================================================================
starting test - test_slice_reset

    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_sched_timeslice_reset.c:79: thread_time_slice: (t <= expected_slice_max is false)
timeslice too big, expected 5025000 got 5026070
FAIL - test_slice_reset
===================================================================
starting test - test_slice_scheduling
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
PASS - test_slice_scheduling
===================================================================
starting test - test_priority_scheduling
ASSERTION FAIL [(u32_t)thread->stack_obj == guard_start] @ ZEPHYR_BASE/arch/arm/core/cortex_m/mpu/arm_core_mpu.c:236
	Guard start (0xffffffe9) not beginning at stack object (0x4fbf)

FATAL: r0/a1:  0x00000004  r1/a2:  0x000000ec  r2/a3:  0x00000000
FATAL: r3/a4:  0x0000000e r12/ip:  0x00000000 r14/lr:  0x00002e7d
FATAL:  xpsr:  0x6100000e
FATAL: Faulting instruction address (r15/pc): 0x0000a4ce
FATAL: >>> ZEPHYR FATAL ERROR 4: Kernel panic
```

ioannisg, 2019-08-23:
We see this sporadically on ARM boards as well, not only qemu