zephyrproject-rtos/zephyr #17870: Incorrect report received lenght and offset in async API

YuraCobain, 2019-07-30:
During testing of streaming mode using async API inconsistent values for evt->data.rx.len and 
evt->data.rx.offset are reported via UART_RX_RDY ready event.
The issue is happened right after switching to the new buffer when current rx is timeout.  

Probably the bug is in incorrect check if there is more data to sent to user in following function:
static void rx_timeout(struct k_timer *timer)
{
        ...
	/* Check if there is data that was not sent to user yet */
	if (data->async->rx_total_byte_cnt
	    != data->async->rx_total_user_byte_cnt) { <--- probably should be compare less than

After changing from not equal to less than comparison the bug is bot observed any more.

Steps to reproduce the behavior:
Due to lack of time I haven't wrote the test for UART async API.
However the scenario is quite simple:

1. Initiate streaming mode for UART using chained buffers with length 32 bytes of one buffer and 
     rx timeout 10ms;
2. Transmit data to receiver with length of 32 bytes and random timeout 8-12 ms;
3. When random tx timeout exceed rx timeout  UART_RX_RDY is event will be generated with incorrect length value, next  UART_RX_RDY will have incorrect offset value.

 - OS: Ubuntu 18.04
 - Toolchain zephyr-sdk-0.10.1
 - zephyr branch master @20ccdab

ioannisg, 2019-07-30:
@YuraCobain Could you indicate the platform you have used?

YuraCobain, 2019-07-30:
I am using particle xenon(nrf52840).

ioannisg, 2019-07-31:
@Mierunski can you follow-up on this bug report?

Mierunski, 2019-08-02:
I'll need more information on this:
- Are you using UART or UARTE?
- If you are using UARTE, are you using hardware RX byte counting? (UART_0_NRF_HW_ASYNC)
- What is the baudrate?

YuraCobain, 2019-08-02:
Here is the configuration of my serial driver:
#
# Serial Drivers
#

# CONFIG_UART_0_NRF_HW_ASYNC is not set
CONFIG_UART_NRFX=y
CONFIG_UART_0_NRF_UARTE=y
CONFIG_UART_0_ASYNC=y
CONFIG_UART_0_NRF_TX_BUFFER_SIZE=32
CONFIG_NRF_UARTE_PERIPHERAL=y

Baudrate is 115200.

YuraCobain, 2019-08-05:
The log and archive with the app that reproduce the bug is is attached bellow.
RX and TX is looped on DT_NORDIC_NRF_UARTE_UART_0_LABEL.
[17870_uart_async_api.log](https://github.com/zephyrproject-rtos/zephyr/files/3466959/17870_uart_async_api.log)

YuraCobain, 2019-08-05:
/* main.c - Application main entry point */

/*
 * Copyright (c) 2016 Intel Corporation
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <stdbool.h>
#include <zephyr/types.h>
#include <stddef.h>
#include <string.h>
#include <errno.h>
#include <sys/printk.h>
#include <sys/byteorder.h>
#include <zephyr.h>
#include <drivers/uart.h>

K_SEM_DEFINE(rx_disabled, 0 ,1);
K_SEM_DEFINE(rx_ready, 0 ,1);

u8_t rx_double_buffer[2][256];
u8_t *next_rx_buf = rx_double_buffer[1];
u8_t tx_buffer[130];

#define TX_STACK_SIZE 256
#define TX_PRIORITY 5

static void tx_thread(void *, void *, void *);

K_THREAD_STACK_DEFINE(tx_stack, TX_STACK_SIZE);
struct k_thread tx_thread_data;

#define RX_STACK_SIZE 256
#define RX_PRIORITY 5

static void rx_thread(void *, void *, void *);

K_THREAD_STACK_DEFINE(rx_stack, TX_STACK_SIZE);
struct k_thread rx_thread_data;

void test_long_buffers_with_timeouts_callback(struct uart_event *evt, void *user_data)
{
	struct device *uart_dev = (struct device *) user_data;

	switch (evt->type) {
	case UART_TX_DONE:
		break;
	case UART_TX_ABORTED:
		break;
	case UART_RX_RDY:
        printk("offset:%zu len:%zu\n", evt->data.rx.offset, evt->data.rx.len);
		break;
	case UART_RX_BUF_REQUEST:
		uart_rx_buf_rsp(uart_dev, next_rx_buf, 256);
		break;
	case UART_RX_BUF_RELEASED:
		next_rx_buf = evt->data.rx_buf.buf;
		break;
	case UART_RX_DISABLED:
		break;
	default:
		break;
	}
}

#define UART_DEVICE_NAME DT_UART_0_NAME
void main(void)
{
	struct device *uart_dev = device_get_binding(UART_DEVICE_NAME);

    k_thread_create(&rx_thread_data, rx_stack,
                                 K_THREAD_STACK_SIZEOF(rx_stack),
                                 rx_thread,
                                 uart_dev, NULL, NULL,
                                 RX_PRIORITY, 0, K_NO_WAIT);

    k_thread_create(&tx_thread_data, tx_stack,
                                 K_THREAD_STACK_SIZEOF(tx_stack),
                                 tx_thread,
                                 uart_dev, NULL, NULL,
                                 TX_PRIORITY, 0, K_NO_WAIT);

    k_sem_take(&rx_disabled, K_FOREVER);
}

void rx_thread(void *p1, void *p2, void *p3)
{
	struct device *uart_dev = p1;

	uart_callback_set(uart_dev, test_long_buffers_with_timeouts_callback, uart_dev);

	uart_rx_enable(uart_dev, rx_double_buffer[0], 256, 10);

    printk("rx ready\n");
    k_sem_give(&rx_ready);
    while(1)
    {
        k_sleep(50);
    }
}

void tx_thread(void *p1, void *p2, void *p3)
{
	struct device *uart_dev = p1;

    k_sem_take(&rx_ready, K_FOREVER);
    printk("tx ready\n");

    while(1)
    {
        static int counter = 0;
        uart_tx(uart_dev, tx_buffer, 67, 5);
        counter++;
        k_sleep(counter % 5 ? 5 : 20);
    }
}

heypiotr, 2019-08-06:
I've run into similar or the same problem, with a similar configuration to @YuraCobain. (Particle Boron, UARTE, no hardware counting, 115200)

A quick session with the debugger:

```
Breakpoint 2, rx_timeout (timer=0x200001cc <uarte0_async+92>) at /Users/piotr/Code/Playground/zephyr/zephyr/drivers/serial/uart_nrfx_uarte.c:609
609				u32_t len = data->async->rx_total_byte_cnt
(gdb) list 
604		if (data->async->rx_total_byte_cnt
605		    != data->async->rx_total_user_byte_cnt) {
606			if (data->async->rx_timeout_left
607			    < data->async->rx_timeout_slab) {
608				/* rx_timeout ms elapsed since last receiving */
609				u32_t len = data->async->rx_total_byte_cnt
610					    - data->async->rx_total_user_byte_cnt;
611				struct uart_event evt = {
612					.type = UART_RX_RDY,
613					.data.rx.buf = data->async->rx_buf,
(gdb) p data->async->rx_total_byte_cnt
$1 = 2393
(gdb) p data->async->rx_total_user_byte_cnt
$2 = 3056
(gdb) n
611				struct uart_event evt = {
(gdb) p len 
$3 = 4294966633
```

… that is, if I understand correctly, total bytes received (2393) is somehow less than total bytes delivered to the user/callback (3056).

Perhaps most importantly, **enabling hardware counting solved this problem for me.**

YuraCobain, 2019-08-06:
@heypiotr, thank you for advice but enabling hardware counting(CONFIG_UARTE_NRF_HW_ASYNC) have not solved the issue for me.
However, as I already mentioned changing condition from **not equal** to **less than** in function:
````
static void rx_timeout(struct k_timer timer)
{
...
/ Check if there is data that was not sent to user yet */
if (data->async->rx_total_byte_cnt
**<** data->async->rx_total_user_byte_cnt) { ...}
```

works for me.