zephyrproject-rtos/zephyr #17876: BME680 sensor sample not building

mrusme, 2019-07-30:
**Describe the bug**

The sensor sample for BME680 is not building as [described here](https://docs.zephyrproject.org/latest/samples/sensor/bme680/README.html).

I solved this by copying the code into an own project, specifying the following definitions inside the CMakeLists and using cmake/ninja instead of west:

```
add_definitions(-DDT_INST_0_BOSCH_BME680_LABEL="BME680" -DDT_INST_0_BOSCH_BME680_BASE_ADDRESS=0x77 -DDT_INST_0_BOSCH_BME680_BUS_NAME="I2C_1" -DDT_BOSCH_BME680_BUS_I2C=1)
```

However, when running the build on the Nucleo board I still get this:

```
***** Booting Zephyr OS build zephyr-v1.14.0-2708-g11d8f1a1e972 *****                                                              │
Device 0x0 name is                                                                                                                 │
FATAL: ***** MPU FAULT *****                                                                                                       │
FATAL:   Instruction Access Violation                                                                                              │
FATAL: r0/a1:  0x00000000  r1/a2:  0x00000025  r2/a3:  0x00000000                                                                  │
FATAL: r3/a4:  0x406100f4 r12/ip:  0x00000012 r14/lr:  0x080006b9                                                                  │
FATAL:  xpsr:  0x20000000                                                                                                          │
FATAL: Faulting instruction address (r15/pc): 0x406100f4                                                                           │
FATAL: >>> ZEPHYR FATAL ERROR 0: CPU exception                                                                                     │
FATAL: Current thread: 0x200001b0 (unknown)                                                                                        │
FATAL: Halting system                                                                                                              │
                                                                                                                                   │
                  
```

I searched the Zephyr code for possible values for these definitions but could not find any. Nor are they documented anywhere (or at least it's not obvious enough so that I could find it).

**To Reproduce**

Steps to reproduce the behavior:
1. `west build -p -b nucleo_l476rg samples/sensor/bme680`

**Expected behavior**

The sample should build and the build should run.

**Screenshots or console output**

```
(local) (master) % west build -p -b nucleo_l476rg samples/sensor/bme680
-- west build: making build dir /home/x90/Projects/myproject/zephyr/build pristine                                              
-- west build: build configuration:                                                                                               
       source directory: /home/x90/Projects/myproject/zephyr/samples/sensor/bme680                                              
       build directory: /home/x90/Projects/myproject/zephyr/build                                                               
       BOARD: nucleo_l476rg (origin: command line)                                                                                
-- west build: generating a build system                                                                                          
Zephyr version: 1.14.99                                                                                                           
-- Found PythonInterp: /home/x90/.virtualenvs/local/bin/python (found suitable version "3.6.5", minimum required is "3.4")        
-- Selected BOARD nucleo_l476rg                                                                                                   
-- Found west: /home/x90/.virtualenvs/local/bin/west (found suitable version "0.5.7", minimum required is "0.5.6")                
-- Loading /home/x90/Projects/myproject/zephyr/boards/arm/nucleo_l476rg/nucleo_l476rg.dts as base                               
-- Overlaying /home/x90/Projects/myproject/zephyr/dts/common/common.dts                                                         
Device tree configuration written to /home/x90/Projects/myproject/zephyr/build/zephyr/include/generated/generated_dts_board.conf
Parsing Kconfig tree in /home/x90/Projects/myproject/zephyr/Kconfig                                                             
Loaded configuration '/home/x90/Projects/myproject/zephyr/boards/arm/nucleo_l476rg/nucleo_l476rg_defconfig'                     
Merged configuration '/home/x90/Projects/myproject/zephyr/samples/sensor/bme680/prj.conf'                                       
Configuration saved to '/home/x90/Projects/myproject/zephyr/build/zephyr/.config'                                               
-- Cache files will be written to: /home/x90/.cache/zephyr                                                                        
-- The C compiler identification is GNU 8.3.0                                                                                     
-- The CXX compiler identification is GNU 8.3.0                                                                                   
-- The ASM compiler identification is GNU                                                                                         
-- Found assembler: /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc                                                       
-- Performing Test toolchain_is_ok                                                                                                
-- Performing Test toolchain_is_ok - Success                                                                                      
Including module: esp-idf in path: /home/x90/Projects/myproject/modules/hal/esp-idf/zephyr                                      
Including module: fatfs in path: /home/x90/Projects/myproject/modules/fs/fatfs                                                  
Including module: qmsi in path: /home/x90/Projects/myproject/modules/hal/qmsi                                                   
[... 30 lines cut ...]
rojects/myproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mthumb -mcpu=cortex-m4 -imacro
s/home/x90/Projects/myproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length 
-Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno
-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/x90/Projects/myproject/zephyr/
samples/sensor/bme680=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/x90/Projects/myproject/zephyr=ZEPHYR_BASE -ffunction-sections -f
data-sections -mabi=aapcs -march=armv7e-m -std=c99 -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj
.d -o CMakeFiles/app.dir/src/main.c.obj   -c /home/x90/Projects/myproject/zephyr/samples/sensor/bme680/src/main.c               
/home/x90/Projects/myproject/zephyr/samples/sensor/bme680/src/main.c: In function 'main':                                       
/home/x90/Projects/myproject/zephyr/samples/sensor/bme680/src/main.c:14:42: error: 'DT_INST_0_BOSCH_BME680_LABEL' undeclared (fi
rst use in this function); did you mean 'DT_INST_0_SOC_NV_FLASH_LABEL'?                                                           
  struct device *dev = device_get_binding(DT_INST_0_BOSCH_BME680_LABEL);                                                          
                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                            
                                          DT_INST_0_SOC_NV_FLASH_LABEL                                                            
/home/x90/Projects/myproject/zephyr/samples/sensor/bme680/src/main.c:14:42: note: each undeclared identifier is reported only on
ce for each function it appears in                                                                                                
[22/127] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/thread_entry.c.obj                                                 
ninja: build stopped: subcommand failed.                                                                                          
ERROR: command exited with status 1: /usr/bin/cmake --build /home/x90/Projects/myproject/zephyr/build                           
run as "west -v build -p -b nucleo_l476rg samples/sensor/bme680" for a stack trace                                              
```

**Environment (please complete the following information):**
 - OS: Gentoo Linux
 - Toolchain: Zephyr
 - SDK-ng 0.10.1

mrusme, 2019-07-31:
@carlescufi maybe you could just point me into the right direction in regard of the defines I mentioned above, so that I know what values they should be set to in order for this to work, so I can figure it out on my own and in the best case set a PR to this repo? :)

Thanks a lot!

carlescufi, 2019-07-31:
@mrusme you need to add a BME680 to your Device Tree file I believe.
@jhn-nordic, @joakimtoe, @jtguggedal or @MaureenHelm can hopefully help you out here

mrusme, 2019-07-31:
Thanks for the hint. I checked [the documentation](https://docs.zephyrproject.org/latest/guides/dts/index.html?highlight=device%20tree) but did not see a way to **extend** the existing device tree for a board with a sensor that was additionally added over I2C (apart from creating a "new" board and including the sensor within the DTS maybe, although this might lead to having to back-merge all changes to the original board every time they occur?).

What is the best approach to take here? 

Thanks a lot!

carlescufi, 2019-07-31:
@mrusme the options you have are:

1. Define your own board (see [here](https://docs.zephyrproject.org/latest/guides/porting/board_porting.html))
2. Add a `nucleo_l476rg.overlay` to your application with the BME680 node, as described [here](https://docs.zephyrproject.org/latest/application/index.html#device-tree-overlays)

carlescufi, 2019-07-31:
@mrusme if you solve the problem by adding the node to DT, can you close this issue?

mrusme, 2019-07-31:
@carlescufi I guess this could be closed then, yes. I would re-open another issue, if something with the DT overlay would not work. Nevertheless it maybe should be documented somewhere. Let me fiddle around with it further, if I manage to get it working I'll set a PR with the changes to the docs.

closed 2019-07-31 by mrusme