zephyrproject-rtos/zephyr #17882: [GATT caching] Database Out of Sync error is not returned as expected

MariuszSkamra, 2019-07-30:
**Describe the bug**
In test case GATT/SR/GAS/BV-05-C, step 8 the Database Out of Sync error is not returned as expected on GATT Characteristic Value Read after, the Zephyr GATT database changed.

**To Reproduce**
Steps to reproduce the behavior:
0. Set TPSX_iut_device_name_in_adv_packet_for_random_address = tester
1. Run bluetooth/shell application
2. bt init
3. gatt register
4. bt clear all (as requested by PTS)
5. gatt metrics on/off (Add/Remove GATT Metrics service)

**Expected behavior**
Test Case pass

**Impact**
Qualification test case GATT/SR/GAS/BV-05-C is failing

**Additional context**
Add any other context about the problem here.

Vudentz, 2019-07-30:
Hmm maybe you just want to toggle the metrics so that guarantees a new hash is generated, I can have a look tomorrow.

Vudentz, 2019-07-30:
It does look like we are reseting change-aware flag which should trigger the right error, so we may need some logs here.

MariuszSkamra, 2019-07-30:
it appeared to be 'registered' flag not be cleared when calling "gatt metrics off"

PR #17939 (2019-08-05): Bluetooth: Shell: Fix unregister GATT Metrics vendor service
The 'registered' flag was not cleared when 'gatt metrics off'
was called so that one was unable to register the service again.
Remove 'registered' variable that is actually not needed as there
is no tracking if service is already registered in similar
cmd_register_test_svc command. If the service is already registered,
the host will log an error.

Fixes: #17882
commit 7c603007 (2019-08-05): Bluetooth: Shell: Fix unregister GATT Metrics vendor service

The 'registered' flag was not cleared when 'gatt metrics off'
was called so that one was unable to register the service again.
Remove 'registered' variable that is actually not needed as there
is no tracking if service is already registered in similar
cmd_register_test_svc command. If the service is already registered,
the host will log an error.

Fixes: #17882
commit b3a4d9c3 (2019-08-15): Bluetooth: Shell: Fix unregister GATT Metrics vendor service

The 'registered' flag was not cleared when 'gatt metrics off'
was called so that one was unable to register the service again.
Remove 'registered' variable that is actually not needed as there
is no tracking if service is already registered in similar
cmd_register_test_svc command. If the service is already registered,
the host will log an error.

Fixes: #17882
closed 2019-08-05 by carlescufi