zephyrproject-rtos/zephyr #17891: nvs_init can hang if no nvs_ate available

pfvogel, 2019-07-30:
**Describe the bug**
While running on a simulated flash file system, nvs_init() failed to return.  Inspection with debugger shows that it stuck in an infinite loop.

**To Reproduce**
Calling nvs_init() on flash file system with no "free" sectors.

**Expected behavior**
If no free space available, return an error.

**Impact**
Potential malfunction of firmware on startup.

**Screenshots or console output**
The code is stuck in the subsys/fs/nvs/nvs.c file in the while(1) loop around line 580.  If the nvs_flash_cmp_const() routine never returns a match, the loop will never break.  The address being checked decrements past zero and wraps.  The fix is to check that addr is greater than zero or return an error.

**Environment (please complete the following information):**
 - OS: Linux - native_posix platform
 - Toolchain Zephyr SDK
 - Commit d8c5c9dcf12eda7e5edb86e3ce4971e750d0c7af, but code as same as current master on github.

**Additional context**
Add any other context about the problem here.

nvlsianpu, 2019-07-31:
@pfvogel 

> To Reproduce
Calling nvs_init() on flash file system with no "free" sectors.

Can I ask you for more details about exact nvs configuration and state?
- Does state was: nvs contains N sectors. N-2 sectors are closed, one sector is open and full, one -sector is empty (it is the sector which takes role of scratch during GC)?
- value of N?

Maybe you have code to reproduce the issue you can share?

nvlsianpu, 2019-07-31:
I see that this issue will be encountered if the last `ate` is close to `data`. Close mean that gap between is smaller than the `ate size`.

nvlsianpu, 2019-07-31:
> The fix is to check that addr is greater than zero or return an error.

Sector might be valid while this happens. I thin solution will be to count data position (so read each ate, what happens anyway) and iterate until gap space is available.

nvlsianpu, 2019-07-31:
@pfvogel Can you test below patch?
https://github.com/nvlsianpu/zephyr/commit/b91410d97b10eee1cac07556390bb0c5b284680d

I want to add a test for the issue, this is why I didn't open PR jet.

Laczen, 2019-07-31:
Hi @pfvogel , @nvlsianpu, I think this problem is not so easy to solve, it has been introduced by the sector close are. It can appear if you write the fs full and finish with a delete. In that case the sector is completely filled and an empty ate will not be found. In this special case the closing ate could be used 
to find the last ate.
But even in this case there still is a problem, what if the closing ate is not properly written?
Taking this into account I think it is better to leave at least one ate as a separation between data and ate's, this could be achieved by adding an extra ate in line 811 of nvs.c (maybe the space for the delete can be removed).

nvlsianpu, 2019-08-01:
@Laczen How to achieve state of storage described above? Apparently I missed something.

Laczen, 2019-08-01:
@nvlsianpu, I think you can achieve this by writing the same 8 byte structure (e.g 2 u32_t) again with different id until the fs is full. Once you have reached this you write a delete of the last id. If you then restart I think you will get this situation.

nvlsianpu, 2019-08-01:
https://github.com/nvlsianpu/zephyr/commit/66460acd08226bee55d346c2e7878ea3eac52d8b#diff-e1cabdfdfe297813e46c8ed0df375340R490-R518

@Laczen I added test you mnetioned. Unfortuantallly it assert on nvs_delete call in line 514.
`

Laczen, 2019-08-01:
@nvlsianpu, took me some time to realize: this is normal: filling_id was never written so it cannot be deleted, the delete call should be filling_id - 1.

nvlsianpu, 2019-08-01:
@Laczen 
> took me some time to realize: this is normal: filling_id was never written so it cannot be deleted, the delete call should be filling_id - 1.

I corrected the  ID to existing one - results was the same.

nvlsianpu, 2019-08-01:
Error is -ENOSPC
```
    Assertion failed at ZEPHYR_BASE/tests/subsys/fs/nvs/src/main.c:515: test_nvs_full_sector: (err == 0 is false)
nvs_delete call failure: -28
FAIL - test_nvs_full_sector
```

nvlsianpu, 2019-08-01:
@Laczen From code I see that delete entry is not treated specially in nvs_write() - it is normal that sector_freespace can be equal 0 while writing it.

nvlsianpu, 2019-08-01:
^^ https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/fs/nvs/nvs.c#L811-L818

Laczen, 2019-08-01:
@nvlsianpu, I think it is impossible to write a sector full so that there is no empty ate: ate_wra is updated each time a write is performed and is already pointing to an empty ate area. The free_sector_space check makes sure that there is at least one ate space.

So probably the bug is something different.

pfvogel, 2019-08-01:
I think one (probably unusual) possible way the sector could look full is if the flash data area had not been erased prior to first use, say if the data area was moved from one area to another.  The nvs_init wouldn't succeed, so nvs_clear can't be called either.  Having a specific error return value from nvs_init for this case would allow the code to call a flash erase routine if this case is encountered.

Laczen, 2019-08-01:
Hi @pfvogel, do you think this is what happened ?

If so a check could be added to nvs_init to make sure that addr does not go below 0.

pfvogel, 2019-08-01:
In the test case that I ran, the flash area was uninitialized, so I think that was what happened.
[nvs_init_hang.zip](https://github.com/zephyrproject-rtos/zephyr/files/3457845/nvs_init_hang.zip)
I added a small test case that runs under sanitycheck for the native posix environment that showed my original problem.

nvlsianpu, 2019-08-02:
@pfvogel  https://github.com/zephyrproject-rtos/zephyr/pull/17982

nvlsianpu, 2019-08-02:
@pfvogel Can you agree that issue you discovered is another manifestation fo this bug: https://github.com/zephyrproject-rtos/zephyr/issues/16899?

pfvogel, 2019-08-02:
Yes, it looks like the same issue.

commit 369510af (2019-08-08): fs/nvs: fix initialization hang if sector is full

During initialization NVS analyzes open sector for
find write addresses. It was possible to infinite loop
in case when there was no space for any new ATE in the sector.

This patch introduce check for space available in the sector which
fix the issue.

fixes #17891
commit e54f97e5 (2019-08-15): fs/nvs: fix initialization hang if sector is full

During initialization NVS analyzes open sector for
find write addresses. It was possible to infinite loop
in case when there was no space for any new ATE in the sector.

This patch introduce check for space available in the sector which
fix the issue.

fixes #17891
closed 2019-08-08 by carlescufi