zephyrproject-rtos/zephyr #17897: k_busy_wait not working when using 32KHz timer driver

albertofloyd, 2019-07-30:
k_busy_wait results in inaccurate delays causing MEC1501 I2C driver to always report timeout.
This breaks I2C tests in this board. 

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2.  cmake -DBOARD=mec15xxevb_assy6853 
3. make

**Expected behavior**
k_busy_wait should result in expected delays. 

**Impact**
This affects behavior from applications relaying in k_sleep.
Additionally usage of this driver cause most timer tests to fail

**Additional context**

andrewboie, 2019-07-31:
@scottwcpg would be the best person to work on this

andyross, 2019-08-14:
Updating for posterity: this is a @scottwcpg bug, but alas we can't assign it to him because he's not administratively added to the  github project yet.

MaureenHelm, 2019-08-19:
> Updating for posterity: this is a @scottwcpg bug, but alas we can't assign it to him because he's not administratively added to the github project yet.

Invite sent to @scottwcpg

pabigot, 2019-08-19:
See #15707 for how this was handled in Nordic.

albertofloyd, 2019-08-22:
This has been fixed as part of https://github.com/zephyrproject-rtos/zephyr/pull/18318
Will close when that's merged

PR #18318 (2019-09-03): Devel mec1501 rtos timer load fix 1
Latest MEC1501 RTOS timer changes. Works around HW taking up to one 32KHz cycle to move load register value into count register. Driver has been tested by MCHP and Intel.

Closes #17897
closed 2019-09-03 by ioannisg