zephyrproject-rtos/zephyr #17899: tests/kernel/mem_protect/stackprot/kernel.memory_protection fails on nsim_em

andrewboie, 2019-07-30:
```
Scanning dependencies of target run
Console now belongs to UART, hit CRTL-] to return to simulator.
***** Booting Zephyr OS build zephyr-v1.14.0-2730-g63af2e08fb33 *****
Running test suite stackprot
===================================================================
starting test - test_create_alt_thread
Starts alternate_thread
alternate_thread: Input string is too long and stack overflowed!

FATAL: ***** Exception vector: 0x7, cause code: 0x0, parameter 0x0
FATAL: Address 0x829a
FATAL: EV_PrivilegeV
FATAL: Privilege violation
FATAL: Faulting instruction address = 0x829a
FATAL: >>> ZEPHYR FATAL ERROR 0: CPU exception
FATAL: Current thread: 0x80001000 (unknown)

    Assertion failed at ZEPHYR_BASE/tests/kernel/mem_protect/stackprot/src/main.c:20: k_sys_fatal_error_handler: (reason not equal to K_ERR_STACK_CHK_FAIL)
wrong error type
FAIL - test_create_alt_thread
===================================================================
starting test - test_stackprot
test_stackprot: Stack ok
test_stackprot: Stack ok
test_stackprot: Stack ok
test_stackprot: Stack ok
test_stackprot: Stack ok
test_stackprot: Stack ok
PASS - test_stackprot
===================================================================
Test suite stackprot failed.
===================================================================
PROJECT EXECUTION FAILED
```

andrewboie, 2019-08-06:
Looks like the problem is that the arch code is reporting a CPU exception instead of a stack overflow like it should be.

vonhust, 2019-08-07:
@andrewboie   yes, the root cuase is a little like #17590. **_StackCheckHandler** will be called when stackoverflow happens.  **_StackCheckHandler** will call **Z_ARCH_EXCEPT** in arch/v2/error.h.

**Z_ARCH_EXCEPT** will call **z_arc_v2_irq_unit_is_in_isr** which only can be called in kernel mode.

But in this test,  it's called in used mode. 

I will make a fix soon.

PR #18079 (2019-08-09): arch: arc: fixes the case triggering a cpu exception in user mode
* raise software triggerd exception unconditionally
* fix the handling when exception raised in irq context
* software-trigger exception only allow oops and stack check fail
* remove the exception of arc in tests/mem_protection/userspace

Fixes #17899 #17590
closed 2019-08-09 by ioannisg