zephyrproject-rtos/zephyr #17902: Support Quad SPI in internal SPI API

fabled, 2019-07-31:
The current SPI API `drivers/spi.h` allows to advertise Dual and Quad SPI support via `SPI_LINES_DUAL` and `SPI_LINES_QUAD`, but it does not provide needed means to tag direction of the each transfer part.

The problem is that in Dual and Quad mode there are several wires which direction needs to be configured properly (for In or Out) for each sub block of the data transfer. Also often the initial header needs to be sent in single mode and only the actual data is then sent or received in quad mode.

One option is to have `struct spi_buf` contain additional fields that describe the desired transfer type (single, dual, quad) and direction (in or out). This allows then the SPI driver to configure the hardware properly.

tbursztyka, 2019-07-31:
Isn't this too low level to be exposed to the user? The way in/out SPI lines are wired is hardware specific so that would be configured once on the pinmuxer and that would be it, no?

fabled, 2019-07-31:
> Isn't this too low level to be exposed to the user? The way in/out SPI lines are wired is hardware specific so that would be configured once on the pinmuxer and that would be it, no?

No. The direction needs to be dynamically changed within single transaction. See e.g. https://en.wikipedia.org/wiki/Serial_Peripheral_Interface#Quad_SPI. The client needs to craft the transaction and make request saying "send bits ABC in single wire mode" and those are command + address that enable quad mode, then the next portion needs to be buffer saying "rest X bits are to be (input | output) in (dual | quad) mode". This kind of bit specific transfer type setting is requirement for Quad mode to work.

And this is exactly what this feature request is about: extending the SPI API to support the above, so that Quad SPI can be implemented.

psidhu, 2019-07-31:
Might as well tack on octo-spi to this request as well.

tbursztyka, 2019-08-02:
@fabled I see, so the flags related to this feature should be part of spi_buf structure and Kconfig controlled, since most of spi controllers do not support this: it would be a waste of 4bytes in spi_buf if such flag attribute was always present. 
Let's try to make it for 2.1

fabled, 2019-08-02:
> @fabled I see, so the flags related to this feature should be part of spi_buf structure and Kconfig controlled, since most of spi controllers do not support this: it would be a waste of 4bytes in spi_buf if such flag attribute was always present.
> Let's try to make it for 2.1

Sounds good! Alternative would be to split off some bits from the `size_t len` field, but separate flags field sounds good too. If making it conditional on Kconfig, then it's probably best to introduce macro accessors for the client and driver side for the conditional items to avoid `#ifdef` flood in code files.

tbursztyka, 2019-10-22:
@fabled got some time to work on this, trying to get something as flexible as possible. Here how it would look like: 
```
/**
 * SPI Enhanced Modes: Double Data Rate, Dual, Quand and Octal modes
 *
 */
#define SPI_INSTRUCTION_STD BIT(0)
#define SPI_INSTRUCTION_DDR BIT(1)
#define SPI_INSTRUCTION_EM BIT(2)
#define SPI_ADDRESS_STD BIT(3)
#define SPI_ADDRESS_DDR BIT(4)
#define SPI_ADDRESS_EM BIT(5)
#define SPI_DATA_DDR BIT(6)

/* bit fields are used here _only_ because it's easy to read.
 * Final version will not use bit fields.
 */
struct spi_enhanced_mode {
	union {
		u16_t flags : 10;
		u16_t instruction_length : 2; /* 4 bits multiplier */
		u16_t address_length : 4; /* 4 bits multiplier */
	} config;
	u8_t rx_wait_cycles;
};

/**
 * @brief SPI buffer structure
 *
 * @param buf is a valid pointer on a data buffer, or NULL otherwise.
 * @param len is the length of the buffer or, if buf is NULL, will be the
 *    length which as to be sent as dummy bytes (as TX buffer) or
 *    the length of bytes that should be skipped (as RX buffer).
 * @param mode_flags is an optional flag used when CONFIG_SPI_ENHANCED_MODES
 *    is enabled, and relates to what is the buffer meant to (instruction,
 *    address, data) and how.
 */
struct spi_buf {
	void *buf;
	size_t len;
#ifdef CONFIG_SPI_ENHANCED_MODES
	struct spi_enhanced_mode mode;
#endif
};
```

This should be fixing: double data rate (DDR), dual, quad and octal modes.  (note that DDR is not tight to any mode).

- idea would be to provide instruction and address in the same spi_buf, and data next.
- if mode flags is 0: it's data.
- the selected  enhanced mode is still configured via spi_config
- data itself is always following spi_config selected mode (+ DDR bit if any)
- wait cycle of 8 bits is maybe a bit large, let's see.

instruction, address can be both on standard (single line), or enhanced mode (EM), or instruction standard and address EM etc... DDR can be set also.

I will provide macros to make setting/reading these in a simpler manner.

I could not find a reference where a controller allows playing with the IOs. Would that be a relevant case?

pabigot, 2019-10-24:
The current SPI API falls short in support for devicetree features.  Particularly there should be `spi-cpha` and `spi-cpol` boolean properties in SPI clients so they don't have to be hard-coded in the drivers.  I suspect other properties should be reviewed and added for QSPI support.

This also seems like something that should be reviewed in the API call as there are between four and eight distinct vendors that have QSPI capability and solutions should be suitable for all of them to implement.  @carlescufi

pabigot, 2019-10-25:
@fabled @psidhu To provide context for a discussion soon to occur in #20069:

Is the desire for Quad SPI based on a need to support a bus peripheral *other than* a (serial flash) memory device?  If so, can you identify an example of such a peripheral?

Can you identify a platform that supports generic Quad SPI transactions that is not a specific vendor peripheral interface that implements read/write operations using JEDEC commands?  (E.g. Nordic QSPI, and [from what I can tell STM32 QSPI](https://lwn.net/Articles/719996/), are not generic.)

tbursztyka, 2019-10-25:
@pabigot there is nothing generic in what you can pass through the bus. It can be anything. And btw, it's not limited to instruction/address/data format. It can be just data as well. dual/quad/octal are just improvements of SPI, but do not affect its usage target.

pabigot, 2019-10-25:
@tbursztyka yes yes, I know that's true for Dual/Quad SPI (and that 1-1-4 and 1-4-4 are different).

The point is that #20069 is based on an unstated goal of adding a driver that uses a special SPI-like peripheral that's dedicated to accessing serial flash memories with standard JEDEC commands.  For that sort of peripheral extending the Zephyr SPI API is not useful, because the peripheral doesn't support generic Quad SPI transactions (or it does, but only as a back door for special custom instructions).

What I want to know is whether the driving use case that motivated **this** issue is for such a device and platform, or for one that would require the full Quad SPI capability as a generic bus interface.

tbursztyka, 2019-10-25:
You definitely want these mode enabled as generic bus interface (through SPI, imo. No need for a new API. Or maybe then some helpers on top of the SPI API if needed).
It's true that a good usage of these modes is mainly towards flash, but it's not standardized to be the only target. It could be anything. Think about high throughput networking devices for instance.

psidhu, 2019-10-25:
@pabigot So I know of only two types of real devices that make use of at least octospi - "HyperFlash" and "HyperRAM". From what I can tell

Here's the HyperBus spec: https://www.cypress.com/file/213356/download

Regarding QSPI, there's a few devices I could see myself wanting to use, including DACs/ADCs, flash devices (I'm assuming JEDEC), and a display driver I've seen. Here's a list of qspi DACs/ADCs: https://www.digikey.com/products/en/integrated-circuits-ics/data-acquisition-adcs-dacs-special-purpose/768?k=qspi

fabled, 2019-10-28:
@pabigot 

> Is the desire for Quad SPI based on a need to support a bus peripheral _other than_ a (serial flash) memory device? If so, can you identify an example of such a peripheral?
>
> Can you identify a [...]

Thank you for working on this!

I would really like to keep the QSPI APIs generic, and not hard code assumptions about operation/address/data sequencing as defined in JEDEC. Otherwise we will be in trouble when devices come that work in different way.  Even though for me the target currently is JEDEC only.

The controller I am interested is CEC1702 QMSPI [http://ww1.microchip.com/downloads/en/DeviceDoc/00002207C.pdf chapter 29] and allows in DMA mode configuring descriptors to support iovec style scatter/gather with mode set per transfer part. My understanding is that other QSPI capable controllers have similar approach.

I was sort of expecting this kind of iovec abstraction where you can per-iovec item say the transfer part length, direction and mode for flexible abstraction. This might also enable using existing APIs and just add few iovec fields to take advantage of QSPI when available instead of requiring using complete new APIs and duplicating code when going from SPI to QSPI. For me QSPI (or the dual/octa modes) is an extension to SPI, not a separate technology.

Perhaps something like this could be considered?

kamlaz, 2019-10-28:
@fabled 

> keep the QSPI APIs generic, and not hard code assumptions about operation/address/data sequencing as defined in JEDEC

QSPI API proposed by me was developed aiming jedec flash memories, but as you analyze them, you will find out that API and argumets are so flexible that you can perform any operation you want. I.e if you want to just use QSPI as SPI, do not use address field.
In my opinion treating QSPI as an extenstion of  SPI in case of API is not good solution since:
- SPI API will get complicated
- each SPI driver has to be modified significantly - those files will become less clear and easy to read
- complexity of drives improves chances for malfunction - its better do not touch something that works

pabigot, 2019-10-28:
@fabled @psidhu thanks.

I looked at a few datasheets for the ADCs referenced and couldn't find any that used anything but the standard 4-wire SPI serial interface.  So there I think the "QSPI" search term was finding "SPI-/QSPI-/MICROWIRE-/TMS320-Compatible 4-Wire Serial Interface".  There may be quad-SPI DACs and ADCs, but I haven't been able to find one: "quad" in all matches refers to something other than SPI.

I could believe that there could be value to Zephyr in extending the existing SPI API to support generic Dual-SPI and Quad-SPI, but it's still a little murky.

NXP's i.MX processors have a flexspi peripheral that maybe is generic, but seems to be pretty biased toward flash devices.  It handles single/dual/quad/octal mode transfers specifically including Hyperflash and QSPI.  It might be a better candidate for a more specialized interface like the one being proposed in #20069, or it might be generic enough to use here.

The CEC1702 does seem like it's a target for this capability, but maybe it's the only one.

To @kamlaz's point: before saying we can provide a generic QSPI API for a peripheral like Nordic QSPI that was designed for quad-spi access to serial flash memories and seems to be extremely specialized, I would like to identify a non-memory device that uses quad-spi and confirm that it can be done.

To the case here: generalizing the existing API should be informed by an understanding of how the extensions would be implemented in Nordic, STM32, NXP, and anybody else that markets a product that claims to support Q(uad-)SPI.  Not doing so risks producing an API that doesn't meet people's needs, or that can't be implemented (e.g. iovec support; see also #20154), and we'd end up having to do a competing specialized API anyway.

mtahirbutt, 2020-04-20:
> The current SPI API `drivers/spi.h` allows to advertise Dual and Quad SPI support via `SPI_LINES_DUAL` and `SPI_LINES_QUAD`, but it does not provide needed means to tag direction of the each [...]

Hi, I am trying to make work quadflash with disco_l475_iot1, I am investigating quadspi in this regard. I just found update on spi.h file for incorporating SPI_LINES_QUAD, but do not know how to configure pins for CS as stm32l475 MCU has multiple pins supporting CS for QUADSPI and additionally how can I configure that with DTS file etc.  please comment, thanks

pabigot, 2020-07-07:
API 2020-07-07: Moved from triage to on-hold.  The current active approach for this class of devices is to treat them as flash devices.  Revisit if that proves to be insufficient to system needs.

manoj153, 2020-12-16:
Hi, what's the current state/decision of Zephyr to support QSPI flash?

pabigot, 2020-12-16:
Currently we continue to write flash drivers rather than attempt to support generic quad SPI operations.

manoj153, 2020-12-17:
@pabigot I see, that makes sense, In my experience, I have only used quad SPI for flash like interface only. Anyway, is there a sample for a particular where quadspi flash supported?

pabigot, 2020-12-17:
`samples/drivers/spi_flash` has checks to support `nrf52840dk_nrf52840` using the Nordic QSPI driver, which I think is the only one in tree so far.  Several things in `samples/subsys/fs` work on flash drivers of different types.

manoj153, 2020-12-17:
Thanks, I will take a look, I am interested in STM32 qspi, I am picking up zephyr now (coming with freeRTOS experience). Will be porting it for stm32 specific

pabigot, 2020-12-17:
You'll want to be looking at #25806 then.  I believe somebody else is trying to pick up that work.

MindaugasMin, 2021-01-30:
Hello, 
I am working with ST32L452 and W25Q16JV NOR flash which is used with QSPI peripheral.

I can't read the serial flash discovery parameter. Any ideas where is the problem?
I am sure that my hardware works fine because I tried to read/write/erase using LL functions. 

This is my DTS quadspi configuration:
I copied everything disco_l475_iot1.dts

```
&quadspi {
	pinctrl-0 = <&quadspi_clk_pe10 &quadspi_ncs_pe11
		     &quadspi_bk1_io0_pe12 &quadspi_bk1_io1_pe13
		     &quadspi_bk1_io2_pe14 &quadspi_bk1_io3_pe15>;	dmas = <&dma1 5 5 0x0000 0x03>;
	dma-names = "tx_rx";	status = "okay";	w25q16jv: qspi-nor-flash@0 {
		compatible = "st,stm32-qspi-nor";
		label = "W25Q16JV";
		reg = <0>;
		qspi-max-frequency = <50000000>;
		/* 128 Megabits = 16 Megabytes */
		size = <0x8000000>;
		status = "okay";		partitions {
			compatible = "fixed-partitions";
			#address-cells = <1>;
			#size-cells = <1>;			slot1_partition: partition@0 {
				label = "image-1";
				reg = <0x00000000 0x000D8000>;
			};			storage_partition: partition@D8000 {
				label = "storage";
				reg = <0x000D8000 DT_SIZE_M(7)>;
			};
		};
	};
};
```

manoj153, 2021-03-26:
hi all, I have quad flash on octopi peripheral on STM32h723xx series MCU, I wonder if thin qspi binding might work?

SpooniSpoon, 2021-04-28:
Hi @manoj153, Are you able to handle quad flash on STM32H723? I want to use quad flash too on a STM32H743.

tbursztyka, 2021-09-08:
Guys, a heads up on this one: there seems to be a few (very few) use cases where using a generic SPI controller to access a qspi flash might be necessary. (unlike existing qspi flash driver that are targeting a dedicated qspi flash controller)

So I'll probably revive #20564, once #38386 is good to be merged, and rework it relevantly.