zephyrproject-rtos/zephyr #17907: BLE Mesh when resend use GATT bearer

menglingao1997, 2019-07-31:
Hello EvenyBody:
 When I use GATT Bearer to control node with phone. The node sends the long packet at the transport layer, and then is split and sent. The mobile phone sends back an ACK response, but the ACK response does not contain all the response packets, so the node will resend the split packet it thinks is lost. The node calls the function ‘’‘bt_mesh_net_resend‘’‘ to send the retransmitted packet, but not through the GATT but only the Adv Bearer.

[image]
[image]
[image]
```C
bt_mesh_adv_send(buf, cb, cb_data); 
```

jhedberg, 2019-07-31:
How is it possible that not all packets are acknowledged? GATT is a reliable bearer without packet loss, unlike advertising. That said, I do agree it's odd behaviour that resending happens over a different bearer, so this should probably be fixed no matter what.

menglingao1997, 2019-08-01:
I agree with you that GATT is a reliable transmission bearer. There is theoretically no packet loss, but this is happening. The most critical issue now is the need to fix the problem of retransmitting data on different bearer.

closed 2019-08-14 by jhedberg

menglingao1997, 2019-08-15:
@jhedberg Thank you very much for your contribution to the open source community.
I have a question that there is a not need to do memory release here.
https://github.com/zephyrproject-rtos/zephyr/blob/70640334730b816c8c485d2cfbe77119fc3783cb/subsys/bluetooth/mesh/net.c#L736-L792
```C
        if (IS_ENABLED(CONFIG_BT_MESH_GATT_PROXY) &&
	    bt_mesh_proxy_relay(&buf->b, dst)) {
		net_buf_unref(buf);//This code should not be executed
	} else {
		bt_mesh_adv_send(buf, cb, cb_data);
	}
```
memory release is automatically performed when the appropriate response packet is received.
https://github.com/zephyrproject-rtos/zephyr/blob/70640334730b816c8c485d2cfbe77119fc3783cb/subsys/bluetooth/mesh/transport.c#L208-L215
https://github.com/zephyrproject-rtos/zephyr/blob/70640334730b816c8c485d2cfbe77119fc3783cb/subsys/bluetooth/mesh/transport.c#L170-L206

jhedberg, 2019-08-15:
@menglingao1997 I don't see any net_buf unrefs in the transport.c code you quoted. Either way, note that we're talking about a reference count here, not explicitly freeing the memory. The advertising queue owns a reference to each net_buf that's in the queue, so once a buffer has been removed from the queue it's the responsibility of adv.c to release that reference by eventually calling net_buf_unref(). If this is the last reference the buffer will be freed, but there could also be code elsewhere (e.g. transport.c) holding a second reference and so the buffer would be freed only when the last reference is removed.

jhedberg, 2019-08-15:
@menglingao1997 ah, I had to scroll down on your quote to see the net_buf_unref, I guess you were referring to this:
```
net_buf_unref(tx->seg[i]); 
```

The tx->seg[i] is its own reference to the buffer. Both the advertising queue reference and this one need to get a net_buf_unref() call.

jhedberg, 2019-08-15:
@menglingao1997 the tx->seg[i] in transport.c gets its own reference here: https://github.com/zephyrproject-rtos/zephyr/blob/70640334730b816c8c485d2cfbe77119fc3783cb/subsys/bluetooth/mesh/transport.c#L407

jhedberg, 2019-08-15:
@menglingao1997 sorry, I thought you were talking about the adv.c change, but I see you were referencing the net.c change to the resend() function. Let me take another look.

jhedberg, 2019-08-15:
@menglingao1997 yes, you're right - `bt_mesh_adv_send()` calls `net_buf_ref()` so doing the `unref()` in `resend()` is not consistent and will cause an imbalance in the reference count. Thanks for pointing this out, and sorry for my initial confusion with what you were trying to explain 😄 

There is also another issue - I'm afraid if there are callbacks provided by the sending code those will not be called in this case.

jhedberg, 2019-08-15:
> There is also another issue - I'm afraid if there are callbacks provided by the sending code those will not be called in this case.

Actually this is fine, since this is just a resend of a segment. transport.c will take of the callbacks once the segment sending procedure is complete.

jhedberg, 2019-08-15:
> > There is also another issue - I'm afraid if there are callbacks provided by the sending code those will not be called in this case.
> Actually this is fine, since this is just a resend of a segment. transport.c will take of the callbacks once the segment sending procedure is complete.

Correcting myself again: it's not fine, transport.c relies on callbacks for each segment resend as well. I'll address this in a subsequent PR.

jhedberg, 2019-08-15:
@menglingao1997 please take a look at #18285 which tries to address the issue you brought up.

PR #18221 (2019-08-14): Bluetooth: Mesh: Fix resending segments on correct bearer
The GATT bearer is a reliable one so there should theoretically never
be a need to resend segments. If however for some strange reason the
proxy client doesn't immediately ack all segments we should do the
resending on the GATT bearer, rather than sending them over
advertising.

Fixes #17907
commit 1c7b6688 (2019-08-14): Bluetooth: Mesh: Fix resending segments on correct bearer

The GATT bearer is a reliable one so there should theoretically never
be a need to resend segments. If however for some strange reason the
proxy client doesn't immediately ack all segments we should do the
resending on the GATT bearer, rather than sending them over
advertising.

Fixes #17907
PR #18285 (2019-08-15): Bluetooth: Mesh: Fix bt_mesh_net_resend()
PR #18221 was incorrect in the buffer reference count handling and incomplete in that it lacked calling the send callbacks. This PR tries to address this. See the discussion started by @menglingao1997 at the end of #17907 for details.

Fixes #17907