zephyrproject-rtos/zephyr #17920: Bluetooth: Security problem

MarekPieta, 2019-07-31:
**Describe the bug**
`bt_conn_cb` - `security_changed` is called with requested security level even if establishing security fails. Information about error may only be found in logs.

**To Reproduce**
Connecting peripheral and central, establishing security (`bt_conn_security(conn, BT_SECURITY_MEDIUM)`). Even if security fails, application receives security changed callback (level 2). Information about error is visible only in logs.

**Expected behavior**
Application should not get `security_changed `callback if security establishing fails. It should be informed that security establishment failed.

**Impact**
Bluetooth security is not working properly.

**Screenshots or console output**

Central:
```
[00026327] <err> bt_smp: Unable to get keys for f9:12:69:88:ef:8a (random)
[00026332] <inf> ble_state: Security with f9:12:69:88:ef:8a (random) level 2
...
[00026351] <wrn> bt_smp: Unexpected SMP code 0x0a
```

Peripheral:
```
[01127946] <inf> ble_adv: Advertising stopped
...
[01149322] <inf> ble_state: Security with f6:7d:a2:fa:f6:a8 (random) level 2
...
[01152555] <err> bt_smp: reason 0x8
[01152674] <wrn> bt_smp: Unexpected SMP code 0x05
```

MarekPieta, 2019-07-31:
@carlescufi, @joerchan

joerchan, 2019-08-05:
@MarekPieta This looks like what was discussed here: https://github.com/zephyrproject-rtos/zephyr/pull/17306#issuecomment-511817499
Could you have a look and confirm?

MarekPieta, 2019-08-08:
@joerchan, additional callback could be a solution for this one (if called properly).
I think that even not calling `securiy_changed` callback here would be enough.

pdunaj, 2019-08-08:
Hi @ioannisg , this looks as high priority bug as it affects security of the link.

ioannisg, 2019-08-09:
> 
> 
> Hi @ioannisg , this looks as high priority bug as it affects security of the link.

Discussed with @carlescufi ; seems that the application _does_ get some notification about the issue, that's why we set it to LOW, but no strong opinion on this

pdunaj, 2019-08-09:
Hi @ioannisg , application is notified that security was set to medium while security setup failed which can be seen in the logs).

joerchan, 2019-08-14:
@pdunaj Yes that seems correct.
I noticed it even gets two different security levels based on where the pairing failure happens.

See attached picture.
Both devices receives pairing failed.
One device receives security changed level 2.
One device receives security changed level 4.

[image]

joerchan, 2019-08-14:
@jhedberg @Vudentz @pdunaj 
Seems that there are these issues regarding security_changed (I'm including #17013 and it's PR in this).

Issue: Pairing failed callback with security_changed security level 2
---
Receive Encryption change event with encryption success, but failed to allocate key storage (because PAIR_MAX has been reached).
Gives pairing failed after security_changed callback with encryption set, i.e security_changed level 2.

Suggested fix: 
 - Check keys in slave before sending security request or pairing response.
 -  Check keys in master before sending pairing request.
 -  If failed to get keys on encryption_change event, then disconnect with authentication failure.

Issue: Pairing failed callback with security_changed security level 2, 3 or 4.
----
Receiving pairing failed SMP packet during Transport Specific Key distribution phase.
Gives pairing_failed callback after security_changed with encryption set and authenticated keys set. i.e security_changed level 2, 3 or 4.

Suggested fix: 
 - Disconnect link with authentication failed if receiving SMP error during Key distribution phase.

Issue: No connection callback to indicate security failed.
----
Only pairing failed which is auth callback and can only have one instance
Suggested fix: 
1. Give security_changed callback with current security level.
2. Give (new) security_failed callback with reason

Issue: No reason given for security/pairing failed.
----
Suggested fix:
1. Add pairing failed reason in pairing_failed callback.
2. Give (new) security_failed callback with reason.

The last two Issues are perhaps the same issue.
Use suggested fix "Give (new) security_failed callback with reason" to solve both.

Action
----
I suggest the following fixes.
 - Check keys in slave before sending security request or pairing response.
 - Check keys in master before sending pairing request.
 - If failed to get keys on encryption_change event, then disconnect with authentication failure.
 - Disconnect link with authentication failed if receiving SMP error during Key distribution phase.
 - Give (new) security_failed callback with reason.

Any objections or better solutions?

jhedberg, 2019-08-14:
@joerchan these all sound good to me. I'd be curious if these are in line with how other stacks, e.g. SoftDevice, behave. In particular the decision to disconnect the link has me thinking a bit, since that impacts all service level connections on top of the link, regardless of their security requirements - IIRC with BlueZ only the L2CAP socket that requested the higher level security will get a failure, whereas others satisfied with lower security are allowed to keep operating as usual.

joerchan, 2019-08-14:
@jhedberg With the SoftDevice it seems I get the same behaviour for the case where peer sends Pairing failed during key dist phase.
```
2019-08-14 14:46:31.012000   --   ConnSecUpdate Event Security Mode1 Level: 2, key size: 16
2019-08-14 14:46:31.113000   --   Auth Status Event - Conn Handle: 0, Status: 0x88 SM1_level: 0b0000 Role: 33
```

By doing the first points 2 we should avoid getting into the situation for 3 and 4. If 3 still happens then I would think that is an implementation error. Maybe assert in this case?
If point 4 happens then I would think that is an implementation error of the peer. If we keep the implementation, the application is notified that security level has increased (which technically it has) but the pairing procedure has not finished with success. Maybe that is good enough?

joerchan, 2019-08-19:
@carlescufi Can you give your opinions on this one?

carlescufi, 2019-08-19:
>Application should not get security_changed callback if security establishing fails. It should be informed that security establishment failed.

There are 2 cases here:

1. Security was successfully set to the requested level, but key distribution/bonding failed after (your case) and in that case it makes all the sense that the app gets a `security_changed()` with the new security level, since the connection security did indeed change
2. For whatever reason, the security could not be updated, and in that case I suggest issuing `security_changed()` with the same sec level as before the operation. There is more discussion about this here: https://github.com/zephyrproject-rtos/zephyr/pull/17306#issuecomment-522646949

PR #17306 (2019-08-26): Bluetooth: SMP: Handle errors during pairing and encryption
Give pairing failed reason in the pairing_failed callback.
Provide pairing_failed and pairing_complete callback for SSP
Fixes: #17013

Stop pairing procedure in the request phase, instead of the key distribution phase. This solves one instance where we would end up with pairing failed after the link has been encrypted.
Provide a callback for encrypting the link has failed. This gives the application feedback when a bond exists (or assumed to exists) for the connected peer, but the peer does not have the bond information anymore.
Fixes [... cut]
closed 2019-08-26 by carlescufi