zephyrproject-rtos/zephyr #17926: device tree error: zephyrproject/zephyr/dts/bindings/can/microchip,mcp2515.yaml (in 'reg'): 'category' from !included file overwritten ('required' replaced with 'optional')

aep, 2019-07-31:
i'm trying to get samples/drivers/CAN working with the nrf52

apparantly you need to enable CAN explicitly in menuconfig, which is already confusing because   prj.conf has CONFIG_CAN=y , but by default all i get is "CAN: Device driver not found."

after enabling the driver, the device tree overlay doesnt seem to be compiled in

```
can_mcp2515.c:610:14: error: 'DT_INST_0_MICROCHIP_MCP2515_BUS_NAME' undeclared here (not in a function)Code
```

however, the mcp2515-dts.overlay file seems broken, because if i add it via

```
set(DTC_OVERLAY_FILE "${CMAKE_CURRENT_SOURCE_DIR}/mcp2515-dts.overlay")
```

in CMakeLists.txt  i get:

```
nrf52_pca10040.dts.pre.tmp:179.22-187.5: ERROR (phandle_references): /soc/spi@40004000: Reference to non-existent node or label "gpioa"
```

nrf's are apparantly called gpio0 instead of gpioa, so i changed that, but that just leads to an even more cryptic error

```
device tree error: zephyr/dts/bindings/can/microchip,mcp2515.yaml (in 'reg'): 'category' from !included file overwritten ('required' replaced with 'optional')Code
``

so i edited scripts/dts/edtlib.py and just uncommented that check, which leads me to:

```
device tree error: 'sjw' is marked as required in 'properties:' in /home/aep/proj/zephyr/zephyrproject/zephyr/dts/bindings/can/microchip,mcp2515.yaml, but does not appear in <Node /soc/spi@40004000/mcp2515@0 in 'nrf52_pca10040.dts.pre.tmp'>
``

so i after removing the broken property, i finally have a working CAN driver.

i would upstream that, but its probably wrong

```
diff --git a/dts/bindings/can/can.yaml b/dts/bindings/can/can.yaml
index 628948e5ed..dfb243a9f6 100644
--- a/dts/bindings/can/can.yaml
+++ b/dts/bindings/can/can.yaml
@@ -12,31 +12,31 @@ child:
 properties:
     "#address-cells":
       type: int
-      category: required
+      category: optional
       description: should be 1.
     "#size-cells":
       type: int
-      category: required
+      category: optional
       description: should be 0.
     label:
-      category: required
+      category: optional
     bus-speed:
       type: int
-      category: required
+      category: optional
       description: bus speed in Baud/s
     sjw:
       type: int
-      category: required
+      category: optional
       description: Resynchronization jump width (ISO 11898-1)
     prop-seg:
       type: int
-      category: required
+      category: optional
       description: Time quantums of propagation segment (ISO 11898-1)
     phase-seg1:
       type: int
-      category: required
+      category: optional
       description: Time quantums of phase buffer 1 segment (ISO 11898-1)
     phase-seg2:
       type: int
-      category: required
+      category: optional
       description: Time quantums of phase buffer 2 segment (ISO 11898-1)
diff --git a/samples/drivers/CAN/CMakeLists.txt b/samples/drivers/CAN/CMakeLists.txt
index 1ec55878dc..68c1f52668 100644
--- a/samples/drivers/CAN/CMakeLists.txt
+++ b/samples/drivers/CAN/CMakeLists.txt
@@ -2,6 +2,7 @@
 
 cmake_minimum_required(VERSION 3.13.1)
 
+set(DTC_OVERLAY_FILE "${CMAKE_CURRENT_SOURCE_DIR}/mcp2515-dts.overlay")
 
 include($ENV{ZEPHYR_BASE}/cmake/app/boilerplate.cmake NO_POLICY_SCOPE)
 project(CAN)
diff --git a/samples/drivers/CAN/mcp2515-dts.overlay b/samples/drivers/CAN/mcp2515-dts.overlay
index 899172080b..6a374ab2a2 100644
--- a/samples/drivers/CAN/mcp2515-dts.overlay
+++ b/samples/drivers/CAN/mcp2515-dts.overlay
@@ -6,12 +6,12 @@
 
 &spi1 {
        status = "okay";
-       cs-gpios = <&gpioa 4 GPIO_DIR_OUT>;
+       cs-gpios = <&gpio0 4 GPIO_DIR_OUT>;
        mcp2515@0 {
                compatible = "microchip,mcp2515";
                spi-port-name = "SPI_1";
                spi-max-frequency = <1000000>;
-               int-gpios = <&gpioa 0 GPIO_INT_ACTIVE_LOW>;
+               int-gpios = <&gpio0 0 GPIO_INT_ACTIVE_LOW>;
                status = "okay";
                label = "CAN_1";
                reg = <0>;
diff --git a/scripts/dts/edtlib.py b/scripts/dts/edtlib.py
index 485a8865fb..7722a8a5c6 100644
--- a/scripts/dts/edtlib.py
+++ b/scripts/dts/edtlib.py
@@ -1114,8 +1114,7 @@ def _bad_overwrite(to_dict, from_dict, prop):
 
     # Allow a property to be made required when it previously was optional
     # without a warning
-    if prop == "category" and to_dict["category"] == "required" and \
-                              from_dict["category"] == "optional":
+    if prop == "category":
         return False
 
     return True

```

ulfalizer, 2019-07-31:
CC @karstenkoenig

jukkar, 2019-08-01:
> ```diff
> +set(DTC_OVERLAY_FILE "${CMAKE_CURRENT_SOURCE_DIR}/mcp2515-dts.overlay")
> ```

I think the DTS overlay file is suppose to be set in cmake command line so that user can override it easily.

karstenkoenig, 2019-08-01:
the overlay file can be added with -DOVERLAY_FILE=/home/foo/bar/whatever-dts.overlay to the cmake call, it seemed to be the way to go when I've added this file to the sample. As the mcp2515 is an external controller it needs to be 'added' to the device tree by the user and can not just be hardcoded to the MCU or the board file :-| 
That's also why it needs to be adjusted for the nrf52 with different gpio device names, I've initially developed it on an STM32.
Same with the project file, it can be set via  -DCONF_FILE=prj.conf.mcp2515 in the cmake call. You should not have needed to configure via menuconfig.
https://docs.zephyrproject.org/latest/application/index.html#important-build-system-variables

In general I tried making it a bit more streamlined in this PR
https://github.com/zephyrproject-rtos/zephyr/pull/15658/commits/726230d8c02194b8b48012bcaf4dced0042dfbb3
but abandoned it, also because the sample seemed to be facing bigger changes anyways. Please have a look there, it moves the project file to a boards folder as recommended to a boards subfolder.

Not so sure about those other DTS changes migrating them from required to optional.

karstenkoenig, 2019-08-05:
Hi @aep ,
I found the issues in the device tree file and think I resolved it more cleanly in my PR here
https://github.com/zephyrproject-rtos/zephyr/pull/15658

In particular this commit, it adds the required fields (which are enforced nowadays :-) ) 
https://github.com/zephyrproject-rtos/zephyr/pull/15658/commits/accea6c9f84e142ff7e6d16cdaf7d9cfc30ebbb8

It also transform the overlay into a shield, so you can just add it to your west call
cmake  -S/home/karsten/zephyrproject/zephyr/samples/drivers/CAN -GNinja -DSHIELD=mcp2515 -DBOARD=nrf52_pca10040

It builds and works cleanly on my nrf52832 dk

aep, 2019-08-06:
@karstenkoenig thanks. unfortunately i don't know how to test this. west is confusing.

```
$ west init -m https://github.com/karstenkoenig/zephyr.git --mr sample_can_cleanup

b2ebf1e84824d88a9da7bffff9544ddeadd1c91d refs/remotes/origin/master
Branch 'master' set up to track remote branch 'master' from 'origin'.
Already on 'master'
usage: west [-h] [-z ZEPHYR_BASE] [-v] [-V] <command> ...
west: error: argument <command>: invalid choice: 'post-init' (choose from 'init', 'update', 'list', 'manifest', 'diff', 'status', 'forall', 'config', 'selfupdate', 'help')

```

karstenkoenig, 2019-08-06:
Eh sorry I said west but actually also used cmake because I am still more used to it :-D 
`cmake -S/home/karsten/zephyrproject/zephyr/samples/drivers/CAN -GNinja -DSHIELD=mcp2515 -DBOARD=nrf52_pca10040`

You need to adjust source (=S) and build path (-B)

aep, 2019-08-06:
nah, i actually don't know how to check out your branch, sorry, probably basics.

karstenkoenig, 2019-08-06:
`git clone git@github.com:karstenkoenig/zephyr.git
git checkout sample_can_cleanup`

aep, 2019-08-06:
right, that would be easy :D but when not using west, the build system doesnt seem to work.

```
zephyr aep@stark: ~/kram/zephyr  cmake -S ~/kram/zephyr/samples/drivers/CAN -GNinja -DSHIELD=mcp2515 -DBOARD=nrf52_pca10040 
Zephyr version: 1.14.99
-- Found PythonInterp: /usr/sbin/python (found suitable version "3.7.4", minimum required is "3.4") 
-- Selected BOARD nrf52_pca10040
-- Found west: /home/aep/.local/bin/west (found suitable version "0.6.0", minimum required is "0.6.0")
FATAL ERROR: no west installation found from "/home/aep/kram/zephyr"; "west list" requires one.

```

karstenkoenig, 2019-08-06:
You can also add my github repo as a new remote in your original zephyr repo where you have setup west. Sorry but please try figuring that out on your own, there is plenty of documentation on either :-|

aep, 2019-08-06:
that's the first thing i tried of course, but that just gave me:

```
arning: CAN_MAX_FILTER (defined at drivers/can/Kconfig.stm32:16, drivers/can/Kconfig.loopback:22)
was assigned the value '5' but got the value ''. You can check symbol information (including
dependencies) in the 'menuconfig' interface (see the Application Development Primer section of the
manual), or in the Kconfig reference at
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_CAN_MAX_FILTER.html (which is updated
regularly from the master branch). See the 'Setting configuration values' section of the Board
Porting Guide as well.

/home/aep/proj/nrf91/zephyr/.checkpatch.conf:1: warning: ignoring malformed line '--mailback'

/home/aep/proj/nrf91/zephyr/.checkpatch.conf:2: warning: ignoring malformed line '--emacs'

[..]

Error: Aborting due to non-whitelisted Kconfig warning
'/home/aep/proj/nrf91/zephyr/.checkpatch.conf:1: warning: ignoring malformed line '--mailback''.
Note: If this warning doesn't point to an actual problem, you can add it to the whitelist at the top
of /home/aep/proj/nrf91/zephyr/scripts/kconfig/kconfig.py.

```

so i'm assuming you have to use this west stuff, like the .. documentation says.
but no worries, sorry for the noise.

karstenkoenig, 2019-08-06:
Now THAT is an issue I actually caused :-) And I should have seen it in the shipable build check, but only looked for the one I knew had to fail.
I still had a leftover config file in my local setup that cmake just picked up, it must have correctly set this field.

Can you please try and add `CONFIG_CAN_MAX_FILTER=5` to `boards/shields/mcp2515/mcp2515.conf` and build again? I left to work before noticing this so I can't address it now :-(

galak, 2019-08-06:
How are we reproducing this?

Very confused about what platform the overlay was suppose to apply to.

karstenkoenig, 2019-08-06:
@galak 
This can be reproduced by building the CAN sample and using the provided mcp2515 dts overlay sample file. I've fixed/addressed it in https://github.com/zephyrproject-rtos/zephyr/pull/18057

alexanderwachter, 2019-08-19:
@aep Are your problems solved now? Can we close this issue?

aep, 2019-08-30:
@alexanderwachter yes, i tested the original PR and it worked. If it's merged, this can be closed. thanks!

closed 2019-08-30 by aep