zephyrproject-rtos/zephyr #17932: BLE Mesh When Friend Send Seg Message To LPN

menglingao1997, 2019-08-01:
When a **friend node** needs to send **large message**, it is reasonable to send all the segment message of high transport layer message, but as shown in the code below, **only the _first packet_ is stored in the friend queue, and then the program _terminates inexplicably_, then the _other Where did the data packet go_?**
``` C
static int send_seg(struct bt_mesh_net_tx *net_tx, struct net_buf_simple *sdu,
                    const struct bt_mesh_send_cb *cb, void *cb_data)
{
    u8_t seg_hdr, seg_o;
    u16_t seq_zero;
    struct seg_tx *tx;
    int i;

    BT_DBG("src 0x%04x dst 0x%04x app_idx 0x%04x aszmic %u sdu_len %u",
           net_tx->src, net_tx->ctx->addr, net_tx->ctx->app_idx,
           net_tx->aszmic, sdu->len);

    if (sdu->len < 1)
    {
        BT_DBG("Zero-length SDU not allowed");
        return -EINVAL;
    }

    if (sdu->len > BLE_MESH_TX_SDU_MAX)
    {
        BT_DBG("Not enough segment buffers for length %u", sdu->len);
        return -EMSGSIZE;
    }

    for (tx = NULL, i = 0; i < ARRAY_SIZE(seg_tx); i++)
    {
        if (!seg_tx[i].nack_count)
        {
            tx = &seg_tx[i];
            break;
        }
    }

    if (!tx)
    {
        BT_DBG("No multi-segment message contexts available");
        return -EBUSY;
    }

    if (net_tx->ctx->app_idx == BLE_MESH_KEY_DEV)
    {
        seg_hdr = SEG_HDR(0, 0);
    }
    else
    {
        seg_hdr = SEG_HDR(1, net_tx->aid);
    }

    seg_o = 0U;
    tx->dst = net_tx->ctx->addr;
    tx->seg_n = (sdu->len - 1) / 12U;
    tx->nack_count = tx->seg_n + 1;
    tx->seq_auth = SEQ_AUTH(BLE_MESH_NET_IVI_TX, bt_mesh.seq);
    tx->sub = net_tx->sub;
    tx->new_key = net_tx->sub->kr_flag;
    tx->cb = cb;
    tx->cb_data = cb_data;

    if (net_tx->ctx->send_ttl == BLE_MESH_TTL_DEFAULT)
    {
        tx->ttl = bt_mesh_default_ttl_get();
    }
    else
    {
        tx->ttl = net_tx->ctx->send_ttl;
    }

    seq_zero = tx->seq_auth & 0x1fff;

    BT_DBG("SeqZero 0x%04x", seq_zero);

    for (seg_o = 0U; sdu->len; seg_o++)
    {
        struct net_buf *seg;
        u16_t len;
        int err;

        seg = bt_mesh_adv_create(BLE_MESH_ADV_DATA, net_tx->xmit,
                                 BUF_TIMEOUT);
        if (!seg)
        {
            BT_DBG("Out of segment buffers");
            seg_tx_reset(tx);
            return -ENOBUFS;
        }

        BLE_MESH_ADV(seg)->seg.attempts = SEG_RETRANSMIT_ATTEMPTS;

        net_buf_reserve(seg, BLE_MESH_NET_HDR_LEN);

        net_buf_add_u8(seg, seg_hdr);
        net_buf_add_u8(seg, (net_tx->aszmic << 7) | seq_zero >> 6);
        net_buf_add_u8(seg, (((seq_zero & 0x3f) << 2) |
                             (seg_o >> 3)));
        net_buf_add_u8(seg, ((seg_o & 0x07) << 5) | tx->seg_n);

        len = MIN(sdu->len, 12);
        net_buf_add_mem(seg, sdu->data, len);
        net_buf_simple_pull(sdu, len);

        tx->seg[seg_o] = net_buf_ref(seg);

        if (IS_ENABLED(CONFIG_BLE_MESH_FRIEND))
        {
            enum bt_mesh_friend_pdu_type type;

            if (seg_o == tx->seg_n)
            {
                type = BLE_MESH_FRIEND_PDU_COMPLETE;
            }
            else
            {
                type = BLE_MESH_FRIEND_PDU_PARTIAL;
            }

            if (bt_mesh_friend_enqueue_tx(net_tx, type, &tx->seq_auth, &seg->b) && BLE_MESH_ADDR_IS_UNICAST(net_tx->ctx->addr))
            {
                /* PDUs for a specific Friend should only go
                 * out through the Friend Queue.
                 */
                net_buf_unref(seg);
                return 0;
            }
        }

        BT_DBG("Sending %u/%u", seg_o, tx->seg_n);

        err = bt_mesh_net_send(net_tx, seg,
                               seg_o ? &seg_sent_cb : &first_sent_cb,
                               tx);
        if (err)
        {
            BT_DBG("Sending segment failed");
            seg_tx_reset(tx);
            return err;
        }
    }

    if (IS_ENABLED(CONFIG_BLE_MESH_LOW_POWER) &&
        bt_mesh_lpn_established())
    {
        bt_mesh_lpn_poll();
    }

    return 0;
}
```

aescolar, 2019-08-01:
@menglingao1997 Thanks for this bug report.

When filling bug reports, please try to follow the bug report template.

Instead of copying Zephyr code into the report link to the file, line and version where you observe the issue. In this case:
https://github.com/zephyrproject-rtos/zephyr/blob/583226a83328b00f385bdf6c11e2f0df0a1453be/subsys/bluetooth/mesh/transport.c#L299-L427

When referring to your terminal output, if possible, instead of pasting screenshots, try to paste the text itself.

This way it should be easier for others to understand the problem and handle it appropriately.

jhedberg, 2019-08-01:
Thanks for the report. It seems you're right - the `return 0;` should in fact be a `continue;`. Are you able to send a pull request for it?

commit 680d778f (2019-08-14): Bluetooth: Mesh: Fix not sending all segments through the Friend Queue

The code was incorrectly bailing out with "return 0" rather than
continue. Also, it was incorrectly making a reference to
tx->seg[seg_o] since when a PDU goes through the friend queue we don't
use the usual retransmission mechanism.

Fixes: #17932
closed 2019-08-14 by jhedberg