zephyrproject-rtos/zephyr #17949: stm32 i2c driver has problems with AHB_PRESCALER, APB1_PRESCALER, APB2_PRESCALER

StefJar, 2019-08-01:
I am running Zephyr on my own board. This is based on the stm32f412.

After having problems with setting up the AHB prescalors, I am now ran into I2C problems.

@erwango fixed the AHB prescalors problems in: https://github.com/zephyrproject-rtos/zephyr/pull/17938

Now having this fix put into action I am facing I2C errors.

it comes down that:

```
CONFIG_CLOCK_STM32_AHB_PRESCALER=1
CONFIG_CLOCK_STM32_APB1_PRESCALER=8
CONFIG_CLOCK_STM32_APB2_PRESCALER=8
```
fails
but
```
CONFIG_CLOCK_STM32_AHB_PRESCALER=2
CONFIG_CLOCK_STM32_APB1_PRESCALER=4
CONFIG_CLOCK_STM32_APB2_PRESCALER=4
```
works

The rest cfg is set to:

```
CONFIG_I2C=y
CONFIG_I2C_STM32=y
CONFIG_I2C_STM32_V1=y
# CONFIG_I2C_STM32_INTERRUPT=y
CONFIG_I2C_1=y

CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=96000000

CONFIG_CLOCK_CONTROL=y

# Clock configuration for Cube Clock control driver
CONFIG_CLOCK_STM32_HSE_CLOCK=24000000
CONFIG_CLOCK_STM32_SYSCLK_SRC_PLL=y

# use HSE as PLL input
CONFIG_CLOCK_STM32_PLL_SRC_HSE=y

# the board have an external oscillator
C
__udivmoddi4() at libgcc2.c:1,024 0x80002d8	
__aeabi_uldivmod() at bpabi.S:258 0x80002c4	
z_impl_k_busy_wait() at sys_clock.h:50 0x8011650	
k_busy_wait() at kernel.h:25 0x80160e2	
stm32_i2c_wait_timeout() at i2c_ll_stm32_v1.c:311 0x80160e2	
stm32_i2c_msg_write() at i2c_ll_stm32_v1.c:335 0x800f81a	
i2c_stm32_transfer() at i2c_ll_stm32.c:145 0x80161a8	
z_impl_i2c_transfer() at i2c.h:252 0x8011ece	
i2c_transfer() at i2c.h:15 0x8011ece	
i2c_write_read() at i2c.h:454 0x8011ece	
i2c_reg_read_byte() at i2c.h:541 0x8011ece	ONFIG_CLOCK_STM32_HSE_BYPASS=n

# produce 96MHz clock at PLL output
CONFIG_CLOCK_STM32_PLL_M_DIVISOR=12
CONFIG_CLOCK_STM32_PLL_N_MULTIPLIER=96
CONFIG_CLOCK_STM32_PLL_P_DIVISOR=2
CONFIG_CLOCK_STM32_PLL_Q_DIVISOR=4
```

In the error case I get:

`i2c_ll_stm32_v1.msg_end: stm32_i2c_msg_write: ERR 1`

The stack trace look like:
```
__udivmoddi4() at libgcc2.c:1,024 0x80002d8	
__aeabi_uldivmod() at bpabi.S:258 0x80002c4	
z_impl_k_busy_wait() at sys_clock.h:50 0x8011650	
k_busy_wait() at kernel.h:25 0x80160e2	
stm32_i2c_wait_timeout() at i2c_ll_stm32_v1.c:311 0x80160e2	
stm32_i2c_msg_write() at i2c_ll_stm32_v1.c:335 0x800f81a	
i2c_stm32_transfer() at i2c_ll_stm32.c:145 0x80161a8	
z_impl_i2c_transfer() at i2c.h:252 0x8011ece	
i2c_transfer() at i2c.h:15 0x8011ece	
i2c_write_read() at i2c.h:454 0x8011ece	
i2c_reg_read_byte() at i2c.h:541 0x8011ece	
```

StefJar, 2019-08-01:
enabling the isr

`CONFIG_I2C_STM32_INTERRUPT=y`

solves the error with

```
CONFIG_CLOCK_STM32_AHB_PRESCALER=1
CONFIG_CLOCK_STM32_APB1_PRESCALER=8
CONFIG_CLOCK_STM32_APB2_PRESCALER=8
```

erwango, 2019-08-02:
Tested I2C(w/o interrupts) ok on nucleo_f412zg with similar clock settings.
Difference would be the use of external HSE, bypassing the HSE Oscillator.

erwango, 2019-11-01:
@StefJar , is this issue still valid?

StefJar, 2019-11-04:
> @StefJar , is this issue still valid?

thx for asking. I don't see this issue any longer. So I am closing it

closed 2019-11-04 by StefJar