zephyrproject-rtos/zephyr #17956: Is POSIX I/O supported on peripheral?

qiutongs, 2019-08-01:
**Is your feature request related to a problem? Please describe.**
I want to know if zephyr supports POSIX I/O on any peripherals, like UART, I2C, SPI.
From what I find in the source code, it seems zephyr only has POSIX I/O for file system but not peripherals. Can you confirm this in case I miss anything?

**Describe the solution you'd like**
If what I find is true, is there any plan that zephyr will support POSIX I/O on peripherals?

**Describe alternatives you've considered**
N/A

**Additional context**
My reference: https://github.com/zephyrproject-rtos/zephyr/tree/master/lib/posix

andrewboie, 2019-08-01:
Currently there is no generic file descriptor access to Zephyr peripherals, you need to use the driver subsystem APIs to interact with them.

I know of no current plans to change this.

qiutongs, 2019-08-02:
Thanks for confirmation.

closed 2019-08-02 by qiutongs