zephyrproject-rtos/zephyr #17965: kernel/sleep/ test failed on reel_board.

chen-png, 2019-08-02:
**Describe the bug**
Running test suite sleep on reel_board,  it failed with error message "elapsed_ms <= UPPER_BOUND_MS is failed, overslept".

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b reel_board -p=auto tests/kernel/sleep/
2. west flash

**Screenshots or console output**
[image]

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-SDK-0.10.1
 - Commit ID: 14db558939db

jfischer-phytec-iot, 2019-08-12:
Weird, I tried it on nrf52840_pca52840 and I get the same result if I flash it with `west flash -r jlink`:

```
***** Booting Zephyr OS build zephyr-v1.14.0-3134-g4524035bc8d5 *****
Running test suite sleep
===================================================================
starting test - test_sleep
Kernel objects initialized
Test thread started: id = 0x2000014c
Helper thread started: id = 0x20000088
Testing normal expiration of k_sleep()
Testing: test thread sleep + helper thread wakeup test
Testing: test thread sleep + isr offload wakeup test
Testing: test thread sleep + main wakeup test thread
Testing kernel k_sleep()
PASS - test_sleep
===================================================================
starting test - test_usleep
elapsed_ms = 2500

    Assertion failed at ZEPHYR_BASE/tests/kernel/sleep/src/usleep.c:67: test_usleep: (elapsed_ms <= UPPE)
overslept
FAIL - test_usleep
===================================================================
Test suite sleep failed.
===================================================================
PROJECT EXECUTION FAILED
```

But it works fine on reel board and nrf52840_pca10056 if I update the firmware using MSD support (e.g. pmount /dev/sdc; cp zephyr/zephyr.bin /media/sdc && pumount /dev/sdc):

```

***** Booting Zephyr OS build zephyr-v1.14.0-3134-g4524035bc8d5 *****
Running test suite sleep
===================================================================
starting test - test_sleep
Kernel objects initialized
Test thread started: id = 0x2000014c
Helper thread started: id = 0x20000088
Testing normal expiration of k_sleep()
Testing: test thread sleep + helper thread wakeup test
Testing: test thread sleep + isr offload wakeup test
Testing: test thread sleep + main wakeup test thread
Testing kernel k_sleep()
PASS - test_sleep
===================================================================
starting test - test_usleep
elapsed_ms = 1500
PASS - test_usleep
===================================================================
Test suite sleep succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

carlescufi, 2019-08-12:
@pizi-nordic could you take a look on our DevKit?

pizi-nordic, 2019-08-13:
I reproduced that bug on nrf52840_pca10056. I will look at it tomorrow.

jfischer-phytec-iot, 2019-08-14:
update: the test passes as long the debug probe is connected to the target, to reproduce on reel board, execute `pyocd commander -t nrf52840`, and press reset button or `set nreset 0; set nreset 1`.

pabigot, 2019-08-16:
I can confirm that when it passes the loop iteration takes 3 ticks.  When it fails it takes 5.

#18343 provides a fix for the test and explanation that's consistent with the documented test design, but I have no idea why the clock behavior is so different depending on whether the debug probe is attached or not.

Further instrumentation shows that when the failure occurs the line below is evaluated when `next == 2`: https://github.com/zephyrproject-rtos/zephyr/blob/6abbed2975a0a3400658819a051e2ae81700c342/kernel/timeout.c#L163

The effect is that the intent to bypass setting an alarm when one is already set and "soon enough" fails, because the invoked set code takes the path through this line:
https://github.com/zephyrproject-rtos/zephyr/blob/6abbed2975a0a3400658819a051e2ae81700c342/drivers/timer/nrf_rtc_timer.c#L167

and the scheduled timeout gets advanced.

@pizi-nordic @andyross I've confirmed that an alternative fix is to change the `imminent` test to check `next <= 2`.  This allows the original version of kernel/sleep/test to pass in the conditions where it fails (the loop iteration takes 4 ticks).

However, to do that is a more invasive change that involves determining a target-specific value for the imminent comparison.  It may be worth considering, though, as it could also be the underlying cause of ongoing latency issues that are blocking closure of #14577.

pizi-nordic, 2019-08-19:
Thank you @pabigot!

pabigot, 2019-08-19:
In the successful case where the iteration body takes 3 ticks, generated by `west flash`:

* The timeout is set for 2 cycles successfully.
* z_set_timeout_expiry() on entry to idle sees the next timeout due in 1 tick, which is imminent, so the compare is not reprogrammed.
* At the end of the announce releasing the sleep the clock is set for more than 2 cycles.

In the failure case where the iteration body is longer, generated by `west flash -r jlink`:

* The timeout is set for 2 cycles successfully.
* z_set_timeout_expiry() on entry to idle sees the next timeout due in 2 ticks, which is not imminent, so the compare is reprogrammed.

There are two variant failure modes.  The most common is:

* z_clock_set_timeout() sets to 2, and the check hits the `Too soon, interrupt won't arrive.` path which reprograms it to 2.  The iteration duration in this path is 5 ticks.

Another is:

* z_clock_set_timeout() sets to 2, and the check hits the `Missed it!` path which forces a timer ISR at point within the middle of a tick. `z_clock_announce()` then makes another attempt at scheduling.  The iteration duration in this path is 6 ticks.

This second path showed up as a result of increased instrumentation, showing that this whole thing is very sensitive to timing.

Any given run may include iterations that follow the three paths described above, but runs started with `-r jlink` are likely to have more failure paths.  The root cause is perhaps JLink presence affecting the stability of the RTC or CPU clocks and so affecting whether the idle loop starts within 1 tick of the deadline.

All very confusing, and worth looking into further though I believe #18343 is sufficient to close this issue.

For future analysis [this gist](https://gist.github.com/pabigot/60d82b2f1633e47d905d644b8104e56c) has a patch relative to 561e6ab7a7e16ae8eb2ea78f93aadbcaaa0c2f4d that has my instrumentation code for nrf52840_pca10056 using a Logic 8 Pro sampling at 250 MHz, along with a compressed capture file showing the behaviors.  You may need Logic version 1.2.29 Beta to read the file.

pabigot, 2019-08-19:
Per @jfischer-phytec-iot this fixes it for the new reel board, but the old one is failing to satisfy the maximum 2500 ms upper bound, getting values like 2625 that suggest about 25% of the iterations are taking the path that requires 6 ticks.  I'm still inclined to think this is associated with crystal stability.  If that seems plausible, I'll update the `MINIMUM_CYCLE_TICKS` value in #18343 to 3, and probably rename it to `MAX_SHORTEST_DELAY_TICKS` since it's clearly not a constant.

pizi-nordic, 2019-08-20:
Thank you @pabigot for looking into the problem in detail.

PR #18343 (2019-08-23): tests/kernel/sleep: relax upper bound to support additional delays
Nordic platforms measure ticks in cycles of a 32 KiHz clock for which
the minimum compare delay is 2 ticks.  The test assumed an upper bound
of four ticks delay per loop iteration, resulting from alignment at
various layers.  This delay is met on Nordic for tick rates at or below
16384, but is too short for the default 32768 Hz tick rate.

Instrumentation confirms that the usleep test loop body on Nordic at 32
KiHz ticks takes 153 ms which is 5 ticks, and that the requested 1 tick
timeout delay is being increased to 2 ticks to  [... cut]
commit 243ea2a1 (2019-08-23): tests/kernel/sleep: relax upper bound to support additional delays

Nordic platforms measure ticks in cycles of a 32 KiHz clock for which
the minimum compare delay is 2 ticks.  The test assumed an upper bound
of four ticks delay per loop iteration, resulting from alignment at
various layers.  This delay is met on Nordic for tick rates at or below
16384, but is too short for the default 32768 Hz tick rate.

Instrumentation confirms that the usleep test loop body on Nordic at 32
KiHz ticks takes 3 ticks as the optimum delay, only when a debug probe
is active.  In other circumstances it can take  [... cut]
closed 2019-08-23 by ioannisg