zephyrproject-rtos/zephyr #17967: drivers/pwm/pwm_api test failed on frdm_k64f board.

chen-png, 2019-08-02:
**Describe the bug**
Running suite pwm_basic_test on frdm_k64f board, it failed on the pwm_cycle test, we can see the error message "Fail to set the period and pulse width".

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b frdm_k64f -p=auto tests/drivers/pwm/pwm_api/
2. west flash

**Screenshots or console output**
[image]

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-SDK-0.10.1
 - Commit ID: 14db558939d

MaureenHelm, 2019-08-13:
Hi @chen-png , please have a look at #18203 and confirm if it fixes the issue for you.

closed 2019-08-13 by galak

chen-png, 2019-08-14:
> Hi @chen-png , please have a look at #18203 and confirm if it fixes the issue for you.

OK, the PR#18203 has fixed this issue and the tests/drivres/pwm/pwm_api test run successfully on frdm_k64f board now. Thanks.

PR #18203 (2019-08-13): tests: pwm_api: Use smaller pwm period cycles on k64-based boards
The mcux pwm drivers use period cycles as a divisor to calculate the pwm
frequency in hertz. This operation can underflow easily with large
values of period cycles relative to the pwm clock source, causing the
driver to return an error code and the pwm_api test to fail.

Updates the test to use the smaller set of period and pulse cycles on
k64-based boards, fixing the test for frdm_k64f and hexiwear_k64 boards.

The test is not changed for i.mx rt boards because the pwm clock source
on these boards is much faster than on k64 boa [... cut]
PR #18227 (2019-11-07): [Backport v1.14-branch] tests: pwm_api: Use smaller pwm period cycles on k64-based boards
Backport #18203.

Fixes #17967