zephyrproject-rtos/zephyr #17971: [zephyr 1.14] Unable to register GATT service that was unregistered before

MariuszSkamra, 2019-08-02:
**Describe the bug**
There is a bug in GATT population of services. One is unable to register service that has been unregistered and was not the last one service in database

**To Reproduce**
Steps to reproduce the behavior (btshell):
1. bt init
2. gatt metrics on
3. gatt register
4. gatt metrics off
5. gatt metrics on
6. See ```<err> bt_gatt: Unable to register handle 0x0010```

**Expected behavior**
Service is registered

**Impact**
Unable to register service

**Screenshots or console output**
```
uart:~$ bt init
Bluetooth initialized
--- 13 messages dropped ---
[00:04:32.108,200] <dbg> bt_gatt.gatt_register: attr 0x200074b0 handle 0x000b uuid 2a04 perm 0x09
[00:04:32.108,300] <dbg> bt_gatt.gatt_register: attr 0x200074c4 handle 0x000c uuid 2a04 perm 0x01
[00:04:32.108,300] <dbg> bt_gatt.gatt_register: attr 0x200074d8 handle 0x000d uuid 2a04 perm 0x01
[00:04:32.108,300] <dbg> bt_gatt.gatt_register: attr 0x200074ec handle 0x000e uuid 2a04 perm 0x01
[00:04:32.108,400] <dbg> bt_gatt.gatt_register: attr 0x20007500 handle 0x000f uuid 2a04 perm 0x01
[00:04:32.108,700] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0001 offset 0 length 2
[00:04:32.108,700] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0002 offset 0 length 5
[00:04:32.108,900] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0005 offset 0 length 5
[00:04:32.109,200] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0007 offset 0 length 5
[00:04:32.109,200] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0009 offset 0 length 2
[00:04:32.109,300] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x000a offset 0 length 5
[00:04:32.109,500] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x000c offset 0 length 5
[00:04:32.109,800] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x000e offset 0 length 5
[00:04:32.110,000] <dbg> bt_gatt: Hash: 
b2 db dc 5b 71 e5 25 37 |...[q.%7
2b e2 c9 47 ba a5 9f 69 |+..G...i
[00:04:32.110,500] <inf> bt_hci_core: Identity: dd:ad:fb:8f:b7:ae (random)
[00:04:32.110,500] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0000, manufacturer 0x05f1
[00:04:32.110,500] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0xffff
uart:~$ gatt metrics on
GATT metrics test Service on
[00:04:49.856,000] <dbg> bt_gatt.gatt_register: attr 0x200076e8 handle 0x0010 uuid 12345678-1234-5678-1234-56789abcde02 perm 0x01
[00:04:49.856,100] <dbg> bt_gatt.gatt_register: attr 0x200076fc handle 0x0011 uuid 12345678-1234-5678-1234-56789abcde02 perm 0x01
[00:04:49.856,200] <dbg> bt_gatt.gatt_register: attr 0x20007710 handle 0x0012 uuid 12345678-1234-5678-1234-56789abcde02 perm 0x03
[00:04:49.874,600] <dbg> bt_gatt.sc_process: start 0x0010 end 0x0012
[00:04:49.874,900] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0001 offset 0 length 2
[00:04:49.874,900] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0002 offset 0 length 5
[00:04:49.875,100] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0005 offset 0 length 5
[00:04:49.875,400] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0007 offset 0 length 5
[00:04:49.875,400] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0009 offset 0 length 2
[00:04:49.875,400] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x000a offset 0 length 5
[00:04:49.875,700] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x000c offset 0 length 5
[00:04:49.875,900] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x000e offset 0 length 5
[00:04:49.875,900] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0010 offset 0 length 16
[00:04:49.876,100] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0011 offset 0 length 16
[00:04:49.876,500] <dbg> bt_gatt: Hash: 
56 a0 aa 70 3c 76 22 7e |V..p<v"~
0e c3 0d a0 1f 72 46 2f |.....rF/
uart:~$ gatt register 
Registering test vendor services
[00:04:56.234,400] <dbg> bt_gatt.gatt_register: attr 0x2000778c handle 0x0013 uuid 2902 perm 0x01
[00:04:56.234,400] <dbg> bt_gatt.gatt_register: attr 0x200077a0 handle 0x0014 uuid 2902 perm 0x01
--- 17 messages dropped ---
[00:04:56.234,500] <dbg> bt_gatt.gatt_register: attr 0x200077b4 handle 0x0015 uuid 2902 perm 0x30
[00:04:56.234,500] <dbg> bt_gatt.gatt_register: attr 0x200077c8 handle 0x0016 uuid 2902 perm 0x01
[00:04:56.234,700] <dbg> bt_gatt.gatt_register: attr 0x200077dc handle 0x0017 uuid 2902 perm 0x43
[00:04:56.234,700] <dbg> bt_gatt.gatt_register: attr 0x200077f0 handle 0x0018 uuid 2902 perm 0x01
[00:04:56.234,800] <dbg> bt_gatt.gatt_register: attr 0x20007804 handle 0x0019 uuid 2902 perm 0x43
[00:04:56.234,900] <dbg> bt_gatt.gatt_register: attr 0x20007818 handle 0x001a uuid 2902 perm 0x01
[00:04:56.235,000] <dbg> bt_gatt.gatt_register: attr 0x2000782c handle 0x001b uuid 2902 perm 0x03
[00:04:56.235,100] <dbg> bt_gatt.gatt_register: attr 0x20007730 handle 0x001c uuid 2902 perm 0x01
[00:04:56.235,100] <dbg> bt_gatt.gatt_register: attr 0x20007744 handle 0x001d uuid 2902 perm 0x01
[00:04:56.235,200] <dbg> bt_gatt.gatt_register: attr 0x20007758 handle 0x001e uuid 2902 perm 0x02
[00:04:56.254,500] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x001a offset 0 length 16
[00:04:56.254,700] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x001c offset 0 length 16
[00:04:56.255,000] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x001d offset 0 length 16
[00:04:56.255,400] <dbg> bt_gatt: Hash: 
1b 78 95 45 ce 32 c2 c2 |.x.E.2..
fd 03 34 d1 fd 3f c7 53 |..4..?.S
uart:~$ gatt metrics off
GATT metrics test Service off
--- 1 messages dropped ---
[00:05:02.704,200] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0001 offset 0 length 2
[00:05:02.704,200] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0002 offset 0 length 5
[00:05:02.704,400] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0005 offset 0 length 5
[00:05:02.704,700] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0007 offset 0 length 5
[00:05:02.704,700] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0009 offset 0 length 2
[00:05:02.704,700] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x000a offset 0 length 5
[00:05:02.704,900] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x000c offset 0 length 5
[00:05:02.705,200] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x000e offset 0 length 5
[00:05:02.705,200] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0013 offset 0 length 16
[00:05:02.705,400] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0014 offset 0 length 16
[00:05:02.705,800] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0016 offset 0 length 16
[00:05:02.706,100] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x0018 offset 0 length 16
[00:05:02.706,300] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x001a offset 0 length 16
[00:05:02.706,500] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x001c offset 0 length 16
[00:05:02.706,900] <dbg> bt_gatt.bt_gatt_attr_read: handle 0x001d offset 0 length 16
[00:05:02.707,400] <dbg> bt_gatt: Hash: 
14 f5 5d 93 41 66 d0 bc |..].Af..
dc ee 47 54 64 61 7a 45 |..GTdazE
uart:~$ gatt metrics on
[00:05:07.157,200] <err> bt_gatt: Unable to register handle 0x0010
```

**Environment (please complete the following information):**
 - nrf52840

**Additional context**
Add any other context about the problem here.

MariuszSkamra, 2019-08-08:
The problem still exists with https://github.com/zephyrproject-rtos/zephyr/pull/17295
```
uart:~$ gatt metrics on
GATT metrics test Service on
uart:~$ gatt register 
Registering test vendor services
uart:~$ gatt metrics off
GATT metrics test Service off
uart:~$ gatt metrics on
[00:00:28.719,200] <err> bt_gatt: Unable to register handle 0x0012
uart:~$ 
```

joerchan, 2019-08-08:
#17295 fixed a bug from #16178, together they fix this.
I have added both to #18122 for backporting to the v1.14-branch.

MariuszSkamra, 2019-08-16:
As @joerchan mentioned, this issue is fixed with #17295 together with #16178

closed 2019-08-16 by MariuszSkamra