zephyrproject-rtos/zephyr #17983: Notification enabled before secured

pdunaj, 2019-08-02:
**Describe the bug**
See `notify_connected` in `subsys/bluetooth/host/conn.c`.
When all callbacks are executed gatt is notified. It seems OK except that it leads to data exchange before security is established (we get information about service's notification).

I seen it while trying to move the code where we sent data for the first time in the app. Can be worked around but maybe should be fixed some other way.

**To Reproduce**
N/A

**Expected behavior**
I expected no exchange before security is set.

**Impact**
?

**Screenshots or console output**
N/A

**Environment (please complete the following information):**
master

**Additional context**
N/A

Vudentz, 2019-08-14:
So what data is being exchanged? If there are notifications/indications that don't require security that should be fine, or there is something else going on?

carlescufi, 2019-08-23:
@pdunaj can you please add more information here?

pdunaj, 2019-08-28:
Hi @Vudentz , when the bug was discovered our app was starting to send HID data when callback informing about notifications being enabled was delivered. This callback can be delivered however before security is established. We added a check for security and now data is sent only when both notification and security are enabled.
Since notification enabled callback is delivered before security is established I was wondering if this is a correct behavior. Theoretically somebody could push the information about notification enabling before security is done and there will be no way of validating that I think. Then device will try to send data to central that did not enable notifications.

pdunaj, 2019-08-30:
Hi, it seems that the problem is inside service that we use as it sets BT_GATT_PERM_READ (instead of BT_GATT_PERM_READ_ENCRYPT) for this notification. This is not configurable. HID service is in NCS and problem was already mentioned to @kapi-no .

The same thing is present in DIS and BAS in Zephyr. They define characteristics with PERM_READ with no option to configure it.

pdunaj, 2019-08-30:
BTW - thanks @joerchan for pointing the problem.

carlescufi, 2019-08-30:
@Vudentz @joerchan 

We now discussed this with Joakim and went back to the discussions we had when we implemented this in the SoftDevice.

The GAP spec says:

```
10.3.1.1 Handling of GATT indications and notifications

A client “requests” a server to send indications and notifications by
appropriately configuring the server via a Client Characteristic Configuration
Descriptor. Since the configuration is persistent across a disconnection and
reconnection, security requirements must be checked against the configuration
upon a reconnection before sending indications or notifications. When a server
reconnects to a client to send an indication or notification for which security is
required, the server shall initiate or request encryption with the client prior to
sending an indication or notification. If the client does not have an LTK
indicating that the client has lost the bond, enabling encryption will fail.
```

Note in particular the sentence `security requirements must be checked against the configuration
upon a reconnection before sending indications or notifications`.
Back when we implemented this internally we decided that trying to notify or indicate a characteristic whose CCC's *write* security permissions were higher than the current security level of the connection should fail, so the corresponding `notify()/indicate()` functions return `ERR_FORBIDDEN`.
In the case of Zephyr we have two choices in my opinion:

1. Return an error when the application calls `bt_gatt_notify()` or `bt_gatt_indicate()` with the current connection security level lower than the *write* security permissions of the CCC.
2. Trigger the security procedure automatically whenever the application calls `bt_gatt_notify()` or `bt_gatt_indicate()` with the current connection security level lower than the *write* security permissions of the CCC.

PR #18898 (2019-09-19): Bluetooth: GATT: Add support to setting permission on CCCD
This adds support to set different permissions to CCCD so security can
be checked when enabling notification which conforms to:

BLUETOOTH CORE SPECIFICATION Version 5.1 | Vol 3, Part G page 2360:

  '3.3.3.3 Client Characteristic Configuration

   Authentication and authorization may be required by the server to
   write the configuration descriptor.'

In addition to that also ensure that notification are not re-enabled
until the proper security level is reached to conform to the following
statement:

  '10.3.1.1 Handling of GATT indi [... cut]
commit 03b9ce48 (2019-09-19): Bluetooth: GATT: Add support to setting permission on CCCD

This adds support to set different permissions to CCCD so security can
be checked when enabling notification which conforms to:

BLUETOOTH CORE SPECIFICATION Version 5.1 | Vol 3, Part G page 2360:

  '3.3.3.3 Client Characteristic Configuration

   Authentication and authorization may be required by the server to
   write the configuration descriptor.'

In addition to that also ensure that notification are not re-enabled
until the proper security level is reached to conform to the following
statement:

  '10.3.1.1 Handling of GATT ind [... cut]
closed 2019-09-19 by jhedberg