zephyrproject-rtos/zephyr #17990: BLE Mesh When IV update test procedure

menglingao1997, 2019-08-03:
https://github.com/zephyrproject-rtos/zephyr/blob/0440a815a91b83bdb27e272b9770f558c1b8975c/subsys/bluetooth/mesh/net.c#L672-L678

When `CONFIG_BT_MESH_IV_UPDATE_TEST` not enable, and `BT_MESH_IVU_TEST` flag is enable, must ignore the limit of IV daration.but just as above code show, does not meet specifications.

aescolar, 2019-08-03:
CC @jhedberg

jhedberg, 2019-08-03:
The `BT_MESH_IVU_TEST` flag can only be changed using the `bt_mesh_iv_update_test()` API, and that API is only available when `CONFIG_BT_MESH_IV_UPDATE_TEST=y`, so I don't see any issue here - the flag cannot be set if the Kconfig isn't also set.

menglingao1997, 2019-08-04:
> The `BT_MESH_IVU_TEST` flag can only be changed using the `bt_mesh_iv_update_test()` API, and that API is only available when `CONFIG_BT_MESH_IV_UPDATE_TEST=y`, so I don't see any issue here - the flag cannot be set if the Kconfig isn't also set.

You are right,I mistake.

closed 2019-08-04 by menglingao1997