zephyrproject-rtos/zephyr #17991: Cannot generate coverage reports on qemu_x86_64

andrewboie, 2019-08-03:
```
$ sanitycheck --coverage -p qemu_x86_64 -T tests/kernel/ -T tests/lib/
JOBS: 6
Cleaning output directory /home/apboie/projects/zephyr3/zephyr/sanity-out
Building testcase defconfigs...
Filtering test cases...
75 tests selected, 19061 tests discarded due to filters
total complete:   75/  75  100%  failed:    0
Generating coverage files...
geninfo: WARNING: no .gcda files found in /home/apboie/projects/zephyr3/zephyr/sanity-out - skipping!
lcov: ERROR: no valid records found in tracefile /home/apboie/projects/zephyr3/zephyr/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/apboie/projects/zephyr3/zephyr/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/apboie/projects/zephyr3/zephyr/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/apboie/projects/zephyr3/zephyr/sanity-out/coverage.info
genhtml: ERROR: no valid records found in tracefile /home/apboie/projects/zephyr3/zephyr/sanity-out/coverage.info
75 of 75 tests passed with 0 warnings in 190 seconds
```

Impact: we are not able to determine code coverage of the kernel on an SMP-enabled system

aescolar, 2019-10-08:
labeling as enhancement assuming this never worked, please change if you disagree

kkiyota, 2020-03-10:
I have the same problem with qemu_cortex_m3.  Looks like "-DCONFIG_COVERAGE=y" is missing with cmake.

cfriedt, 2020-05-21:
I can't actually seem to run the qemu_cortex_m3 target. This was working fine quite recently.
```
[143/144] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: cortex-m3
qemu-system-arm: -serial unix:/tmp/slip.sock: Failed to connect socket /tmp/slip.sock: No such file or directory
qemu-system-arm: -serial unix:/tmp/slip.sock: could not connect serial device to character backend 'unix:/tmp/slip.sock'
FAILED: zephyr/CMakeFiles/run 
cd /home/cfriedt/workspace/zephyrproject/zephyr/build && /home/cfriedt/zephyr-sdk-0.11.2/sysroots/x86_64-pokysdk-linux/usr/bin/qemu-system-arm -cpu cortex-m3 -machine lm3s6965evb -nographic -vga none -net none -pidfile qemu.pid -chardev stdio,id=con,mux=on -serial chardev:con -mon chardev=con,mode=readline -serial unix:/tmp/slip.sock -kernel /home/cfriedt/workspace/zephyrproject/zephyr/build/zephyr/zephyr.elf
ninja: build stopped: subcommand failed.
```

nashif, 2020-09-11:
this is not supported and this qemu platform is not enabled for coverage, so reverting back to enhancement.

dcpleung, 2021-03-10:
Done in #29873

closed 2021-03-10 by dcpleung