zephyrproject-rtos/zephyr #17996: BUILD_ASSERT not active in three of five shippable platforms

pabigot, 2019-08-04:
Lack of a terminating semicolon after a `BUILD_ASSERT()` macro in a push to #17749 was diagnosed only in two platforms; three shippable builds plus my development environment didn't see it and passed.

I'm wondering if the recent appearance of these messages, captured from an initial `west build` targeting nrf52840_pca10056, in conjunction with the description in #8160, is why three of five shippable platforms, as well as development builds with SDK 0.10.1, do not enable `BUILD_ASSERT`:

```
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- The ASM compiler identification is GNU
...
-- Configuring C Compiler
-- Performing Test HAVE_C_FLAG_STD_C11
-- Performing Test HAVE_C_FLAG_STD_C11 - Failed
-- Performing Test HAVE_C_FLAG_STD_C99
-- Performing Test HAVE_C_FLAG_STD_C99 - Failed
-- Performing Test HAVE_C_FLAG_STD_C89
-- Performing Test HAVE_C_FLAG_STD_C89 - Failed
```
Claims that GCC 8.3.0 doesn't have those flags are blatantly false, and `BUILD_ASSERT` should be default-enabled everywhere it could possibly be supported.

carlescufi, 2019-08-05:
I believe those new messages are triggered by the inclusion of civetweb as a module, whose `CMakeLists.txt` seems to add a lot of processing, including the messages that you show here. This should be limited to civetweb therefore.

SebastianBoe, 2019-08-05:
It seems civetweb is missing a

if(CONFIG_CIVETWEB)

in it's build scripts.

@PiotrZierhoffer , @tbursztyka

PiotrZierhoffer, 2019-08-07:
I moved the check earlier, the PR is here: https://github.com/civetweb/civetweb/pull/778

SebastianBoe, 2019-08-07:
> For the sake of sanity I left the cmake version verification before the relevant ifs, I don't think it should affect the rest of the build process

AFAICT the policy changes will not have any affect and are therefore harmless.

pabigot, 2019-08-07:
This definitely needs to be in 2.0.0 (actually, needs to be in ASAP) regardless of when upstream takes the patch.

PiotrZierhoffer, 2019-08-07:
I created the PR here: https://github.com/zephyrproject-rtos/zephyr/pull/18107

I must admit I don't fully understand what does this issue mean, but now Civetweb's cmake is stopped much earlier.

PiotrZierhoffer, 2019-08-08:
The PR is merged - what should we do with this issue is an open question now.

SebastianBoe, 2019-08-08:
Could @pabigot re-test and see if it is still so that "BUILD_ASSERT not active in three of five shippable platforms" ?

pabigot, 2019-08-08:
#18121 failed on all platforms as it should have.  I have a feeling something still isn't right, but until it shows up again we'll call this good.

closed 2019-08-08 by pabigot

SebastianBoe, 2019-08-08:
Great, thank you @pabigot

pabigot, 2019-08-08:
For the record: The [shippable run](https://app.shippable.com/github/zephyrproject-rtos/zephyr/runs/47565/summary/console) that kicked off this issue showed failures in matrix builds 3 and 5 due to lack of a semicolon after a `BUILD_ASSERT()`.  I inferred this was due to that macro expanding to nothing, which was probably incorrect, as was my count of failed platforms.

I can't see why the error would only show up in two of five platforms; maybe there's a compiler difference under the hood.