zephyrproject-rtos/zephyr #17997: fix extern "C" use throughout Zephyr

pabigot, 2019-08-04:
Inspired by #17993, the latest in an unending series of patches to fix broken C++ support, I did:
```
lilith[632]$ git grep -A12 'extern "C"' | grep '#include' | sed -e 's@\.h-.*$@.h@' | uniq | wc
    303     303   15609
```
That's at least 303 include files in the Zephyr tree that probably get this wrong, i.e. have an `#include` within the first ten lines after an `extern "C"` block.  Excluding `ext/hal` (found in atmel, nordic, nxp, openisa, ti) reduces it to a mere 88.

As long as Zephyr claims C++ compatibility, even if only for application code, this should be cleaned up at least in non-HAL code before 2.0, and there should be some attempt toward adding a check for new occurrences to the commit checking code.

jukkar, 2019-08-05:
I created a PR #18023 that changes the include file location in networking related include files.

pfalcon, 2019-08-05:
> Inspired by #17993,

I fail to see to see any inspiration there, it just says it moves headers around without explaining why. Some time later, another guy will submit patch which moves them back. Of course, without any explanation either.

pabigot, 2019-08-05:
@pfalcon It would be fair to request that the commits and/or PR explain why it's done that way.  Not everybody will recognize the problem.  Or just link back here to this sample text:

Declarations that use C linkage should be placed within extern "C" so the language linkage is correct when the header is included by a C++ compiler.

Similarly #include directives should be outside the extern "C" to ensure the language-specific default linkage is applied to any declarations provided by the included header.

See: https://en.cppreference.com/w/cpp/language/language_linkage

pfalcon, 2019-08-05:
> Similarly #include directives should be outside the extern "C" to ensure the language-specific default linkage is applied to any declarations provided by the included header.

Makes sense, hopefully authors of specific patches might mention that in PR commit messages, because the above fact might not be immediately obvious, and I guess effect of not following that rule might differ from header to header (so, some people might place one header inside `extern "C"`, see no ill effects, and decide that moving headers around is useless sinecure; or worse, someone might decide that instead of putting `extern "C"` *inside a header*, it's "better" to put *a header include inside* `extern "C"` block).

pabigot, 2019-08-05:
@pfalcon Good point; I've updated the explanation and moved it to the description of this issue.  It should probably go into the developer area of the documentation.  I'll try to remember to do that if/when I get back to doing C++ stuff in Zephyr.

pabigot, 2019-08-12:
The following script can be used to detect situations where an `extern "C"` block contains additional `#include` directives.  The suite of patches in #18196 results in six remaining files that do this, mostly because they're including generated headers or the conditions for inclusion are more complex to refactor than is worth the effort.

```
git grep -l 'extern "C"' -- '**/*.h' \
| grep -v '^ext/' \
| while read LINE ; do
  sed -n -e '/__cplusplus/,/__cplusplus/p' ${LINE}\
    | grep '\#.*include' \
    >/tmp/yout$$
  if [ -s /tmp/yout$$ ] ; then
    echo ${LINE}
  fi
done
rm -f /tmp/yout$$
```

ioannisg, 2019-08-13:
@pabigot @carlescufi can this issue be closed now?

pabigot, 2019-08-13:
@ioannisg no, there are another ~15 patches in #18196 that need to be pulled out into targeted PRs that won't request the entire world to review them.  I'll close this when it's all done.

PR #18245 (2019-08-19): aggregated remaining patches to correct use of extern "C"
This PR, in combination with #18243 and #18244, should be sufficient to close #17997.

If the complexity is too high it can be broken down into chunks.  I figured one big PR was better than the eight smaller ones that fell out of an initial partition.
closed 2019-08-19 by ioannisg