zephyrproject-rtos/zephyr #17998: STM32 (Nucleo L476RG) SPI pins floating

mrusme, 2019-08-04:
Not sure whether that's a bug or simply me holding it wrong, but since there's so little documentation available on this I cannot figure out what's going on. I'm using the following code to test:

```cpp
static const struct spi_config spi_cfg = {
    .frequency = 4000000,
    .operation = SPI_OP_MODE_MASTER | SPI_WORD_SET(8) | SPI_TRANSFER_MSB | SPI_MODE_CPOL,
    // .slave = 0,
};

struct device * spi_dev;

static bool spi_init(void)
{
    spi_dev = device_get_binding(DT_SPI_1_NAME);

    if (spi_dev == NULL) {
        printk("Could not get %s device\n", DT_SPI_1_NAME);
        k_sleep(2000);
        return false;
    }

    return true;
}

uint8_t spi_test_send(uint8_t cmd)
{
    int err;
    static u8_t tx_buffer[1];
    static u8_t rx_buffer[1];

    tx_buffer[0] = cmd;

    const struct spi_buf tx_buf = {
        .buf = tx_buffer,
        .len = sizeof(tx_buffer)
    };
    const struct spi_buf_set tx = {
        .buffers = &tx_buf,
        .count = 1
    };

    struct spi_buf rx_buf = {
        .buf = rx_buffer,
        .len = sizeof(rx_buffer),
    };

    const struct spi_buf_set rx = {
        .buffers = &rx_buf,
        .count = 1
    };

    err = spi_transceive(spi_dev, &spi_cfg, &tx, &rx);
    if (err) {
        printk("SPI error: %d\n", err);
    } else {
        printk("TX sent: %x\n", tx_buffer[0]);
        printk("RX recv: %x\n", rx_buffer[0]);
        return rx_buffer[0];
        // tx_buffer[0]++;
    }
}

uint8_t myData[] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00};

int main(void)
{
    bool spiInitialized = false;

    while(!spiInitialized) {
        spiInitialized = spi_init();
    }

    spi_test_send(1);

    for(int i=0; i<10; i++) {
        k_sleep(15);
        myData[i] = spi_test_send(0);
    }

     k_sleep(10000);
}
```

My prj.conf looks like this:

```
CONFIG_SERIAL=y

CONFIG_CPLUSPLUS=y
CONFIG_PRINTK=y
CONFIG_STDOUT_CONSOLE=n
CONFIG_I2C=y
CONFIG_SENSOR=y
CONFIG_MULTITHREADING=y
CONFIG_GPIO=y

CONFIG_PINMUX=y
CONFIG_SPI=y
CONFIG_SPI_1=y
```

After deploying this on the L476RG, I'm measuring the voltage on CN8/A2 (-> PA_4), as this is documented to be [the default CS pin used by Zephyr](https://docs.zephyrproject.org/latest/boards/arm/nucleo_l476rg/doc/index.html#default-zephyr-peripheral-mapping). The pin is floating at ~1.8V instead of being at 0V and 3.3V.

I've made a test using STM32-Arduino and deploying the following sketch on my board:

```cpp
void setup() {
  Serial.begin(115200);
  pinMode(A2, OUTPUT);
}

void loop() {
  digitalWrite(A2, LOW);
  delay(5000);
  digitalWrite(A2, HIGH);
  delay(5000);
}
```

Here I could successfully measure the same pin alternating between 0V and 3.3V.

Is there something missing in order to get this working correctly or is it simply a bug?

 - OS: Gentoo Linux
 - Toolchain: Zephyr SDK, latest release

FRASTM, 2019-08-06:
Hello,

not sure what your SPI configuration, but actually the PA4 is SPI1_NSS pin when Alternate function (AF5) is choosen.
Did you check the refMan of th STM32L476  about the "42.4.5 Slave select (NSS) pin management" to configure the SPI bus and

erwango, 2019-08-30:
@mrusme, did you manage to solve your issue?

I had a look to general NSS pin set up and indeed, there might be an issue with _NSS pins definitions. Changing definition will make it alternate.
```
#define STM32L4X_PINMUX_FUNC_PA4_SPI1_NSS                                      \
	(STM32_PINMUX_ALT_FUNC_5 | STM32_PUSHPULL_PULLUP)
```
I'll check the reason why it was initially set to `STM32_PUSHPULL_NOPULL` before raising an issue. I suspect NSS pins (output enable) are not often used, explaining why this was not seen earlier (either CS pin or software mgmt would be preferred).

Then please note that toggling NSS pin is not equivalent to Arduino GPIO toggling.
NSS toggling is done by SPI IP while Arduino GPIO toggling is done by GPIO controller. Then same could be done in Zephyr like in blinky sample.

mrusme, 2019-08-30:
@erwango thanks for getting back. To be honest, I gave up on Zephyr after encountering more issues down the road which I couldn't figure out myself within a reasonable amount of time. Tried getting help on the IRC channel, which also didn't lead to much success back then.

Since I diverted to another stack this has become irrelevant to me, hence I can close this issue.

erwango, 2019-09-02:
@mrusme, sorry to hear that. Btw, IRC is not much in use these days since we switched to slack some months ago(https://zephyrproject.slack.com/).

mrusme, 2019-09-02:
@erwango unfortunately that's of not much use for anyone else but you guys:

[image]

[Pointed this out over a month ago](https://twitter.com/mrusme/status/1155866788736983040) to your somewhat active Twitter account, haven't heard back about it, which gave me the impression this channel is not for the community but rather only the companies working directly with you.

erwango, 2019-09-02:
@mrusme, well that's strange, for sure there are people from the open community in there. So not only people from companies. Wonder why access is/looks blocked.

erwango, 2019-09-02:
@mrusme :
Actually you need to access via the Slack invite link first : https://tinyurl.com/y5glwylp.
Maybe this need to be documented better indeed.

mrusme, 2019-09-02:
@erwango maybe change the link on [the Zephyr contact page](https://www.zephyrproject.org/about/#contact-us) as it still is invalid? The one you just wrote seems to be valid, without actually going through the whole steps of registration.

erwango, 2019-09-02:
@mrusme, this would make sense. Will submit a change.

erwango, 2019-09-03:
Reopenning the issue, as it reveals a real issue that should be fixed.

PR #19017 (2019-09-10): drivers/pinmux: stm32: Use pull-up for _SPI_NSS pins
To work efficiently, SPI_NSS pins require pull-up configuration.
Fix this for whole STM32 series.

Fixes #17998
commit 90df6a32 (2019-09-10): drivers/pinmux: stm32: Use pull-up for _SPI_NSS pins

To work efficiently, SPI_NSS pins require pull-up configuration.
Fix this for whole STM32 series.

Fixes #17998
closed 2019-09-10 by MaureenHelm