zephyrproject-rtos/zephyr #18002: Flash using open source stlink, instead of SEGGER jlink?

jiapei100, 2019-08-05:
Hi, all:

It seems I cannot install **JLink v648b**:

```console
➜ jlink sudo dpkg -i JLink_Linux_V648b_x86_64.deb
[sudo] password for lvision: 
Selecting previously unselected package jlink.
(Reading database ... 506734 files and directories currently installed.)
Preparing to unpack JLink_Linux_V648b_x86_64.deb ...
Removing /opt/SEGGER/JLink ...
/opt/SEGGER/JLink not found (OK)
Unpacking jlink (6.482) ...
dpkg: dependency problems prevent configuration of jlink:
jlink depends on libncurses5 (>= 5.5); however:
Package libncurses5 is not installed.

dpkg: error processing package jlink (--install):
dependency problems - leaving unconfigured
Errors were encountered while processing:
jlink
```

Is it possible to set the **flash tool** from [SEGGER jlink](https://www.segger.com/products/debug-probes/j-link/) to [stlink](https://github.com/texane/stlink)?

Cheers
Pei

vanwinkeljan, 2019-08-05:
Once you made a build you can use the west flash command to list the available _runners_ and select one that suits your needs
```
 west flash -H
```

One question on the jlink install issue, any reason why you did not install the missing dependencies (`sudo apt-get install -f` )?

jiapei100, 2019-08-05:
@vanwinkeljan 

1.

```console
➜  build git:(master) ✗ west flash -H
ninja: no work to do.
All Zephyr runners which support flash:
  em-starterkit, bossac, dfu-util, esp32, hifive1, jlink, nios2,
  nrfjprog, arc-nsim, openocd, pyocd, intel_s1000, blackmagicprobe
(Not all may work with this build, see available runners below.)
Build directory:
  ....../zephyrproject/zephyr/build
Board:
  hifive1_revb
CMake cache:
  ....../zephyrproject/zephyr/build/CMakeCache.txt
Available flash runners:
  jlink
Additional options for available flash runners:
jlink options:
  --dt-flash {Y,y,N,n,yes,no,YES,NO}
                        If 'yes', use configuration generated by device tree
                        (DT) to compute flash addresses.
  --device DEVICE       device name
  --iface IFACE         interface to use, default is swd
  --speed SPEED         interface speed, default is autodetect
  --tui                 if given, GDB uses -tui
  --gdbserver GDBSERVER
                        GDB server, default is JLinkGDBServer
  --gdb-port GDB_PORT   pyocd gdb port, defaults to 2331
  --tool-opt TOOL_OPT   Additional options for JLink Commander, e.g.
                        '-autoconnect 1'
  --commander COMMANDER
                        J-Link Commander, default is JLinkExe
  --erase               if given, mass erase flash before loading
  --reset-after-load, --no-reset-after-load
                        reset after loading? (default: no)
Default flash runner:
  jlink
Cached common runner configuration:
  --build_dir=....../zephyrproject/zephyr/build
  --board_dir=....../zephyrproject/zephyr/boards/riscv/hifive1_revb
  --elf_file=....../zephyrproject/zephyr/build/zephyr/zephyr.elf
  --hex_file=....../zephyrproject/zephyr/build/zephyr/zephyr.hex
  --bin_file=....../zephyrproject/zephyr/build/zephyr/zephyr.bin
  --gdb=/opt/zephyr-sdk/riscv32-zephyr-elf/bin/riscv32-zephyr-elf-gdb
  --openocd=/usr/local/bin/openocd
  --openocd_search=/usr/local/share/openocd/scripts
Runner-specific information:
  jlink:
    Cached runner-specific options:
      --dt-flash=y
      --device=FE310
      --iface=JTAG
      --speed=4000
      --tool-opt=-jtagconf -1,-1
      --tool-opt=-autoconnect 1
    Capabilities:
      RunnerCaps(commands={'attach', 'debug', 'flash', 'debugserver'}, flash_addr=True)
```

There is NO libncurses5 in my repository at all...

```console
➜  build git:(master) ✗ apt show libncurses5.so
N: Unable to locate package libncurses5.so
N: Couldn't find any package by glob 'libncurses5.so'
N: Couldn't find any package by regex 'libncurses5.so'
N: Unable to locate package libncurses5.so
N: Couldn't find any package by glob 'libncurses5.so'
N: Couldn't find any package by regex 'libncurses5.so'
E: No packages found
➜  build git:(master) ✗ apt show libncurses6   
Package: libncurses6
Version: 6.1+20181013-2ubuntu2
Status: install ok installed
Priority: required
Section: libs
Source: ncurses
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Craig Small <csmall@debian.org>
Installed-Size: 334 kB
Depends: libtinfo6 (= 6.1+20181013-2ubuntu2), libc6 (>= 2.14)
Recommends: libgpm2
Homepage: https://invisible-island.net/ncurses/
Download-Size: unknown
APT-Manual-Installed: no
APT-Sources: /var/lib/dpkg/status
Description: shared libraries for terminal handling
 The ncurses library routines are a terminal-independent method of
 updating character screens with reasonable optimization.
 .
 This package contains the shared libraries necessary to run programs
 compiled with ncurses.
```

In fact, the **BEST** solution is to let **SEGGER JLink** support **libncurses6**.
Any further suggestions?

Cheers
Pei

vanwinkeljan, 2019-08-05:
> In fact, the BEST solution is to let SEGGER JLink support libncurses6.

True, but that is upto SEGGER

> Any further suggestions?

According to the [ubuntu package repository](https://packages.ubuntu.com/search?keywords=libncurses&searchon=names&suite=all&section=all) libncurses5 still exists for all supported ubuntu versions, you could try to install one of those.

carlescufi, 2019-08-05:
Yes please, this is a Segger issue, not a Zephyr one. Closing this, please report this to Segger instead.

closed 2019-08-05 by carlescufi

jiapei100, 2019-08-05:
Problem solved by **dpkg-deb**. JLink is **NOW** copied and pasted **WITHOUT** installation, which seems to work ALREADY. 
Thank you... please close the issue.