zephyrproject-rtos/zephyr #18005: BLE Mesh When Friend Clear Procedure Timeout

menglingao1997, 2019-08-05:
https://github.com/zephyrproject-rtos/zephyr/blob/77db273f6f84b3d94d34b5e72070ec6bc31e41cf/subsys/bluetooth/mesh/friend.c#L667-L693

According to the specification,the Friend Clear Procedure timer shall be started with `two times` of Friend Poll Timeout value.

``` C
duration = k_uptime_get_32() - frnd->clear.start;
frnd->clear.start = k_uptime_get_32() + (2 * frnd->poll_to);
```

aescolar, 2019-08-05:
CC @jhedberg

jhedberg, 2019-08-13:
I don't see a bug here. `frnd->clear.start` is the timestamp when the procedure was started, i.e. it should be initialised to `k_uptime_get_32()`. Am I misunderstanding something, or are you? 😄

jhedberg, 2019-08-14:
Closing this since the current code looks correct to me. I think you might have confused frnd->clear.start being a timeout/timer when in fact it's just a timestamp that's used later to determine if enough time has passed. @trond-snekvik FYI

closed 2019-08-14 by jhedberg