zephyrproject-rtos/zephyr #18013: BLE Mesh On Net Buffer free issure

menglingao1997, 2019-08-05:
**FUNCTION1**
https://github.com/zephyrproject-rtos/zephyr/blob/2b84c726c8893eaaa1bf26dfddf2798b159cd47d/subsys/bluetooth/mesh/friend.c#L90-L111

**FUNCTION2**
https://github.com/zephyrproject-rtos/zephyr/blob/2b84c726c8893eaaa1bf26dfddf2798b159cd47d/subsys/bluetooth/mesh/adv.c#L242-L252

**FUNCTION3**
https://github.com/zephyrproject-rtos/zephyr/blob/2b84c726c8893eaaa1bf26dfddf2798b159cd47d/subsys/bluetooth/mesh/friend.c#L166-L205

**FUNCTION4**
https://github.com/zephyrproject-rtos/zephyr/blob/2b84c726c8893eaaa1bf26dfddf2798b159cd47d/subsys/bluetooth/mesh/adv.c#L159-L196

when the buffer create from **FUNCTION1**, then the buf `ref=1`, we send this buffer use **FUNCTION2**, this time the buf `ref=2`, when the friend terminated buf buf also in `adv_queue`, the **FUNCTION3** will be call, in this function unref the last_buf, also set BUSY=0, so buf `ref=1`,when **FUNCTION4** get cpu times, will get buffer from `adv_queue`, but the BUSY flag we also cleared, so this buffer not send, but the buf also `ref=1`. so at this time the **memory leak** occurs.

**If I infer that it is correct, Can be modified like this:**
``` C
 	if (BT_MESH_ADV(buf)->busy) { 
 		BT_MESH_ADV(buf)->busy = 0U; 
 		adv_send(buf); 
 	} 
        else{
                net_buf_unref(buf);
        }

```

menglingao1997, 2019-08-05:
@jhedberg

jhedberg, 2019-08-05:
@menglingao1997 your suggested fix looks correct. Do you want to try creating a pull request for it? I'm still on vacation, so I'll be able to do it earliest next week.

jhedberg, 2019-08-05:
Isn't this a duplicate of #17936 ?

menglingao1997, 2019-08-06:
look like this, and may be able to solve in the same way.

commit 5f6043bc (2019-08-14): Bluetooth: Mesh: Fix canceled buffer memory leak

In case a queued buffer is canceled before sending we have to unref
it, since that's what adv_send() would do as well.

Fixes #17936
Fixes #18013
closed 2019-08-14 by jhedberg